bb-qq/r8152 #254: DS920 + & UGREEN USB C to 2.5G Ethernet Adapter (Type-C)

DudeUseNas, 2022-10-17:
### Description of the problem

Hi all,

I'm new here.
I'm trying to upgrade my connection with NAS and my brand New MAcBookPro using an Ethernet Adaptor.
I've already installed a Gemini package from her. Connection is ok. Can work on the NAS for photo editing.
But I can not succeed in iincreasing the speed ...   

Thanks for your help !

My best,
D.

### Description of your products

> -  SYNOLOGY NAS DS 920 +
> - Last version of DSM (updated today)
> - UGREEN USB C to 2.5G Ethernet Adapter (Type-C)
> - CalDigit TS3 plus

### Description of your environment

> - Connect directly to my MacBook using CalDigit TS3 plus in USB-C (Same behavior when connected directly to the laptop)
> - Apple M1 Pro
> - Ethernet switch: CalDigit TS3 plus
> - UGreen Cable category Cat.7)

DudeUseNas, 2022-10-25:
Thanks for your answer but CallDigit TS3 Plus as a dedicated data USB-C port with 10Gbps.
By the way, when I connect the UGreen Ethernet Adapter directly on the MacBook, this does not work  ....

Jamestec, 2022-10-25:
Is your setup NAS <-> UGREEN 2.5GbE <-> Ethernet cable <-> CalDigit TS3 plus <-> 10Gbps Type C <-> MacBook?

Sjekke, 2022-10-25:
CalDigit TS3 is not a switch. 
It only provides 1Gb on Ethernet
To use the 10gbit USB-C, I think you need an extra ethernet adapter ...

DudeUseNas, 2022-10-25:
> Is your setup NAS <-> UGREEN 2.5GbE <-> Ethernet cable <-> CalDigit TS3 plus <-> 10Gbps Type C <-> MacBook?

No, My Setup is : NAS <->Ethernet cable <-> UGREEN 2.5Gbps <-> 10Gbps USB-C on CalDigit TS3 plus <-> MacBook

DudeUseNas, 2022-10-25:
> CalDigit TS3 is not a switch. It only provides 1Gb on Ethernet To use the 10gbit USB-C, I think you need an extra ethernet adapter ...

But this should work if I bypass the CallDigit, isn'it ?

Jamestec, 2022-10-25:
> > Is your setup NAS <-> UGREEN 2.5GbE <-> Ethernet cable <-> CalDigit TS3 plus <-> 10Gbps Type C <-> MacBook?
> 
> No, My Setup is : NAS <->Ethernet cable <-> UGREEN 2.5Gbps <-> 10Gbps USB-C on CalDigit TS3 plus <-> MacBook

The Ethernet ports on the 920+ are only 1Gbps.

Neither the 920+ or CalDigit provide 2.5GbE, so you'll need two 2.5GbE adaptors.

DudeUseNas, 2022-10-25:
Oh ok. Thanks @Jamestec ! So the issues come from my 920+ Synology Nas. I need another ethernet switch between the NAS and the cable ? Do you have any suggestions ?

Jamestec, 2022-10-25:
I'd buy one that has a USB Type A end, since the 920+ only has USB Type A. Also connect it to the front port since the rear can be dodgy.
You can see compatible dongles here, search for 920+: https://github.com/bb-qq/r8152/wiki/Compatibility
I personally have the 920+ and the Alxum 2.5Gbps Ethernet Adapter since that is the cheapest adaptor available to me.

DudeUseNas, 2022-10-25:
Fine. I will try.
So, there is no possibility to use Ethernet port on this model of NAS as both are only 1Gbps ... Is that correct ?
That's why to talk about using USB port. But what is the difference between the front ad the back ? They should be the same ones ...

Jamestec, 2022-10-25:
> So, there is no possibility to use Ethernet port on this model of NAS as both are only 1Gbps ... Is that correct ?

If you transfer many files and have a switch with at least one 2.5Gbps Ethernet port, you can connect both 1GbE of the 920+ to the switch (two Ethernet cables and different IP addresses), then connect the 2.5Gbps port of the switch to the UGREEN USB C to 2.5G Ethernet Adapter and MacBook.
      1GbE <-> 1GbE or more
920+                                   Switch   2.5GbE <-> UGREEN <-> ( CalDigit maybe) <-> MacBook
      1GbE <-> 1GbE or more
You should be able to get > 1Gbps with this setup if you spit up the files between the two IP addresses of the 920+ (e.g. connect to two SMB servers, 192.168.1.120 and 192.168.1.121 and split the files between the two).

I believe there's an option to bond the two the 1GbE ports of the 920+, but you'll still need a switch with at least one 2.5GbE and it's only > 1GbE when there's traffic both ways, i.e. if you're only downloading from the NAS, you get 1Gbps, but if you're downloading and uploading, they both will be 1Gbps, but combined traffic can be > 1Gbps (at least the last time I tested, this was the behaviour).

> But what is the difference between the front ad the back ? They should be the same ones ...

Front is soldered directly to the PCB with the CPU etc., rear is actually via a cable. Although the cable seemed secure on mine, sometimes the 920+ refuses to run the 2.5GbE adaptor at 2.5Gbps, only 1Gbps when using the rear port. Front port doesn't have this problem. It's a common problem on Synology units.

DudeUseNas, 2022-10-25:
Thanks a lot for these precise informations.
I do believe that this configuration works. But, I also use my NAS connected to the internet in order to get files when I'm out of office... So, I need to reserve one 1GbE port for this.
Tha Alxum 2.5Gbps Ethernet Adapter might be my only option to increase the speed ...

Sjekke, 2022-10-25:
> Front is soldered directly to the PCB with the CPU etc., rear is actually via a cable. Although the cable seemed secure on mine, sometimes the 920+ refuses to run the 2.5GbE adaptor at 2.5Gbps, only [...]

I don't have this problem on my 916+ ...

Sjekke, 2022-10-25:
> Thanks a lot for these precise informations.
> 
> I do believe that this configuration works. But, I also use my NAS connected to the internet in order to get files when I'm out of office... So, I need to reserve one 1GbE port for this.
> 
>

Auch ... what a security issue ...

DudeUseNas, 2022-10-25:
@Sjekke I'm aware that's a DAS is more secure. But NAS solution is much more suitable for me in my professionnal use. What about your solution to increase speed connection on your 916+ ?

Sjekke, 2022-10-26:
> @Sjekke I'm aware that's a DAS is more secure. But NAS solution is much more suitable for me in my professionnal use. What about your solution to increase speed connection on your 916+ ?

I don't have a DAS but a simple NAS 916+ with a Plugable USB 2,5gbit adapter connected to a ASUS AX11000 on 2,5Gb port. I access my NAS via a VPN, I will never connect my NAS to directly to the Internet

jglathe, 2022-12-17:
There are cases where you get only 1GbE and 10GbE as transfer modes from 10GbE adapters. I also have an Intel X540 T2 (?) adapter card which works nicely with 10GbE, but not on 5G or 2,5GbE. It then always defaults to 1GbE. So, these cards have to go to 10GbE ports on the other side... Have this in a DS1517+ with this card. I can get transfer rates >1Gb/s to this device from 2,5GbE endpoints, but additionally the DS1517+ itself needs a bit of runway (SHR2 with 6 disks 20TB?), so this may come out as a bit disappointing either. OTOH, a DS218+ with a UGreen 2,5G adapter gives really good speeds to my desktop, with a RTL8125 2,5GbE interface, and a 5GbE backbone.

bb-qq, 2022-12-31:
@DudeUseNas, have your questions been answered? If so, I would like to close the issue.

closed 2023-07-16 by bb-qq