bb-qq/r8152 #255: No issues with bb-qq driver on my nas - but question about slow speed with windows driver from club34 cac-1420

mrjohnpoz, 2022-10-18:
I find it hard to believe this has anything to do with the bb-qq driver, but ran into a weird thing, and this seems to be a active site with users using the club 3d cac-1420 driver.

And was wondering if anyone else seen this, or have a way to correct it?  So when using the windows driver realtek 10.38.117.2020 dated 9/25/2015 I get very respectable 2.37gbps doing IPerf to my ds918+ using the 2.16.3-2 driver

But if I update the driver on windows 10, to one available from club3d 
DriverVer   = 11/02/2021,10.49.20.1102

I only get  

`[ ID] Interval           Transfer     Bitrate
[  5]   0.00-10.00  sec  1.61 GBytes  1.39 Gbits/sec                  sender
[  5]   0.00-10.02  sec  1.61 GBytes  1.38 Gbits/sec                  receiver`

I looked at settings of the driver advanced, and the newer driver adds some new features like ss timeout, etc. 

I would not report this is an issue, but maybe there is something in this newer club3d windows driver that doesn't sit well with the bb-qq driver on the nas?

This is a cac-1420 on a windows 10 machine, plugged directly into another cac-1420 that is on my nas..  I have rolled back the driver on windows and get back my normal 2.37 speeds.

`[ ID] Interval           Transfer     Bitrate
[  5]   0.00-10.00  sec  2.77 GBytes  2.38 Gbits/sec                  sender
[  5]   0.00-10.01  sec  2.77 GBytes  2.37 Gbits/sec                  receiver`

I was hoping someone here might have some insight?  If not, and the powers that be don't think this sort of question is suited for here, please delete - or if you need any more info.. Just ask.. thanks.

pigsyn, 2022-10-18:
Hey, DS 920+ here
Got one club 3d cac-1420 a couple of days ago, it was impossible to negotiate link above 100M using 3 different OS (linux, macOS, win11) and 2 different switches (managed 2.5G qnap and 2.5G TrendNet ).

I tried everything, ethtool, different drivers (but not 2015 ones like you did), cables, forced 2.5G link on switch, every setting in device manager advanced settings in windows, couldn't make it work on 1 single machine.

Asked Amazon for an exchange and got Asustor AS-U2.5G2. Works everywhere.
So it has to be Realtek's fault imho.

mrjohnpoz, 2022-10-18:
I haven't had any issues with getting 2.5ge speeds until I changed the windows driver, and even then its neg a 2.5gig connection - just the performance sucks.. 

If I look up the driver for that, its the same version number that I get good speeds on the 10.38.117.2020, and its a realtek chip, etc.

loopeey, 2022-10-21:
Try the latest driver and update only in device manager? Just had a problem with receive speed but it turned out to be nas stuck writing forever that had to be fixed. Currently on 10.38 i think dated 2020 but will try latest maybe.

https://www.realtek.com/en/component/zoo/category/network-interface-controllers-10-100-1000m-gigabit-ethernet-usb-3-0-software

mrjohnpoz, 2022-10-21:
Thanks - that seems to work fine.. Grabbed the driver direct from realtek via your link.. And now it shows version 10.54.20.608, dated 6/8/2022 and that seems to work fine, iperf showing my 2.37gbps speed.. And now more options exposed in the config.

bb-qq, 2022-11-23:
Can we close this issue?

mrjohnpoz, 2022-11-23:
Sorry yeah

closed 2022-11-23 by mrjohnpoz