bb-qq/r8152 #256: DS425+ 2.16.3.2/2.16.1 version not working

Githwx, 2022-10-20:
### Description of the problem

 i upgraded the driver to 2.16.3.2 from 2.15.0.10 version (Before upgrading, the 2.5G usb worked with driver installed.) It failed to activate and is stopped and  need a repair.  i already tried a few times follow the process as Uninstall package -> Reboot -> Install package -> Run "sudo install -m 4755 -o root -........" command -> Install package but the issue remains even i tried the older version from 2.15.0.1 to 7 . Not sure how to proceed further. Mine is DS415+ with DSM 7.1.1 42962 update 1

### Description of your products

> DS 415+
> DSM 7.1.1 42962 update 1
> ethernet dongle ULT-Unite 2.5G USB 

### Description of your environment

> - NAS connects to  Ethernet Switcher 
>     - PC: Mac MIni M1
>     - Ethernet switch: TPlink SH 1005
> - cable category (Cat.6e and  Cat.7)

Githwx, 2022-11-22:
Can anyone share some advises and help? The issue lasts till now and it's frustrated that don't know how to solve it.

bb-qq, 2022-12-04:
The repair loop issue should be fixed in the latest release.
https://github.com/bb-qq/r8152/releases/tag/2.16.3-3

So I close this thread, but please feel free to re-open the thread if your issue is not fixed.

closed 2022-12-04 by bb-qq