bb-qq/r8152 #258: Unable to connect NAS DS920+ using Plugable 2.5G adapter

agreatdayeveryday, 2022-10-25:
### Description of the problem

I purchased the Plugable 2.5G adapter and it arrived today. I installed the 2.16.3-1 driver and entered the `sudo install -m 4755 -o root -D /var/packages/r8152/target/r8152/spk_su /opt/sbin/spk_su` command when it failed to install the first time. I restarted the device and was not able to connect to my NAS anymore. 

I tried both setting a static IP and using DHCP while reserving the address on my router page. 

I have tried the following steps to debug:

**Debug Method 1**
1. Plug ethernet to LAN 1
2. Set static IP in network settings to 192.168.1.96
3. Create a task scheduler on boot-up (user root) 
4. Setting the following command: synonet --set_ip -4 eth0 add 192.168.1.96/24
5. Shutdown NAS
6. Plug ethernet into Plugable 2.5G adapter
7. Start NAS
8. Unable to access https://192.168.1.96:5001

**Debug Method 1**
1. Plug ethernet to LAN 1
2. Set DHCP in network settings
3. Create a task scheduler on boot-up (user root) 
4. Setting the following commands: 
ifconfig eth0 up
synonet –dhcp eth0
6. Shutdown NAS
7. Plug ethernet into Plugable 2.5G adapter
8. Start NAS
9. Unable to access https://192.168.1.96:5001

### Description of your products

> Synology DS920+
>     - DSM version 7.1.1-42962 Update 2
> - Plugable 2.5G USB C and USB to Ethernet Adapter

### Description of your environment

> - PC: ASUS Dark Hero motherboard has 2.5G ethernet port, connected to 2.5G switch (Realtek PCIe 2.5GbE Family Controller)
> - Ethernet switch: TRENDnet 5-Port Unmanaged 2.5G Switch
> - cable category (Cat6)

Payn, 2022-10-25:
USB adapter is `eth2` (given your NAS has 2 builtin LANs) and you can't use task scheduler with boot-up trigger on it without artificial delay cause USBs are initialized later.

agreatdayeveryday, 2022-10-25:
@Payn I see, there was a comment in another issue about setting a delay or something but I'm not sure if that's relevant for my issue.

Yes, the DS920+ has two LAN ports. Previously I was connected to LAN 1.

If I can't use task scheduler to set the static IP what are my options? Should I uninstall and reinstall the drivers? The drivers are installed **before** the 2.5G adapter is connected, right? Then once drivers are installed, shutdown/restart and then connect the 2.5G adapter. If I connect the adapter before the drivers are installed I cannot access the NAS anymore.

Payn, 2022-10-25:
Well from what you wrote, you are setting static IP or DHCP on `eth0` which is LAN1, i suppose you wanted to use `eth2` (your adapter). 
And of course you can set it from task scheduler but like i said, you need a delay when using boot-up trigger.
Put `sleep 60` before your commands.

agreatdayeveryday, 2022-10-25:
Does the adapter create a new LAN 3 in the network settings? The driver is running when I last checked. I wasn't able to confirm since I couldn't access the NAS. I'll change to eth2 and add the delay tonight and reboot and see if that works. Thank you so much! Appreciate the help.

Payn, 2022-10-25:
Yes it adds third network interface (eth2) and it works the same as built-in ones. Just configure it first while connected through LAN1 and i don't see why it wouldn't work.
If other issues arise after (slow speed, disconnecting,..), that's another story.

agreatdayeveryday, 2022-10-25:
Oh! Was I supposed to keep LAN 1 connected and have a separate Ethernet cable for the 2.5G adapter? If so, I got confused as I thought I was supposed to unplug LAN 1 in order to achieve faster transfer speeds to/from the NAS to my personal PC. I did a test on my PC last night transferring a few large files from an external HDD and I was seeing over 300MB/s transfer speeds.

Payn, 2022-10-25:
You don't have to, i guess you can work with adapter only. 
As you can see on my screenshot, i have both PC and NAS connected to router for management and then separate 2.5gbit connection from PC to NAS for data transfer only (network share in windows). This way you always have a connection to NAS even if adapter fails.

agreatdayeveryday, 2022-10-25:
That makes a lot of sense, I hadn't thought about it that way only because I figured I wasn't sure what the transfer speeds would look like if I had both a gigabit and 2.5G connection.

My setup is currently:
PC connects to cat6 wall jack, from the patch panel I find the cable that is for my PC and connect to the 2.5G switch. Another cable connects my 24 port gigabit switch (used for the patch panel) to the 2.5G switch. A third cable connects to my Netgear Orbi router.

Payn, 2022-10-25:
Transfer speeds have nothing to do with what you got connected. Only what connection you are sending data through. For example i can use a connection through router and get 1gbit speed. Or i can use my direct connection through adapter and get 2.5 gbit. You are just using different IPs for each connection and can map both of them to network shares through with you will be accessing NAS.

So connect NAS (LAN1) to your router and use whatever IP it gets for management (that will be 1gbit connection).
Connect PC with separate cable to adapter plugged in NAS (LAN3), give it IP and map to network share (this will be 2.5 gbit connection).

agreatdayeveryday, 2022-10-25:
Is the 2.5G adapter connecting to a 2.5G unmanaged switch a direct connection if I also have my PC connected to the same 2.5G switch? 

My PC's motherboard has two separate Ethernet ports as well, but I'm only using the 2.5G port. In theory I can connect the 2.5G cable directly to the NAS via adapter and use the gigabit port for accessing the internet.

Payn, 2022-10-25:
I'm using both 1G and 2.5G ports on my motherboard but since you got 2.5gbit switch you could probably use just one cable to pc for everything.
Trying to visualize your setup, can you try something like this?
PC - 2.5G switch - NAS adapter
PC - 2.5G switch - router
NAS - router
So in the end your PC will be connected to both NAS and router, and your NAS will be connected both to your PC and the router. A triangle. Then just use IP of LAN3 for 2.5gbit transfers and LAN1 for management through router.
Of course if you don't plan accessing the NAS from any other device you could skip the router completely and just connect both NAS's LANs to 2.5G switch.

agreatdayeveryday, 2022-10-25:
To clarify a bit, here is my exact setup:

- My office has 4 cat6 wall jacks
- PC has two Ethernet ports (1G and 2.5G)

PC - wall jack - patch panel - 2.5G switch
NAS LAN1 - 2.5G switch
24-port Gigabit switch - 2.5G switch - Router

I don't have any other computers (i.e. HTPC) to access the NAS but eventually I plan on setting up Plex which would be connected from the Nvidia Shield Pro device.

Payn, 2022-10-25:
If i'm not mistaken, only thing you need is to just connect the 2.5gbit adapter (LAN3) to 2.5gbit switch.
Then you should be able to access NAS through both it's LANs.
I have my setup at home and sometimes need to access NAS from phone so i need the WIFI router in between.
So try that and add a network share with LAN3 IP and you should be set up with 2.5gbit.

agreatdayeveryday, 2022-10-25:
Will do, praying everything works tonight! 🙏 

Unrelated to this but one of my brand new drives (less than 2 weeks old) crashed last night, not sure if it was related to me installing 16GB ram (total 20GB now) but I have to wait until it finishes repairing/rebuilding tonight before I mess with it again.

agreatdayeveryday, 2022-10-27:
@Payn THANK YOU SO MUCH! It worked!

I connected a new ethernet cable to the adapter plugged into the front USB port while keeping LAN1 connected. I disabled and re-enabled the driver in order for LAN3 to appear then setup a static IP address, changed jumbo frames to 9000 and on my router page I reserved the new IP it assigned.

Should I still go ahead and setup a script on boot-up to run these commands?

ifconfig eth2 up
synonet –dhcp eth2

I guess I can restart the NAS now and see if it reconnects.

Getting about 230-240MB/s on transfers now. 

[image]

agreatdayeveryday, 2022-10-27:
Spoke to soon, restarted the NAS again and even though I can still see LAN3 in the network settings I cannot map the network drive anymore :( 

Tried disabling and enabling the driver. Any ideas?

Payn, 2022-10-27:
You don't need any scripts. It should work. But in the latest version there is an issue I already reported about disconnecting the adapter, where i need to restart driver every day. 
But it doesn't sound like your problem. How does it look like in Windows? I your network share red as disconnected?

agreatdayeveryday, 2022-10-27:
I'm not sure what I did but it's fixed again. When it was not mapping I checked my attached devices page on my router's page and it was not listed so I knew it wasn't going to map. I changed it back to DHCP instead of static IP. Network drive is mapped to the IP for LAN3 now.

agreatdayeveryday, 2022-10-27:
For now, I'll close the issue and mark as completed, thanks again!!

closed 2022-10-27 by agreatdayeveryday

Payn, 2022-10-27:
DHCP is not a good idea for network drives. You will have to change settings every time IP changes. And it shouldn't even go through the router. 
Only thing needed is having PC connected to LAN3 (directly or through switch) and giving them static IPs from the same subnet, same as when connecting two PCs together. Everything else is automagic and will work after restarts and doesn't need any commands/scripts at all.

TheS1R, 2022-10-27:
Workaround: Reserve static IP on router for MAC address, and then specify DHCP on NAS for LAN3.

agreatdayeveryday, 2022-10-28:
@TheS1R I can reserve the IP address in the router for LAN3 so that should work even if I use DHCP in the NAS settings.

agreatdayeveryday, 2022-10-28:
@Payn I reserved IP addresses for my main PC, NAS LAN1 and LAN3 in the router admin so even if I use DHCP it should be fine.

Payn, 2022-10-28:
Oh so your router is also 2.5Gbit? Thought it's just a 1Gbit, that's why i suggested connecting LAN3 to PC without router so you get the full speed. Well, in that case you should be fine with reserved IPs. Hope it stays working now :)

agreatdayeveryday, 2022-10-28:
Nope, my router is not 2.5G. The WAN port is 2.5G, but the rest are just gigabit. I'd still want to access the NAS when I'm remote (i.e. traveling). Thank you for all the help, appreciate it! Now onto the next project, configuring Plex :D

Sjekke, 2022-10-28:
> 

Do you a 2,5gbit Internetconnection?

agreatdayeveryday, 2022-10-29:
Nope @Sjekke , but I have gigablast fiber (1,000 down, 1,000 up).

Sjekke, 2022-10-29:
> Nope @Sjekke , but I have gigablast fiber (1,000 down, 1,000 up).

Nice! 

Can't you reconfigure your current 2.5 Gbit port for LAN instead of the Internet and a 1 Gbit port for your WAN? Then you can connect your NAS to the 2.5 Gbit. I can do this with my ASUS router

Payn, 2022-10-29:
That's what i was suggesting from the beginning. Typical layout:
[image]

This way you access your NAS both through router (1gbit) for remote connections or directly through adapter for full speed 2.5gbit transfers. Got it set up same way except without the switch.

agreatdayeveryday, 2022-10-29:
> > Nope @Sjekke , but I have gigablast fiber (1,000 down, 1,000 up).
> 
> Nice!
> 
> Can't you reconfigure your current 2.5 Gbit port for LAN instead of the Internet and a 1 Gbit port for your WAN? Then you can connect your NAS to the 2.5 Gbit. I can do this with my ASUS router

Unfortunately my router (Netgear Orbi) doesn't have any 2.5G LAN ports, the satellite has one but it's located downstairs in my living room.

agreatdayeveryday, 2022-10-29:
> That's what i was suggesting from the beginning. Typical layout: [image]
> 
> This way you access your NAS both through router (1gbit) for remote connections or directly through adapter for full speed 2.5gbit transfers. Got it set up same way except without the switch.

I don't have any other spare ports on my router which is why I also connected the LAN1 to the 2.5G switch. I'd have to figure out which devices I can unplug from my router in order to make that happen but even if I connect the LAN1 through the switch I should still be able to access my NAS remotely, no?

Payn, 2022-10-29:
Not sure how routing would work in that case. Because you have both NAS's ports connected to router technically, receiving IPs from it. So even if you access your NAS through LAN3 it will still got through the router limiting it to 1Gbit. Maybe it would bypass router and go directly through switch to PC due to some networking magic?
Only way i know this would work is to use only your 1gbit port on PC to connect to switch and the 2.5gbit port to connect to NAS directly (my setup).

Payn, 2022-10-29:
This might work
[image]

Sjekke, 2022-10-29:
What type of router do you have. 

I would never implement a solution like this. Secondly putting your NAS on the internet ... you really must be crazy for this.

Payn, 2022-10-29:
Public IP may not be a good idea. He can use QuickConnect for safer remote access or any other solution. There is nothing wrong on being able to access your NAS when not home.
Otherwise what's the problem with this solution? How else do you want to get both 2.5gbit speed to NAS and having it available to other devices on the network?

Sjekke, 2022-10-29:
Simple ... 

[image]

Payn, 2022-10-29:
Except now you share the connection with internet and won't get full speed from NAS in case of big downloads/etc. Not saying it will happen often but why wasting the bandwidth if both his PC and NAS already have multiple interfaces?

agreatdayeveryday, 2022-10-29:
@Sjekke I have a Netgear Orbi Pro, model SXK80

agreatdayeveryday, 2022-10-29:
> This might work. Grey is 1gbit, blue 2.5gbit. [image]
> 
> Then you just map 192.168.2.1 on PC as your network share to get full speed. And for accessing remotely just use your public_IP:5000.

I'll give this a try later today. This is almost the setup I have but currently just the one cable from PC to the 2.5G switch. I have the network shared mapped to the 2.5G address.

agreatdayeveryday, 2022-10-29:
> What type of router do you have.
> 
> I would never implement a solution like this. Secondly putting your NAS on the internet ... you really must be crazy for this.

I have a default firewall rule for all ports, all source IP to deny. So unless they can VPN in they won't be able to access anything is my understanding.

Sjekke, 2022-10-29:
> Except now you share the connection with internet and won't get full speed from NAS in case of big downloads/etc. Not saying it will happen often but why wasting the bandwidth if both his PC and NAS already have multiple interfaces?

- Sharing the connection with Internet? Explain ..
- In your picture he needs a managed L3 switch €€
- And it seems you don't bother about security

Payn, 2022-10-29:
You have only single 2.5Gbit connection to PC, how do you expect to get 3.5Gbit from it (in case downloading from both internet + NAS)? I know, probably edge case but still.
And i don't bother about security? :D I don't have my NAS connected to internet at all, @agreatdayeveryday  wanted remote connection which he can use QuickConnect for like I said.
But you are probably right about unmanaged switch, so it would require either skip it and connect directly or live with the above issue..

Sjekke, 2022-10-29:
> You have only single 2.5Gbit connection to PC, how do you expect to get 3.5Gbit from it (in case downloading from both internet + NAS)? I know, probably edge case but still.

**Simple** ... goto to 10Gbit in your PC if your really want be sure that you need simultaneous 3,5gbit **AND** the internet is not used by anyone else at the same moment **AND** the file you are downloading is on a fast internetserver **AND** your router can handle this speeds ... overkill4sure. 

> And i don't bother about security? :D I don't have my NAS connected to internet at all, @agreatdayeveryday wanted remote connection which he can use QuickConnect for like I said. 

**Not safe**. It's a 3rd party proxy service and there are many other points in the chain between your NAS and you outside of your network. So VPN it is = end2end security, so not Quickconnect.  

> But you are probably right about unmanaged switch, so it would require either skip it and connect directly or live with the above issue.. 

Is this really an issue? 

**Conclusion** : If you really need that speed every day ... go for a 10Gbit network. You could also install a 5gbit adapter in your current NAS and PC but I would go for 10Gbit. OR buy the new DS923+ + 10Gbit adapter ...

Sjekke, 2022-10-29:
> @Sjekke I have a Netgear Orbi Pro, model SXK80

Is your NAS and PC connected on the same ORBI device?