bb-qq/r8152 #259: Network Adapter seams not working anymore (DSM 7.1.1-42962 Update 1)

jensche, 2022-10-29:
### Description of the problem

Since DSM 7.1.1-42962 Update 1 it is no longer possible to access directly via the USB network adapter. 

The USB network adapter is cleanly recognized as a 2.5Gbe adapter, and can also be used as standard. However, it is not accessible on the internal network via IP. 

This problem only exists since the update. 

The synology is only accessible via standard Ethernet adapter. 

### Description of your products

> - the product name of your NAS model
>     - Synology DS916+
>     - DSM 7.1.1-42962 Update 1 
> - USB 10/100/1G/2.5G LAN - Realtek Semiconductor Corp. (3D Dongle)

TheS1R, 2022-10-29:
If you are using a static IP on the NAS, try:

- updating driver to 2.16.3-2 DSM7.x; or
- use DHCP (_reserve static IP on router by MAC address of Realtek dongle_).

jensche, 2022-10-29:
Thanks... worked...

closed 2022-10-29 by jensche