bb-qq/r8152 #261: DSM 7.1.1-42962U2 driver 2.16.3-2 DSM7.x fails to load automatically at startup/does not default to DHCP

varunnangia, 2022-11-04:
**Description of problem**
Under DSM 7.1.1-42962 Update 2, driver release version [2.16.3-2 DSM7.x](https://github.com/bb-qq/r8152/releases/tag/2.16.3-2), the package does not appear to startup automatically. When started manually, the package does not remember to default to DHCP. I have tried both front and rear ports.

**Description of products**
Synology DS1019+ (apollolake) with 2 x [Crucial DDR3 4GB RAM](https://smile.amazon.com/gp/product/B005LDLVAO/)
[Plugable 2.5G USBC-E2500](https://smile.amazon.com/gp/product/B084L4JL9K/), connected to rear port
[Ubiquiti Switch Flex XG](https://store.ui.com/collections/unifi-network-switching/products/unifi-flex-xg)

**Description of environment**
NAS connects to Flex XG, M1 Mac Mini with built-in 10Gbe connects to Flex XG. Flex XG is [managed](https://www.ui.com/consoles) by a [UDM-SE](https://store.ui.com/collections/unifi-network-unifi-os-consoles/products/dream-machine-se). 

**Relevant ifconfig output**
Small redactions indicated by [redacted]. ztc interface is a tunnel created by the use of [ZeroTier](https://www.zerotier.com/).
```
docker0   Link encap:Ethernet  HWaddr 02:42:F6:0A:8D:77  
          inet addr:172.17.0.1  Bcast:172.17.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth0      Link encap:Ethernet  HWaddr 00:11:32:[redacted]  
          inet addr:169.254.208.56  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:125 

eth1      Link encap:Ethernet  HWaddr 00:11:32:[redacted]  
          inet addr:169.254.102.224  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:126 base 0x6000  

eth2      Link encap:Ethernet  HWaddr 8C:AE:4C:[redacted]  
          inet addr:10.0.0.8  Bcast:10.0.0.255  Mask:255.255.255.0
          inet6 addr: fe80::8eae:4cff:fedd:4cb0/64 Scope:Link
          inet6 addr: [redacted]:600::79d/128 Scope:Global
          inet6 addr: [redacted]:600:8eae:4cff:fedd:4cb0/64 Scope:Global
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:448052788 errors:0 dropped:0 overruns:0 frame:0
          TX packets:48855053 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:667697253904 (621.8 GiB)  TX bytes:6693756101 (6.2 GiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:385624 errors:0 dropped:0 overruns:0 frame:0
          TX packets:385624 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:85200456 (81.2 MiB)  TX bytes:85200456 (81.2 MiB)

ztc25ff5v Link encap:Ethernet  HWaddr EA:BB:D4:E7:76:1D  
          inet addr:10.147.17.204  Bcast:10.147.17.255  Mask:255.255.255.0
          inet6 addr: fe80::e8bb:d4ff:fee7:761d/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:32935 errors:0 dropped:0 overruns:0 frame:0
          TX packets:37406 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:7083610 (6.7 MiB)  TX bytes:7019577 (6.6 MiB)
```
**Workaround**
I've created a task at boot-up that runs as root that toggles the package off and on, and sets the interface to DHCP:
```
synoservice --disable pkgctl-r8152
synoservice --enable pkgctl-r8152
synonet --dhcp eth2
```
This appears to work and link has been stable for a bit, including ~600GiB transferred. Not sure why this is required though and may be an issue that needs resolution.

mrjohnpoz, 2022-11-10:
Well I finally got around to shutting down the nas to put in my little velco strips to try and help with the new noisier drives - not sure if it helped any.  But anyway upon restart the bb-qq drivers started up on their own.

varunnangia, 2022-11-11:
> I have not seen this issue normally. BUT!! I have recently redone my nas and did a clean install of the 7.1.1 update 2.. And I did at one point reboot and I had to manually start the package..
> 
> I will will check next time I reboot. Thanks for the possible work around if am also seeing this problem. I don't use dhcp on that interface though. But when I started it it did come up with my static settings and everything worked.

Sorry, I'm not sure what your specific environment is, but I have continued to experience this issue at every boot-up. The workaround continues to function, so I'm not fussed about it, but because I'm short on ports on my home network at the moment (waiting for a switch to come back in stock), I don't have the option of keeping a second cable plugged in for troubleshooting.

mrjohnpoz, 2022-11-11:
My setup is a ds918+ same dsm 7.1.1 u2, but using club 3d usb..  Static IP set, and other end is plugged into pc also using another CAC-1420 usb nic.

Running the latest 2.16.3-2 bb-qq.. I rarely reboot my nas.. Other then when update the dsm software.  Recently I did an redo of it with bigger drives.  I did notice having to run the driver at one point in the process.  But when I just recently shut it down to add some velco strips to try and reduce the noise of the drives.  When I rebooted it everything came up fine.  I will for sure keep an eye on it next time I reboot.

bb-qq, 2022-11-12:
Could you paste `/var/packages/r8152/var/log/start-stop-status.log` here?

dotWee, 2022-11-12:
Just ran into the same issue. I've updated my DS920+ about 7 days ago; today i had to perform a manually triggered reboot.  

After rebooting, the package was in "manually stopped" state and missing its configuration for the USB network interface.

@bb-qq - these are my `start-stop-status.log` logs for the time windows ~1h before reboot & ~1h after reboot:

```
Sat Nov 12 10:57:13 CET 2022
+ grep -E '^(version|create_time)' /var/packages/r8152/INFO
version="2.16.3-2"
create_time="20221011-21:59:21"
+ echo status
status
+ case $1 in
+ /sbin/lsmod
+ grep r8152
r8152                 220725  0 
usbcore               200524  10 etxhci_hcd,r8152,usblp,uhci_hcd,usb_storage,ehci_hcd,ehci_pci,usbhid,xhci_hcd,xhci_pci
+ exit 0
+ echo ----------
----------
+ date
Sat Nov 12 10:57:19 CET 2022
+ grep -E '^(version|create_time)' /var/packages/r8152/INFO
version="2.16.3-2"
create_time="20221011-21:59:21"
+ echo status
status
+ case $1 in
+ /sbin/lsmod
+ grep r8152
r8152                 220725  0 
usbcore               200524  10 etxhci_hcd,r8152,usblp,uhci_hcd,usb_storage,ehci_hcd,ehci_pci,usbhid,xhci_hcd,xhci_pci
+ exit 0
+ echo ----------
----------
+ date
Sat Nov 12 10:57:55 CET 2022
+ grep -E '^(version|create_time)' /var/packages/r8152/INFO
version="2.16.3-2"
create_time="20221011-21:59:21"
+ echo status
status
+ case $1 in
+ /sbin/lsmod
+ grep r8152
r8152                 220725  0 
usbcore               200524  10 etxhci_hcd,r8152,usblp,uhci_hcd,usb_storage,ehci_hcd,ehci_pci,usbhid,xhci_hcd,xhci_pci
+ exit 0
+ echo ----------
----------
+ date
Sat Nov 12 11:57:10 CET 2022
+ grep -E '^(version|create_time)' /var/packages/r8152/INFO
version="2.16.3-2"
create_time="20221011-21:59:21"
+ echo status
status
+ case $1 in
+ /sbin/lsmod
+ grep r8152
+ echo ----------
----------
+ date
Sat Nov 12 11:57:10 CET 2022
+ grep -E '^(version|create_time)' /var/packages/r8152/INFO
version="2.16.3-2"
create_time="20221011-21:59:21"
+ echo status
status
[... 254 lines cut ...]
+ [[ -n '' ]]
+ '[' '!' -e /sys/class/net/sit0/device/driver ']'
+ continue
+ for interface_name in ${all_interface_names}
+ [[ -n '' ]]
+ '[' '!' -e /sys/class/net/syno_ovs_bonds/device/driver ']'
+ continue
+ /sbin/rmmod /volume1/@appstore/r8152/r8152/r8152.ko
+ '[' -e /volume1/@appstore/r8152/r8152/usbnet.ko ']'
+ '[' -e /volume1/@appstore/r8152/r8152/mii.ko ']'
+ exit 0
+ echo ----------
----------
+ date
Sat Nov 12 13:37:36 CET 2022
+ grep -E '^(version|create_time)' /var/packages/r8152/INFO
version="2.16.3-2"
create_time="20221011-21:59:21"
+ echo status
status
+ case $1 in
+ /sbin/lsmod
+ grep r8152
+ exit 3
```

varunnangia, 2022-11-18:
> Could you paste `/var/packages/r8152/var/log/start-stop-status.log` here?

Certainly. Would you like me to disable the workaround or keep it when sharing the file?

Please accept my apologies for the delayed response, @bb-qq — I'm not getting notifications of replies to this thread.

bb-qq, 2022-12-04:
Yes, I wanted a log of when the problem was occurring.
If you are still experiencing the same problem with the latest release, please provide the log.
https://github.com/bb-qq/r8152/releases/tag/2.16.3-3

CangShui, 2023-01-12:
I use this :
`synosystemctl  restart pkgctl-r8152
synonet --dhcp eth1`

Tranel, 2023-01-15:
Stil have the issue on my DS215+ with the latest driver. After a reboot the driver is stopped.

Chris260377, 2023-01-20:
I have the same issue with DS920+ with driver 2.16.3-3 (geminilake). After reboot (because of firmware update) the USB network port will not start automatically, status is stopped

Description of products:
Synology DS920+ with DSM 7.1.1-42962 Update 3
Plugable 2.5G ULT-WIIQ (https://www.amazon.de/gp/product/B09GBFZ9DZ/), connected to front port

Sjekke, 2023-01-20:
> I have the same issue with DS920+ with driver 2.16.3-3 (geminilake). After reboot (because of firmware update) the USB network port will not start automatically, status is stopped
>
>
>
> Description of [...]

This is not a from the brand Plugable ...

Chris260377, 2023-01-20:
> This is not a from the brand Plugable ...

I have the same issue like mentioned in the thread description. The problem is the driver, not the plug

Tranel, 2023-02-01:
The funny thing is that after the first restart it failed to load the driver automatically. Then I fixed it and now it works fine.

reykjavi, 2023-02-09:
I can confirm that the driver (or service) didn't start after reboot just once, the very first time after installation. The second time it did start automatically.

MilesTEG1, 2023-02-10:
Hello,
I've got, I think, the same problem.
This morning, I connected my USB3<->Ethernet 2,5G adaptater on a 2,5G switch.
I just check if it was truly connected in 2,5G mode, and I don't see the ETH3 interface on my DS920+. The driver's package was stopped. I had to start it mannually.
I recall installing the DSM 7.1.1-42962 Update 4 two days ago, and then the NAS had to reboot I think.

Another thing I noticed is that status of DHCP6 isn't saved : I deactivated IPV6, and after starting the driver, there is an IPv6 :
[image]
I had to activate fully IPv6, and then deactivated it again to have no more the IPv6 :
[image]

What's the best way to ensure the driver is loaded every time ?
- A programmed task executing a script at startup in which there is the command `synosystemctl  restart pkgctl-r8152`
- A planned task executing every hour the same script ?
- A program task executing every hour the start of the service :
[image]

Add, is there a way to save the settings and keep it after a restart of the driver's service ?

Romeo1984, 2023-03-08:
> synoservice --disable pkgctl-r8152
> synoservice --enable pkgctl-r8152

I have a DS720+ and my driver package only fails to start when I complete a DSM upgrade. I created a service task to disable/enable the package on boot, so we have to wait until the next DSM Update to see if this works.

Romeo1984, 2023-03-08:
> Well I finally got around to shutting down the nas to put in my little velco strips to try and help with the new noisier drives - not sure if it helped any. But anyway upon restart the bb-qq drivers started up on their own.

bb-qq, 2023-04-22:
If it did not work, I might be able to investigate if you could paste the contents of `/var/packages/r8152/var/log/start-stop-status.log` here.

bb-qq, 2023-07-16:
Is anyone else encountering this problem with 2.16.3-4 or later versions?
I added a small improvement to this version.

lucascosti, 2023-12-01:
I had this happen on my first installation of 2.17.1-1 with my DS418play and a Plugable USBC-E2500.

On the first restart after installation, the driver didn't start, and I had to start it manually via Package Center.

After that, it's been working fine after 2 more restarts and one full shutdown/start.

<details>

<summary>Here's my log from `/var/packages/r8152/var/log/start-stop-status.log`</summary>

```shell
$ cat /var/packages/r8152/var/log/start-stop-status.log
+ echo ----------
----------
+ date
Fri Dec  1 11:14:55 AEST 2023
+ grep -E '^(version|create_time)' /var/packages/r8152/INFO
version="2.17.1-1"
create_time="20230715-10:17:26"
+ echo start
start
+ case $1 in
+ '[' -e /var/packages/r8152/scripts/apply-memory-setting ']'
+ '[' -w /sys/module/usbcore/parameters/autosuspend ']'
+ echo -1
++ count_usb_interfaces
++ ls -l /sys/class/net/
++ wc -l
++ grep /usb
+ initial_count=0
+ '[' -e /volume1/@appstore/r8152/r8152/mii.ko ']'
+ '[' -e /volume1/@appstore/r8152/r8152/usbnet.ko ']'
+ /sbin/insmod /volume1/@appstore/r8152/r8152/r8152.ko
+ '[' -r /usr/lib/udev/rules.d/51-usb-r8152-net.rules ']'
++ seq 1 60
+ for i in `seq 1 60`
++ count_usb_interfaces
++ ls -l /sys/class/net/
++ grep /usb
++ wc -l
+ '[' 0 -ne 0 ']'
+ sleep 1
+ for i in `seq 1 60`
++ count_usb_interfaces
++ wc -l
++ grep /usb
++ ls -l /sys/class/net/
+ '[' 1 -ne 0 ']'
+ break
+ set_interfaces up
+ action=up
+ target_interface_name=
++ ls /sys/class/net
+ all_interface_names='docker-32865520
docker0
docker537c062
docker7dfff2a
dockerf76e93a
eth0
eth1
eth2
lo
sit0'
+ '[' -n ']'
+ '[' '!' -e /sys/class/net/ ']'
+ for interface_name in ${all_interface_names}
+ [[ -n '' ]]
+ '[' '!' -e /sys/class/net/docker-32865520/device/driver ']'
+ continue
+ for interface_name in ${all_interface_names}
[... 1693 lines cut ...]
status
+ case $1 in
+ /sbin/lsmod
+ grep r8152
r8152                 224445  0
usbcore               210297  10 etxhci_hcd,r8152,usblp,uhci_hcd,usb_storage,ehci_hcd,ehci_pci,usbhid,xhci_hcd,xhci_pci
+ exit 0
+ echo ----------
----------
+ date
Fri Dec  1 12:04:40 AEST 2023
+ grep -E '^(version|create_time)' /var/packages/r8152/INFO
version="2.17.1-1"
create_time="20230715-10:17:26"
+ echo status
status
+ case $1 in
+ grep r8152
+ /sbin/lsmod
r8152                 224445  0
usbcore               210297  10 etxhci_hcd,r8152,usblp,uhci_hcd,usb_storage,ehci_hcd,ehci_pci,usbhid,xhci_hcd,xhci_pci
+ exit 0
```

</details>

bb-qq, 2024-06-23:
It is a necessary step to start manually from the Package Center.
I added this point to the documentation.

I would like to close this issue as the initial problem seems to have been resolved.

closed 2024-06-23 by bb-qq