bb-qq/r8152 #262: Driver for Synology Armada 37xx

kastnerka, 2022-11-10:
Dear bb-qq,

Would you mind releasing driver for Synology Armada 37xx? 

There are for Armada 375 and 38x, will it be compatible to use any of them? Or could you editing it to make it work for 37xx? Thanks!

kastnerka, 2022-11-23:
Whoa, great work! Thank you bb-qq, I will install and test it shortly.

bb-qq, 2022-11-23:
Sorry, I uploaded the wrong packages, so I re-uploaded the correct ones. Please re-download the package.

bb-qq, 2022-12-04:
I close this issue since the request of creating a driver was filled. If you have any problems, please create a new issue.
Also, I appreciate it if you report the result to this issue: https://github.com/bb-qq/r8152/issues/135

closed 2022-12-04 by bb-qq