bb-qq/r8152 #263: DS716+II and ULT-unite 2.5G Ethernet Adapter (RTL8156B) driver works 100%

VAllens, 2022-11-13:
The driver is working, but devices can only connect with static IP.
DHCP devices have no access to the nas.

My environment:
Product model: [DS716+II](https://www.synology.com/en-us/support/download/DS716+II?#docs)
DSM version: [7.1.1-42962 update 2](https://www.synology.com/en-us/releaseNote/DSM?model=DS716%2BII#7_1)
Port: USB3.0 at back
ULT-unite 2.5G Ethernet Adapter (RTL8156B)
driver: [r8152-braswell-2.16.3-2.spk](https://github.com/bb-qq/r8152/releases/download/2.16.3-2/r8152-braswell-2.16.3-2.spk)

iperf3 test result

```powershell
❯ .\iperf3.exe -c 192.168.31.129 -i 1 -t 60 -p 520
Connecting to host 192.168.31.129, port 520
[  4] local 192.168.31.14 port 2427 connected to 192.168.31.129 port 520
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec   221 MBytes  1.85 Gbits/sec
[  4]   1.00-2.00   sec   233 MBytes  1.96 Gbits/sec
[  4]   2.00-3.00   sec   232 MBytes  1.94 Gbits/sec
[  4]   3.00-4.00   sec   233 MBytes  1.96 Gbits/sec
[  4]   4.00-5.00   sec   234 MBytes  1.96 Gbits/sec
[  4]   5.00-6.00   sec   232 MBytes  1.94 Gbits/sec
[  4]   6.00-7.00   sec   223 MBytes  1.87 Gbits/sec
[  4]   7.00-8.00   sec   230 MBytes  1.93 Gbits/sec
[  4]   8.00-9.00   sec   233 MBytes  1.95 Gbits/sec
[  4]   9.00-10.00  sec   234 MBytes  1.96 Gbits/sec
[  4]  10.00-11.00  sec   234 MBytes  1.96 Gbits/sec
[  4]  11.00-12.00  sec   232 MBytes  1.95 Gbits/sec
[  4]  12.00-13.00  sec   221 MBytes  1.85 Gbits/sec
[  4]  13.00-14.00  sec   233 MBytes  1.96 Gbits/sec
[  4]  14.00-15.00  sec   232 MBytes  1.95 Gbits/sec
[  4]  15.00-16.00  sec   233 MBytes  1.96 Gbits/sec
[  4]  16.00-17.00  sec   229 MBytes  1.92 Gbits/sec
[  4]  17.00-18.00  sec   232 MBytes  1.95 Gbits/sec
[  4]  18.00-19.00  sec   219 MBytes  1.84 Gbits/sec
[  4]  19.00-20.00  sec   234 MBytes  1.96 Gbits/sec
[  4]  20.00-21.00  sec   233 MBytes  1.95 Gbits/sec
[  4]  21.00-22.00  sec   234 MBytes  1.96 Gbits/sec
[  4]  22.00-23.00  sec   234 MBytes  1.96 Gbits/sec
[  4]  23.00-24.00  sec   228 MBytes  1.92 Gbits/sec
[  4]  24.00-25.00  sec   211 MBytes  1.77 Gbits/sec
[  4]  25.00-26.00  sec   234 MBytes  1.96 Gbits/sec
[  4]  26.00-27.00  sec   232 MBytes  1.95 Gbits/sec
[  4]  27.00-28.00  sec   234 MBytes  1.96 Gbits/sec
[  4]  28.00-29.00  sec   234 MBytes  1.96 Gbits/sec
[  4]  29.00-30.00  sec   233 MBytes  1.95 Gbits/sec
[  4]  30.00-31.00  sec   221 MBytes  1.85 Gbits/sec
[  4]  31.00-32.00  sec   232 MBytes  1.95 Gbits/sec
[  4]  32.00-33.00  sec   228 MBytes  1.91 Gbits/sec
[  4]  33.00-34.00  sec   229 MBytes  1.92 Gbits/sec
[  4]  34.00-35.00  sec   229 MBytes  1.92 Gbits/sec
[  4]  35.00-36.00  sec   228 MBytes  1.91 Gbits/sec
[  4]  36.00-37.00  sec   217 MBytes  1.82 Gbits/sec
[  4]  37.00-38.00  sec   232 MBytes  1.94 Gbits/sec
[  4]  38.00-39.00  sec   228 MBytes  1.91 Gbits/sec
[  4]  39.00-40.00  sec   225 MBytes  1.89 Gbits/sec
[  4]  40.00-41.00  sec   228 MBytes  1.91 Gbits/sec
[  4]  41.00-42.00  sec   227 MBytes  1.91 Gbits/sec
[  4]  42.00-43.00  sec   219 MBytes  1.84 Gbits/sec
[  4]  43.00-44.00  sec   229 MBytes  1.92 Gbits/sec
[  4]  44.00-45.00  sec   228 MBytes  1.91 Gbits/sec
[  4]  45.00-46.00  sec   229 MBytes  1.92 Gbits/sec
[  4]  46.00-47.00  sec   226 MBytes  1.89 Gbits/sec
[  4]  47.00-48.00  sec   227 MBytes  1.90 Gbits/sec
[  4]  48.00-49.00  sec   220 MBytes  1.84 Gbits/sec
[  4]  49.00-50.00  sec   230 MBytes  1.93 Gbits/sec
[  4]  50.00-51.00  sec   227 MBytes  1.90 Gbits/sec
[  4]  51.00-52.00  sec   227 MBytes  1.91 Gbits/sec
[  4]  52.00-53.00  sec   227 MBytes  1.90 Gbits/sec
[  4]  53.00-54.00  sec   227 MBytes  1.91 Gbits/sec
[  4]  54.00-55.00  sec   216 MBytes  1.81 Gbits/sec
[  4]  55.00-56.00  sec   224 MBytes  1.88 Gbits/sec
[  4]  56.00-57.00  sec   229 MBytes  1.92 Gbits/sec
[  4]  57.00-58.00  sec   227 MBytes  1.91 Gbits/sec
[  4]  58.00-59.00  sec   227 MBytes  1.91 Gbits/sec
[  4]  59.00-60.00  sec   226 MBytes  1.90 Gbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-60.00  sec  13.4 GBytes  1.92 Gbits/sec                  sender
[  4]   0.00-60.00  sec  13.4 GBytes  1.92 Gbits/sec                  receiver

iperf Done.
```

VAllens, 2022-11-13:
`ULT-unite USB 3.0 to 2.5G Ethernet Adapter (RTL8156B)` may be the same as `ULT-WIIQ USB 3.0 to 2.5G Ethernet Adapter (RTL8156b)`

bb-qq, 2022-11-19:
Could you paste `/var/packages/r8152/var/log/start-stop-status.log` here when you configure DHCP?

VAllens, 2022-11-21:
[start-stop-status.log](https://github.com/bb-qq/r8152/files/10057355/start-stop-status.log)

bb-qq, 2022-12-04:
Could you try this test version?
https://github.com/bb-qq/r8152/releases/download/2.16.3-3/r8152-braswell-2.16.3-4test.spk

VAllens, 2022-12-04:
> Could you try this test version? https://github.com/bb-qq/r8152/releases/download/2.16.3-3/r8152-braswell-2.16.3-4test.spk

Very good !!! Thanks :)
DHCP service is worked.

[start-stop-status.log](https://github.com/bb-qq/r8152/files/10148193/start-stop-status.log)