bb-qq/r8152 #265: DS218+, DSM7.1.1.42962, DriverGenius CUG025 KO

patator100, 2022-11-18:
Hello

I have tried to install my compatible drivergenius adapter - no success. 
I have followed correctly all instructions (including the script in ssh mode after installation) - no success:
`sudo install -m 4755 -o root -D /var/packages/r8152/target/r8152/spk_su /opt/sbin/spk_su
`
The package does not want to launch and only show a button repair :
[image]

[image]

I have relaunched the installation several time (after uninstalling, reinstalling...) no difference.

I've tried to unplug/plug the adapter, no difference.
It is correctly seen by the NAS:
[image]

I have also restarted the NAS twice, no difference

Any idea what I'm doing wrong
Thanks in advance
pat

patator100, 2022-11-19:
After uninstalling, reboot and reinstalling (r8152-apollolake-2.16.3-3test.spk), still the same. Here is the log

`e)
2022/11/19 01:07:34 (system) trigger SynologyDrive 3.2.0-23232 Begin start-stop-                                                                             status start
2022/11/19 01:07:39 (system) trigger SynologyPhotos 1.3.0-0317 End start-stop-st                                                                             atus start ret=[0]
2022/11/19 01:07:54 (system) trigger SynologyDrive 3.2.0-23232 End start-stop-st                                                                             atus start ret=[0]
2022/11/19 01:07:56 (system) trigger Docker 20.10.3-1308 End start-stop-status s                                                                             tart ret=[0]
2022/11/19 01:07:56 (system) trigger Virtualization 2.6.1-12139 End start-stop-s                                                                             tatus start ret=[0]
2022/11/19 01:08:04 Use given volume to install, volume=[]
2022/11/19 01:08:04 Use any available volume to install, volume=[/volume1]
2022/11/19 01:08:19 Use given volume to install, volume=[]
2022/11/19 01:08:19 Use any available volume to install, volume=[/volume1]
2022/11/19 01:08:21 Use given volume to install, volume=[]
2022/11/19 01:08:21 Use any available volume to install, volume=[/volume1]
2022/11/19 01:08:21 install r8152 2.16.3-3 Begin preinst
2022/11/19 01:08:21 install r8152 2.16.3-3 End preinst ret=[0]
2022/11/19 01:08:23 install r8152 2.16.3-3 Begin /bin/mkdir -p /volume1/@appstor                                                                             e/r8152 -m 755
2022/11/19 01:08:23 install r8152 2.16.3-3 End /bin/mkdir -p /volume1/@appstore/                                                                             r8152 -m 755 ret=[0]
2022/11/19 01:08:23 install r8152 2.16.3-3 Begin /bin/rm -rf /volume1/@appstore/                                                                             r8152
2022/11/19 01:08:23 install r8152 2.16.3-3 End /bin/rm -rf /volume1/@appstore/r8                                                                             152 ret=[0]
2022/11/19 01:08:23 install r8152 2.16.3-3 Begin /bin/mv -f /volume1/@tmp/synopk                                                                             g/install.SMq6a6/package /volume1/@appstore/r8152
2022/11/19 01:08:23 install r8152 2.16.3-3 End /bin/mv -f /volume1/@tmp/synopkg/                                                                             install.SMq6a6/package /volume1/@appstore/r8152 ret=[0]
2022/11/19 01:08:23 install r8152 2.16.3-3 Begin /bin/rm -rf /var/packages/r8152
2022/11/19 01:08:23 install r8152 2.16.3-3 End /bin/rm -rf /var/packages/r8152 r                                                                             et=[0]
2022/11/19 01:08:23 install r8152 2.16.3-3 Begin /bin/mkdir -p /var/packages/r81                                                                             52 -m 755
2022/11/19 01:08:23 install r8152 2.16.3-3 End /bin/mkdir -p /var/packages/r8152                                                                              -m 755 ret=[0]
2022/11/19 01:08:23 install r8152 2.16.3-3 Begin /bin/touch /var/packages/r8152/                                                                             installing
2022/11/19 01:08:23 install r8152 2.16.3-3 End /bin/touch /var/packages/r8152/in                                                                             stalling ret=[0]
2022/11/19 01:08:23 install r8152 2.16.3-3 Begin /bin/mv -f /volume1/@tmp/synopk                                                                             g/install.SMq6a6/INFO /var/packages/r8152/INFO
2022/11/19 01:08:23 install r8152 2.16.3-3 End /bin/mv -f /volume1/@tmp/synopkg/                                                                             install.SMq6a6/INFO /var/packages/r8152/INFO ret=[0]
2022/11/19 01:08:23 install r8152 2.16.3-3 Begin /bin/rm -rf /var/packages/r8152                                                                             /scripts
2022/11/19 01:08:23 install r8152 2.16.3-3 End /bin/rm -rf /var/packages/r8152/s                                                                             cripts ret=[0]
2022/11/19 01:08:23 install r8152 2.16.3-3 Begin /bin/mv -f /volume1/@tmp/synopk                                                                             g/install.SMq6a6/scripts /var/packages/r8152/scripts
2022/11/19 01:08:24 install r8152 2.16.3-3 End /bin/mv -f /volume1/@tmp/synopkg/                                                                             install.SMq6a6/scripts /var/packages/r8152/scripts ret=[0]
2022/11/19 01:08:24 install r8152 2.16.3-3 Begin /bin/rm -rf /var/packages/r8152                                                                             /conf
2022/11/19 01:08:24 install r8152 2.16.3-3 End /bin/rm -rf /var/packages/r8152/c                                                                             onf ret=[0]
2022/11/19 01:08:24 install r8152 2.16.3-3 Begin /bin/mv -f /volume1/@tmp/synopk                                                                             g/install.SMq6a6/conf /var/packages/r8152/conf
2022/11/19 01:08:24 install r8152 2.16.3-3 End /bin/mv -f /volume1/@tmp/synopkg/                                                                             install.SMq6a6/conf /var/packages/r8152/conf ret=[0]
[... cut ...]
root@NAS_SALEM:/var/log# 2022/11/19 01:09:30 upgrade r8152: start version 2.16.3-3 failed, result 0
-ash: 2022/11/19: No such file or directory
root@NAS_SALEM:/var/log# 2022/11/19 01:09:30 upgrade r8152 from 2.16.3-3 to 2.16.3-3 successfully
-ash: 2022/11/19: No such file or directory
root@NAS_SALEM:/var/log# 2022/11/19 01:09:31 (system) trigger r8152 2.16.3-3 Begin start-stop-status stop
-ash: syntax error near unexpected token `('
root@NAS_SALEM:/var/log# 2022/11/19 01:09:31 (system) trigger r8152 2.16.3-3 End start-stop-status stop r  ^C
root@NAS_SALEM:/var/log#
`

patator100, 2022-11-19:
To help a bit more and going through previous advises, I have run the following commands:
sudo /var/packages/r8152/scripts/start-stop-status stop
[stop result.txt](https://github.com/bb-qq/r8152/files/10046751/stop.result.txt)

sudo /var/packages/r8152/scripts/start-stop-status start
[start result.txt](https://github.com/bb-qq/r8152/files/10046750/start.result.txt)

bb-qq, 2022-11-19:
Could you try this version?
https://github.com/bb-qq/r8152/releases/download/2.16.3-2/r8152-apollolake-2.16.3-3test2.spk

After installing this driver, you may need to configure the IP settings again.
If the driver still does not work properly, please let us know again content of the log and the result of the following command.
```
grep . /etc/sysconfig/network-scripts/*
```

patator100, 2022-11-19:
wauouhhhh ! great; it works 👍 THANKS A LOT
[image]

[image]

with a simple CAT5e cable and a 2.5Go adaptaters (no ssd in my NAS), I have managed to download a file at a rate of 200mb/s...impressive.

I now have another issue as my virtual machine (hosting a jeedom docker) doesn't have any network anymore. Should I open another thread or do we carry on here ?

thanks again for your help and great contribution to the community
pat

bb-qq, 2022-11-20:
> Should I open another thread or do we carry on here ?

Yes, please.

closed 2022-11-20 by patator100