bb-qq/r8152 #266: DS218play works OK!

andrus2049, 2022-11-20:
After the success with my DS218 I tested also my older DS218play, which differs only for the RAM that in the DS218play is only 1 GB.

Synology DS218play
Realtek RTD1296 SoC
DSM 7.1.1-42962 Update 2

USB model:
Digitus USB Type-C™ Gigabit Ethernet Adapter 2.5G, USB-C™ + USB A (USB3.1/3.0)
https://www.digitec.ch/de/s1/product/digitus-usb-type-c-gigabit-ethernet-adapter-25g-usb-c-usb-a-usb3130-usb-c-usb-31-netzwerkadapter-16185124

USB 3 port used: on the back
Working reliably, no interruptions during transfer of large GB files.
Speed to the DS218 from a NUC Intel with 2.5G Ethernet Adapter: about 140-150 MBs.

Details:

With driver: 2.16.3-2 DSM7.x
[r8152-rtd1296-2.16.3-2.spk](https://github.com/bb-qq/r8152/releases/download/2.16.3-2/r8152-rtd1296-2.16.3-2.spk)
Assigned static IP in GUI is correctly working after reboot.

**Many thanks to [bb-qq](https://github.com/bb-qq)!!!**

[image]

cqwangding, 2022-11-22:
Similar issue on my ds220j, copy to Nas work well with 200+MB speed, copy from Nas disconnected immediately.

bb-qq, 2022-12-03:
There have been many other similar reports on the RTD1296 platform, most likely due to a problem with the USB host controller, for which there is no known workaround at this time.

bb-qq, 2022-12-04:
Topics on rtd1296 are merged into issue #275.
If you notice anything new, please comment there.

closed 2022-12-04 by bb-qq