bb-qq/r8152 #267: DS218+, DSM7.1.1.42962, r8152-apollolake-2.16.3-3test2.spk and Virtual Machine

patator100, 2022-11-20:
Hello

I'm currently using with success a DriverGenius 2.5Go adapter. Nevertheless, I can't manage to add the adapter to my virtual switch so that I could use the 2.5Go connection for my jeedom virtual machine.
I can correctly see both connections in my network:
[image]

In the virtual switch, I can only see the Lan1 connection:
[image]

[image]

Any idea how to fix this issue ?
Which log would you need ?
thanks

mrjohnpoz, 2022-11-20:
Odd, I see it available on mine.

[image]

I am on 7.1.1, but update 2

patator100, 2022-11-20:
that's weird. It might depends on the 2.5Go adapter driver. 
Have you rebooted your NAS to get the new LAN adapter to appear ?

mrjohnpoz, 2022-11-20:
I am using the same r8152-apollolake-2.16.3-2.spk driver - I think he released one without test in the name day later or something.  I have never noticed before if I could add that to my vm switch.. Since for me it is direct connection just to my pc - but upon you saying your having a problem I wanted to see if seeing the same thing.  Which I am not, it lists it there to add to new switch in VM network settings.

And yeah is way quicker than 1 gig ;)  I see 2.37gpbs via iperf testing to and from my nas.

patator100, 2022-11-26:
ouff ! managed to get it working. After rebooting, the magic happened !
[image]

thank for your help !
pat

closed 2022-11-26 by patator100