bb-qq/r8152 #268: Version 2.16.3-3 require root access in package manager

redhook62, 2022-11-23:
### Version 2.16.3-3 require root access in package manager

So , how to deploy the spk file with ssh ?

### Description of your products

> - Synology DS420+
>     Linux DiskStation 4.4.180+ #42962 SMP Tue Oct 18 15:07:03 CST 2022 x86_64 GNU/Linux synology_geminilake_420+
>     DSM 7.1.1-42962 Update 2

myssv, 2022-11-23:
I have the same

German message:
[image]

bb-qq, 2022-11-23:
Sorry, I uploaded the wrong packages, so I re-uploaded the correct ones. Please re-download the package.

redhook62, 2022-11-23:
Great !

Installed without any problems.

best regards

closed 2022-11-23 by redhook62