bb-qq/r8152 #269: 2.16.3-3 DSM7.x : unable to install because it runs with root privileges

andrus2049, 2022-11-23:
Attempt to install the new release 2.16.3-3 DSM7.x on my DS218 (DSM 7.1.1-42962 Update 2) which had already installed the previous driver 2.16.3-2 DSM7.x fails with the message above.

Used driver:
[r8152-rtd1296-2.16.3-3.spk](https://github.com/bb-qq/r8152/releases/download/2.16.3-3/r8152-rtd1296-2.16.3-3.spk)

To be sure I also reissued the command from the SSH terminal.

Just to verify the difference between the 2 drivers, I uninstalled the previous version 2.16.3-2 DSM7.x and reinstalled it: the previuos driver was successfully installed without the message, so something is different in the new release.

Thanks bb-qq !

bb-qq, 2022-11-23:
Sorry, I uploaded the wrong packages, so I re-uploaded the correct ones. Please re-download the package.

bb-qq, 2022-12-03:
This Issue is closed as the installation issue has been resolved.

closed 2022-12-03 by bb-qq