bb-qq/r8152 #271: I can't get 2.5G Speed

dateno1, 2022-11-26:
### Description of the problem

When i using 6.2.4&Driver for 6.2.4 All OK (I can get 270~285M/s by FTP or SMB)

but I can't get full speed with 7.1.1 (DSM 6.2.4 -> 7.1.1 Upgraded)

Currenttly speed is 115~128M/s (It slower then 2.5" 5400RPM*4@RAID5 before I was replaced Disk by SSD)

I was tested it on 2 client but same result

Now What i need to do?

### Description of your products

> - the product name of your NAS model
>     - and output of ``uname -a`` command
>     - DSM version x.x
> - the product name of the ethernet dongle
DSM 7.1.1-42962
DS918+ (xpenology/Running by ARPL)
i5-6500T/8G/4T SATA SSD*4@RAID5

uname -a
Linux NAS 4.4.180+ #42962 SMP Sat Sep 3 22:30:29 CST 2022 x86_64 GNU/Linux synology_apollolake_918+

### Description of your environment

> - connect directly to a PC or using a Ethernet switch
>     - PC: NIC model name or chipset name
>     - Ethernet switch: model name
> - cable category (Cat.5e or Cat.6 or Cat.7)
RTL8125 2.5G USB NIC (PC/NAS Both side using same model)
Using QNAP 2.5G 5P Switch
Using Cat.6 Cable (Already replaced cable but result is same)

Trim enabled (1 time per day)

Manually run by
fstrim -v /
but Nothing changed

Client is Windows 10 x64/Windows 10 x64@ESXi
### Output of ``dmesg`` command
[dmesg.txt](https://github.com/bb-qq/r8152/files/10095428/dmesg.txt)

### Output of ``lsusb`` command

Direct Connected on Device
lsusb
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:14.0) hub
  |__1-7         0bda:0129:3960 ff  2.00  480MBit/s 500mA 1IF  (Generic USB2.0-CRW 20100201396000000)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:14.0) hub
  |__2-1         f400:f400:0100 00  3.00 5000MBit/s 896mA 1IF  (SanDisk Ultra Fit 4C530001071029109505)
  |__2-3         0bda:8156:3000 00  3.20 5000MBit/s 512mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 000000001)

### Output of ``ifconfig -a`` command
ifconfig -a
docker0   Link encap:Ethernet  HWaddr 02:42:CF:62:C2:8E
          inet addr:172.17.0.1  Bcast:172.17.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth0      Link encap:Ethernet  HWaddr 9C:5C:8E:20:F5:25
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:43173 errors:0 dropped:0 overruns:0 frame:0
          TX packets:12525 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:4909039 (4.6 MiB)  TX bytes:10908409 (10.4 MiB)
          Interrupt:126 base 0xc000

eth1      Link encap:Ethernet  HWaddr A0:CE:C8:C8:14:5F
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:2375269 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1403161 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:3180266972 (2.9 GiB)  TX bytes:1603646223 (1.4 GiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:7637 errors:0 dropped:0 overruns:0 frame:0
          TX packets:7637 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:8452210 (8.0 MiB)  TX bytes:8452210 (8.0 MiB)

ovs-syste Link encap:Ethernet  HWaddr 4E:E3:10:B8:1F:79
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

ovs_eth0  Link encap:Ethernet  HWaddr 9C:5C:8E:20:F5:25
          inet addr:192.168.1.129  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: 2001:470:24:2ff:9e5c:8eff:fe20:f525/64 Scope:Global
          inet6 addr: fe80::9e5c:8eff:fe20:f525/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:39818 errors:0 dropped:0 overruns:0 frame:0
          TX packets:9525 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:4093516 (3.9 MiB)  TX bytes:10876649 (10.3 MiB)

ovs_eth1  Link encap:Ethernet  HWaddr A0:CE:C8:C8:14:5F
          inet addr:192.168.11.1  Bcast:192.168.11.255  Mask:255.255.255.0
          inet6 addr: fe80::a2ce:c8ff:fec8:145f/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:494542 errors:0 dropped:0 overruns:0 frame:0
          TX packets:388438 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:3105037892 (2.8 GiB)  TX bytes:1550761485 (1.4 GiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

tap021132 Link encap:Ethernet  HWaddr 52:18:A5:BA:A0:9B
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:513 errors:0 dropped:0 overruns:0 frame:0
          TX packets:29260 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:31760 (31.0 KiB)  TX bytes:2508955 (2.3 MiB)

I used 2 NIC (RTL8111 + RTL8125)
ovs_eth1/eth1 is RTL 8125

bb-qq, 2022-11-26:
* Is the network configuration exactly the same as before the upgrade?
  * Topology, link speeds with the switch, IP settings, MTU values, etc.
* Is the speed measured using iperf with the same results?
* Could you try increasing the MTUs to see if that changes performance?

mrjohnpoz, 2022-11-26:
I for sure would take disks and smb out of the equation first and test with just iperf.  You can install the syno package

https://synocommunity.com/package/synocli-monitor

Which includes iperf, I ran into slow issue awhile back when trying to upgrade the driver for the usb nic on my windows pc - I grabbed the driver directly from realtek and this corrected the problem.

I'm just using the standard 1500 mtu, and I see 2.37gbps in both directions with iperf.. And file copies in both directions can hit 280MBps, but normally run in 260-270ish range.. 

Here are 2 tests, one in each direction one without -R and one with. 

`ash-4.4# iperf3 -s
-----------------------------------------------------------
Server listening on 5201 (test #1)
-----------------------------------------------------------
Accepted connection from 192.168.10.9, port 2951
[  5] local 192.168.10.10 port 5201 connected to 192.168.10.9 port 2952
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec   283 MBytes  2.37 Gbits/sec                  
[  5]   1.00-2.00   sec   283 MBytes  2.37 Gbits/sec                  
[  5]   2.00-3.00   sec   283 MBytes  2.37 Gbits/sec                  
[  5]   3.00-4.00   sec   283 MBytes  2.37 Gbits/sec                  
[  5]   4.00-5.00   sec   283 MBytes  2.37 Gbits/sec                  
[  5]   5.00-6.00   sec   283 MBytes  2.37 Gbits/sec                  
[  5]   6.00-7.00   sec   283 MBytes  2.37 Gbits/sec                  
[  5]   7.00-8.00   sec   283 MBytes  2.37 Gbits/sec                  
[  5]   8.00-9.00   sec   283 MBytes  2.37 Gbits/sec                  
[  5]   9.00-10.00  sec   283 MBytes  2.37 Gbits/sec                  
[  5]  10.00-10.01  sec  3.09 MBytes  2.37 Gbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-10.01  sec  2.77 GBytes  2.37 Gbits/sec                  receiver
-----------------------------------------------------------
Server listening on 5201 (test #2)
-----------------------------------------------------------
Accepted connection from 192.168.10.9, port 2953
[  5] local 192.168.10.10 port 5201 connected to 192.168.10.9 port 2954
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   282 MBytes  2.36 Gbits/sec   85    500 KBytes       
[  5]   1.00-2.00   sec   282 MBytes  2.37 Gbits/sec    2    500 KBytes       
[  5]   2.00-3.00   sec   282 MBytes  2.36 Gbits/sec   22    416 KBytes       
[  5]   3.00-4.00   sec   283 MBytes  2.38 Gbits/sec    0    502 KBytes       
[  5]   4.00-5.00   sec   282 MBytes  2.37 Gbits/sec    0    510 KBytes       
[  5]   5.00-6.00   sec   283 MBytes  2.37 Gbits/sec    0    522 KBytes       
[  5]   6.00-7.00   sec   283 MBytes  2.37 Gbits/sec    0    522 KBytes       
[  5]   7.00-8.00   sec   281 MBytes  2.36 Gbits/sec   15    499 KBytes       
[  5]   8.00-9.00   sec   283 MBytes  2.37 Gbits/sec    0    509 KBytes       
[  5]   9.00-10.00  sec   281 MBytes  2.36 Gbits/sec   19    496 KBytes       
[  5]  10.00-10.00  sec   816 KBytes  8.58 Gbits/sec    0    496 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  2.76 GBytes  2.37 Gbits/sec  143             sender
-----------------------------------------------------------
Server listening on 5201 (test #3)
-----------------------------------------------------------
`

dateno1, 2022-11-27:
> 

Currently MTU is 1500 and Not changed

I can't use jumbo packet (Some of client not support it)

dateno1, 2022-11-27:
Sorry i found what is problem

It from Driver for windows :( (Linux client is OK)

closed 2022-11-27 by dateno1