bb-qq/r8152 #274: R1000 platform possible?

decay71, 2022-12-02:
Hello, 

is there any chance to compile this for R1000 platform? I i can do it myself can you point me somewhere on how to do that?

Want to try it on a 923+

Thanks

Dirk

bb-qq, 2022-12-03:
Could you try the test release?
https://github.com/bb-qq/r8152/releases/download/2.16.3-3/r8152-r1000-2.16.3-3.spk

decay71, 2022-12-03:
Thank you, it says invalid file format

[image]

bb-qq, 2022-12-03:
Could you retry this?
https://github.com/bb-qq/r8152/releases/download/2.16.3-3/r8152-r1000-2.16.3-4test.spk

decay71, 2022-12-03:
Great, works so far, BackPort, DHCP.. I will send detailed test results later!

decay71, 2022-12-03:
Ok, i had NAS-PC Speed 74gb file 170mb/sec but  PC-NAS started with 200 but slowed down to 40 when i stopped halfway. I had to use a direct connection, i will do more tests when my switch arrives (hopefully today.. or monday) an post it in the compability thread then. Thanks a lot!

armaan-novus, 2022-12-22:
Jumping into the thread to track progress. I have a DS1522+ (same R1000 platform) that I'm going to be testing the [r1000-2.16.3-4test.spk](https://github.com/bb-qq/r8152/issues/274#issuecomment-1336123355) on shortly also.

decay71, 2022-12-22:
Hi, i posted it in the compability thread as mentioned.. https://github.com/bb-qq/r8152/issues/135

Rock stable since weeks now!

bb-qq, 2022-12-31:
Thanks for the report. I close this Issue as it seems to be working fine.

closed 2022-12-31 by bb-qq