bb-qq/r8152 #276: the usual lecture from the local System Administrator

davidhoppcpa, 2022-12-18:
Hi folks, I am a complete noob.  After running the command in SSH terminal, I get the "We trust you have received the usual lecture from the local System Administrator..." message; and it asks for a "Password:."  Which password do I enter here?  Thanks for your time.  (I apologize if this is the wrong place for this "issue."  Thank you again.)

armaan-novus, 2022-12-22:
Hi @davidhoppcpa ,

That message is a relic from a bygone era. `sudo` is a program you run from a normal user account to elevate your privileges to that of an administrator (the "super user" or `su`). Presumably, it's necessary at the beginning of that command because `install` requires admin privileges to run. It used to be the case that Unix-like systems were mostly servers with many users connected to them at once. Running `sudo` on such systems would give you system-wide access, granting you the ability to overwrite other users' files (whether by accident or not). 

Anyway, the password prompt there is for the same user password that you used to ssh into the NAS to begin with. Certain commands like `sudo` just want your password again.

armaan-novus, 2022-12-22:
By the way, it looks like you don't have your User Home Directories enabled. It's a small matter, but it might be worth enabling them if you're going to SSH into the NAS frequently. To do so, follow the steps under the `User Home` heading here: https://kb.synology.com/en-af/DSM/help/DSM/AdminCenter/file_user_advanced?version=7

bb-qq, 2022-12-31:
@davidhoppcpa, have your questions been answered? If so, I would like to close the issue.

closed 2023-07-16 by bb-qq