bb-qq/r8152 #277: 2.5G adapter defaulting to 1G

rtadams89, 2022-12-30:
**Description of the problem**

The network interface falls back to 1000Mbit speed. I can manually set the speed to 2500 via `ethtool -s eth2 speed 2500 duplex full autoneg on` but after a reboot the adapter defaults to 1000 again.

**Description of your products**

DS918+ running DSM 7.1.1
rtl8156b based USB adapter
QNAP QSW-1105-5T 5-Port Unmanaged 2.5GbE Switch

**Description of your environment**

rtl8156b based USB adapter connected to DS918+. CAT 6 cable connecting adapter to network switch.

**Output of lsusb command**

> |__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-1         1d5c:5011:0100 09  2.10  480MBit/s 0mA 1IF  (Fresco Logic, Inc. USB2.0 Hub ffffffd6ffffffa3ffffffebffffffcb) hub
    |__1-1.3     051d:0002:0090 00  2.00   12MBit/s 2mA 1IF  (American Power Conversion Back-UPS XS 1500M FW:947.d10 .D USB FW:d10      3B2113X67220  )
    |__1-1.4     0bda:8156:3100 00  2.10  480MBit/s 200mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 00150067A)
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F00949B163A7D54)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-1         1058:2620:1018 00  3.10 5000MBit/s 896mA 1IF  (Western Digital Elements 2620 57583732443230445048464A)

**Output of ifconfig -a command**

> eth2      Link encap:Ethernet  HWaddr 44:A9:2C:55:06:7A
          inet addr:192.168.0.4  Bcast:192.168.0.255  Mask:255.255.255.0
          inet6 addr: fe80::46a9:2cff:fe55:67a/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:9000  Metric:1
          RX packets:204 errors:0 dropped:0 overruns:0 frame:0
          TX packets:150 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:9384 (9.1 KiB)  TX bytes:25762 (25.1 KiB)

**Output of ethtool command (after reboot)**

> Settings for eth2:
        Supported ports: [ MII ]
        Supported link modes:   10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
                                2500baseX/Full
        Supported pause frame use: No
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
        Advertised pause frame use: Symmetric Receive-only
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Link partner advertised link modes:  10baseT/Half 10baseT/Full
                                             100baseT/Half 100baseT/Full
                                             1000baseT/Half 1000baseT/Full
        Link partner advertised pause frame use: No
        Link partner advertised auto-negotiation: Yes
        Link partner advertised FEC modes: Not reported
        Speed: 1000Mb/s
        Duplex: Full
        Port: MII
        PHYAD: 32
        Transceiver: internal
        Auto-negotiation: on
        Supports Wake-on: pumbg
        Wake-on: g
        Current message level: 0x00007fff (32767)
                               drv probe link timer ifdown ifup rx_err tx_err tx_queued intr tx_done rx_status pktdata hw wol
        Link detected: yes

**Output of ethtool command (after manually setting speed to 2500)**

> Settings for eth2:
        Supported ports: [ MII ]
        Supported link modes:   10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
                                2500baseX/Full
        Supported pause frame use: No
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  2500baseX/Full
        Advertised pause frame use: Symmetric Receive-only
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Speed: Unknown!
        Duplex: Unknown! (255)
        Port: MII
        PHYAD: 32
        Transceiver: internal
        Auto-negotiation: on
        Supports Wake-on: pumbg
        Wake-on: g
        Current message level: 0x00007fff (32767)
                               drv probe link timer ifdown ifup rx_err tx_err tx_queued intr tx_done rx_status pktdata hw wol
        Link detected: no

bb-qq, 2022-12-31:
Can you try a direct connection to another switch or PC?

rtadams89, 2022-12-31:
I have attempted connecting it to another device with a 2.5G port. The connection still only establishes at 1000Mb/s. ethtool output shows the Synology is only advertising up to 1000Mb/s link modes, so it makes sense that the connection would auto negotiate to that. I also just noticed seems the device is recognized as supporting "2500**baseX**/Full" which is odd as "basex" would imply the use of fiberoptic for the physical layer, which is not present here. Is this perhaps the issue; the driver doesn't know 2500base**t**/full is supported, and therefore doesn't advertise it on a twisted pair port?

rtadams89, 2022-12-31:
Taking a shot in the dark here, but... Line 79 of https://github.com/bb-qq/r8152/blob/master/compatibility.h
`#define ETHTOOL_LINK_MODE_2500baseT_Full_BIT	ETHTOOL_LINK_MODE_2500baseX_Full_BIT`

Is that right?

bb-qq, 2022-12-31:
Yes, old kernels do not support 2500baseT, so the link mode is disguised as 2500baseX for ethtool in Realtek drivers.

The actual negotiation is completed in hardware, and the chip only has an interface that specifies a link rate such as 2500. So 2500baseT is used actually.

> I can manually set the speed to 2500 via ethtool -s eth2 speed 2500 duplex full autoneg on but after a reboot the adapter defaults to 1000 again.

Also, this command does not work.
That is the reason why a slightly special parameter must be specified to ethtool in order to fix the link speed at 2500Mbps.
https://github.com/bb-qq/r8152/wiki/Troubleshooting#disable-auto-negotiation
https://github.com/bb-qq/r8152/blob/master/ReadMe.txt#L19

bb-qq, 2022-12-31:
Also, the device may not have enough power. You might try disconnecting all other USB devices as a test.

rtadams89, 2022-12-31:
> Yes, old kernels do not support 2500baseT, so the link mode is disguised as 2500baseX for ethtool in Realtek drivers
Ahh, that makes sense.
> Also, this command does not work.
Interesting, as that command does work for me. Once run, the link renegotiates to 2500 and even if I disconnect and reconnect the Ethernet cable, it again connects at 2500. It's only after a reboot that the link returns to 1000, which is expected.
I've tried running `ethtool -s eth2 autoneg on advertise 0x802f` and that produces the same effect.

rtadams89, 2022-12-31:
> Also, the device may not have enough power. You might try disconnecting all other USB devices as a test.

I just tried with my external usb drive removed and still get the same behavior. Maybe it's just the particular adapter I am using, though it all seems odd that it can be "fixed" by telling it to advertise 2500 from software.

bb-qq, 2023-03-18:
There does not seem to be anything more I can do about this issue.
Can we close this issue?

rtadams89, 2023-03-18:
The fact that I can make this work just by forcing the advertised speed to 2500 from software makes me think there is something in software that is wrong to start with. If there is any additional data points I can provide to help figure out what that is, let me know. If you disagree and think this is a hardware issue or otherwise not fixable in the software, feel free to close out.

bb-qq, 2023-03-18:
As for auto-negotiation, it is done at the hardware level between the Ethernet adapter and the switch. The driver just specifies whether auto-negotiation is performed or not, and which link speeds it advertises as supported.
https://github.com/bb-qq/r8152/blob/master/r8152.c#L9168

There is a non-zero possibility that some kind of timing is affecting this, but it isn't easy to corroborate. So please let me close this issue.

closed 2023-03-18 by bb-qq