bb-qq/r8152 #278: DS218+ works with Anker USB-Hub with Gigabit Ethernet

eauw, 2023-01-12:
I have a DS218+ and just tried it with a 3 Port USB hub with Gigabit Ethernet by Anker. The setup worked and the data transfer seems to be the same as with the internal ethernet.

Unfortunately I couldn't not find any technical details of the adapter.

I have two issues/questions though:
1. In the DSM networks menu I can't find any option to setup link aggregation like I can on DiskStations with two internal ethernet ports. Is it possible to enable link aggregation somehow?
2. The installed driver doesn't start automatically after a restart of the DiskStation. This means that I have to start it manually to enable the usb ethernet. Is there a way to make it start up automatically?

Thanks for this repository. Very good work!

TheS1R, 2023-01-12:
"Gigabit Ethernet" is the exact same as internal NIC.

eauw, 2023-01-12:
> "Gigabit Ethernet" is the exact same as internal NIC.

What does that mean or what do you want to tell me with this? :-)

TheS1R, 2023-01-12:
You need a RTL8152/RTL8153-based 2.5 Gbps USB adapter for this driver. The adapter that you describe will not work.

1. Yes. Control Panel >> Network >> Network interface >> Create >> Create Bond.
2. Buy a cpmpatible adapter.

eauw, 2023-01-12:
@TheS1R thank you. So both my issues are related to an actual incompatible adapter? Appearing in the network interfaces list made me think everything worked.