bb-qq/r8152 #280: Can't perform installation due to empty `/var/packages/` dir

altokarev, 2023-01-17:
Hey there! 👋 

I'm not able to install package due to

```
The operation failed. Please sign in to DSM again and retry.
```
error.

[image]

I assume that this behavior is OK, but there is no `r8152` directory appears at my `/var/packages/`. So I can't install anything with `install` command.

DS920+, Gemini Lake, DSM version 7.1.1-42962.

Thanks in advance! 🙏

altokarev, 2023-01-19:
I guess I figured this out.

The described issue persists only in Safari on Mac, but there are no problems with Chrome and Windows.

I assume this should be added to README in case this is a feature not a bug :)

closed 2023-01-19 by altokarev

bb-qq, 2023-01-28:
I added a warning of  Safari.