bb-qq/r8152 #281: [Success] `DS920+` + `D-Link DUB-2315` compatibility report

altokarev, 2023-01-20:
### Description

Combability report for `2.16.3-3 DSM7.x`.

Configured with SMB3 support, Jumbo Frames & static IP.

### Description of your products

* **Synology DiskStation DS920+** (DSM `v7.1.1-42962`)
* **[D-Link DUB-2315](https://eu.dlink.com/uk/en/products/dub-2315-usb-c-to-2-5g-ethernet-adapter)** USB-C to 2.5 Gigabit Ethernet network adapter with USB-A to USB-C adapter ([Realtek RTL8156B](https://www.realtek.com/en/products/communications-network-ics/item/rtl8156b-s-cg))

### Description of your environment

Connected directly to a PC using D-Link DGE-562T PCI Express 1x adapter with 100/1000/2.5GBase-T ([Realtek RTL8125BG](https://www.realtek.com/en/products/communications-network-ics/item/rtl8125bg-s-cg)) & cable Cat.6.

### Output of ``iperf3`` command

````shell
Connecting to host 192.168.2.3, port 5201
[  4] local 192.168.2.2 port 64939 connected to 192.168.2.3 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec   268 MBytes  2.25 Gbits/sec
[  4]   1.00-2.00   sec   279 MBytes  2.34 Gbits/sec
[  4]   2.00-3.00   sec   280 MBytes  2.34 Gbits/sec
[  4]   3.00-4.00   sec   282 MBytes  2.36 Gbits/sec
[  4]   4.00-5.00   sec   281 MBytes  2.36 Gbits/sec
[  4]   5.00-6.00   sec   281 MBytes  2.35 Gbits/sec
[  4]   6.00-7.00   sec   279 MBytes  2.34 Gbits/sec
[  4]   7.00-8.00   sec   281 MBytes  2.36 Gbits/sec
[  4]   8.00-9.00   sec   281 MBytes  2.36 Gbits/sec
[  4]   9.00-10.00  sec   281 MBytes  2.36 Gbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-10.00  sec  2.73 GBytes  2.34 Gbits/sec                  sender
[  4]   0.00-10.00  sec  2.73 GBytes  2.34 Gbits/sec                  receiver
````

altokarev, 2023-01-20:
Seems there's an issue with reboot. The interface doesn't start after rebooting the NAS. Log attached.

@bb-qq could you please take a look? 🙏 

[start-stop-status-log.txt](https://github.com/bb-qq/r8152/files/10471026/start-stop-status-log.txt)

DervilRus, 2023-01-25:
I stand by your every word. The problem only works after rebooting the DS or after juggling the USB device. After that, I need to restart the driver manually.

bb-qq, 2023-01-28:
If I reboot the device one more time, does the problem reproduce?

The first boot in the log shows no configuration file and the IP settings are skipped, but the second boot seems to have saved the configuration file correctly, so it looks like it would succeed the next time as well.

If the symptoms persist, please let me know the results of `dmesg` and `ifconfig -a` immediately after the reboot.

altokarev, 2023-01-30:
👋

Surprisingly, the second and all subsequent reboots were flawless! The interface start up as it should without any issues.

I think this should be added to the FAQ list.

altokarev, 2023-01-30:
BTW, I've tried my USB adapter with a rear port of the NAS and it works as stable & fast as front one:

```shell
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-10.00  sec   593 MBytes   498 Mbits/sec                  sender
[  4]   0.00-10.00  sec   593 MBytes   498 Mbits/sec                  receiver
[  6]   0.00-10.00  sec   589 MBytes   494 Mbits/sec                  sender
[  6]   0.00-10.00  sec   589 MBytes   494 Mbits/sec                  receiver
[  8]   0.00-10.00  sec   586 MBytes   491 Mbits/sec                  sender
[  8]   0.00-10.00  sec   586 MBytes   491 Mbits/sec                  receiver
[ 10]   0.00-10.00  sec   581 MBytes   487 Mbits/sec                  sender
[ 10]   0.00-10.00  sec   581 MBytes   487 Mbits/sec                  receiver
[ 12]   0.00-10.00  sec   576 MBytes   483 Mbits/sec                  sender
[ 12]   0.00-10.00  sec   576 MBytes   483 Mbits/sec                  receiver
[SUM]   0.00-10.00  sec  2.86 GBytes  2.45 Gbits/sec                  sender
[SUM]   0.00-10.00  sec  2.86 GBytes  2.45 Gbits/sec                  receiver

iperf Done.
```

mizxco, 2024-07-15:
Hey sorry it's been a while but trying to to the same on my DS920+, any issues for far with the rear USB port?