bb-qq/r8152 #282: Speed is low when copying files to/from NAS

Bkmz-GH, 2023-01-22:
### Description of the problem

Hello! Card is connected to front USB port, not identified by the system when connected to rear ports.
Card is identified by NAS as 2.5G, LEDs on card and unmanaged 2.5G capable switch are saying the same.
Iperf3 tests from PC get expected results around 2.2-2.4G on send and receive. Card speed was also tested in direct connection between 2 PCs and with switch between them, both tests showing speed around 2.5G.

The strange thing is when it comes to copying files to/from NAS (SMB/FTP was tested) - speed is around 1G (~100Mb/s).
When copying from NAS speed is always around 1G, but when uploading to NAS it starts at 2.5G but after some time (around 10 seconds) starts to decline ending at 1G.

### Description of your products

NAS: DS718+
output of ``uname -a`: Linux bkmz-nas 4.4.180+ #42962 SMP Tue Oct 18 15:05:01 CST 2022 x86_64 GNU/Linux synology_apollolake_718+
DSM Version: DSM 7.1.1-42962 Update 2
NIC: Noname (10$ on Aliexpress) USB (Type A connector) 2.5G, identified by Windows as "RealTek PCIe 2.5 GBE family controller"

### Description of your environment
> - connection using a Ethernet switch
> - PC: NIC model name - same as on NAS, also tried Intel I225-V
> - Ethernet switch: TP-LINK TL-SH1005
> - cable category: tried Cat.5e, then Cat.6

### Output of ``dmesg`` command
[dmesg.txt](https://github.com/bb-qq/r8152/files/10474885/dmesg.txt)

### Output of ``lsusb`` command

|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F00883A2FCA6047)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-3         0bda:8156:3104 00  3.20 5000MBit/s 256mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 4013000001)
|__usb3          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.180+ etxhci_hcd-170202 Etron xHCI Host Controller 0000:02:00.0) hub
|__usb4          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.180+ etxhci_hcd-170202 Etron xHCI Host Controller 0000:02:00.0) hub

````

### Output of ``ifconfig -a`` command
docker0   Link encap:Ethernet  HWaddr 02:42:76:72:BB:D9
          inet addr:172.17.0.1  Bcast:172.17.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth0      Link encap:Ethernet  HWaddr 00:11:32:99:B4:65
          UP BROADCAST SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:85 errors:0 dropped:0 overruns:0 frame:0
          TX packets:3 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:7471 (7.2 KiB)  TX bytes:726 (726.0 B)

eth1      Link encap:Ethernet  HWaddr 00:11:32:99:B4:66
          UP BROADCAST SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:48 errors:0 dropped:0 overruns:0 frame:0
          TX packets:3 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:4123 (4.0 KiB)  TX bytes:1026 (1.0 KiB)

eth2      Link encap:Ethernet  HWaddr 1C:BF:CE:2F:C4:81
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:9000  Metric:1
          RX packets:15536398 errors:0 dropped:0 overruns:0 frame:0
          TX packets:7434048 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:15306195839 (14.2 GiB)  TX bytes:8528407115 (7.9 GiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:115960 errors:0 dropped:0 overruns:0 frame:0
          TX packets:115960 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:12019961 (11.4 MiB)  TX bytes:12019961 (11.4 MiB)

ovs-syste Link encap:Ethernet  HWaddr 62:A2:CD:A0:4C:48
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

ovs_eth0  Link encap:Ethernet  HWaddr 00:11:32:99:B4:65
          inet addr:169.254.42.18  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

ovs_eth1  Link encap:Ethernet  HWaddr 00:11:32:99:B4:66
          inet addr:169.254.208.142  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

ovs_eth2  Link encap:Ethernet  HWaddr 1C:BF:CE:2F:C4:81
          inet addr:192.168.149.32  Bcast:192.168.149.255  Mask:255.255.255.0
          inet6 addr: fe80::1ebf:ceff:fe2f:c481/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:8100147 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1078249 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:12752778237 (11.8 GiB)  TX bytes:6021876373 (5.6 GiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

tap021132 Link encap:Ethernet  HWaddr F2:DA:D6:F0:D8:C3
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:187 errors:0 dropped:0 overruns:0 frame:0
          TX packets:36693 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:16776 (16.3 KiB)  TX bytes:6499070 (6.1 MiB)

tap021132 Link encap:Ethernet  HWaddr 42:66:52:FB:17:0E
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:43172 errors:0 dropped:0 overruns:0 frame:0
          TX packets:87752 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:10680584 (10.1 MiB)  TX bytes:25195419 (24.0 MiB)

tap021132 Link encap:Ethernet  HWaddr FE:C7:EC:44:3E:8E
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:225561 errors:0 dropped:0 overruns:0 frame:0
          TX packets:265272 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:198501018 (189.3 MiB)  TX bytes:209792830 (200.0 MiB)

tap021132 Link encap:Ethernet  HWaddr DE:AF:93:C2:50:10
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:2208287 errors:0 dropped:0 overruns:0 frame:0
          TX packets:2261274 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:2114792621 (1.9 GiB)  TX bytes:2181858290 (2.0 GiB)

### Output of ``ethtool eth2`` command
[ethtool.txt](https://github.com/bb-qq/r8152/files/10474886/ethtool.txt)

Bkmz-GH, 2023-01-26:
Small info update: rear ports are working with active USB-hub but speed is also
 around 1GB/s. As far as I understand similar problems were encountered [here](https://github.com/bb-qq/r8152/issues/33#issuecomment-1274703481)  and [here](https://github.com/bb-qq/r8152/issues/135#issuecomment-895321625). Attached screens for iperf3 results and PC to PC copying file through the same switch as NAS attached to. So the problem is almost definitely NAS/driver specific.

[image]
[image]

nagyrobi, 2023-02-12:
Look at CPU usage of NAS during file copy. The USB driver relies on the CPU, thus having a CPU bottleneck will lower USB network speeds too.

Bkmz-GH, 2023-02-15:
Hm, copying file takes more I/O waits.

File copy CPU load:
[image]
Iperf CPU load:
[image]

bb-qq, 2023-03-19:
Have you [set the SMB version](https://github.com/bb-qq/r8152/wiki/Troubleshooting#enable-smb3) to `SMB3`?

IAMakarov, 2023-03-26:
hello, 
[image]

bb-qq, 2023-04-22:
Are you using SSD cache?
It appears that disk access is the bottleneck, since iperf has no problems and the actual file transfer is fast for 10 seconds when uploading to the NAS. (I assume the first 10 seconds are being written to the cache in memory)

IAMakarov, 2023-04-22:
You seem to be right. Don't how and when I'll be able to check with faster disks, but in case you are right your driver works on DS718+ with that noname card sold on aliexpress:
[image]

Thanks for the great work!

bb-qq, 2023-04-23:
Can we close this issue and re-open it if the symptoms do not resolve after verifying with a faster disk?

IAMakarov, 2023-04-23:
> Can we close this issue and re-open it if the symptoms do not resolve after verifying with a faster disk?

ok, sure

closed 2023-04-30 by bb-qq