bb-qq/r8152 #285: Club3D adapter falls back to 100 Mbps, driver reinstall temporarily fixes it

IrealiTY, 2023-02-02:
### Description of the problem

Club 3D USB 3.2 Gen1 Adapter (vid:0bda pid:8156) seems to randomly drop to 100 Mbps link only until I re-run the `sudo install -m 4755 -o root -D /var/packages/r8152/target/r8152/spk_su /opt/sbin/spk_su` command and reinstall the driver via the package center.

### Description of your products

> - DS720+
>     - Linux 4.4.180+ #42962 SMP Tue Oct 18 15:07:03 CST 2022 x86_64 GNU/Linux synology_geminilake_720+
>     - DSM version 7.1.1-42962 Update 3
> - Club 3D USB 3.2 Gen1 (https://www.amazon.de/gp/product/B07Q626XK2)

### Description of your environment

> - connect directly to a Ethernet switch 2.5Gbe Switch which is connected to a 2.5Gbe Port on my router
>     - PC: NIC model name or chipset name
>     - Ethernet switch: Qnap QSW-1105-5T
>     - Router: Asus RT-AX86U with MerlinFW
> - cable category Cat.6

### Output of ``dmesg`` command

````
[dmesg.txt](https://github.com/bb-qq/r8152/files/10567416/dmesg.txt)

````

### Output of ``lsusb`` command

````
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F00097B16345A47)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-2         0bda:8156:3000 00  3.20 5000MBit/s 512mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 000000001)

````

### Output of ``ifconfig -a`` command

````
[ifconfig.txt](https://github.com/bb-qq/r8152/files/10567426/ifconfig.txt)
````

bb-qq, 2023-03-18:
Perhaps there is a problem with the router's auto-negotiation function.
If applying the latest firmware does not resolve the problem, try fixing the link speed by referring to the [troubleshooting page](https://github.com/bb-qq/r8152/wiki/Troubleshooting).

bb-qq, 2023-07-16:
I would like to close this issue as I have not heard back. If you have any questions, please feel free to re-open it.

closed 2023-07-16 by bb-qq

IrealiTY, 2023-07-16:
My bad, yes, it was a faulty cable!