bb-qq/r8152 #286: [Success] DS920+ and Club 3D CAC-1420

vseven, 2023-02-06:
### Description of the problem

Compatibility report

### Description of your products

DSM 7.1.1-42962 Update 1
Club 3DE CAC-1420

### Description of your environment

Ubiquiti UDM-Pro SE
Ubiquiti U6-Enterprise
2x Mikrotik S+RJ10 SFP+ adapter (Ports #10 & #11 both configured as LAN)

Both U6-Enterprise and Synology report connection speed at 2500.  Mikrotik adapters are internally bridged and can provide the 10Gb the UDM-Pro wants and the 2.5Gb the U6-Enterprise and Club 3D want.

ipref3 running within Docker on NAS.  Note - this is from a AX211 network card wirelessly.  This is partial output because I had to use -P  to saturate the link and no one wants to see all that.  Also a fun fact - this is 25 feet with LOS to the AP, putting the laptop closer actually hurt performance slightly, about 10% slower at 6 ft away.

Laptop to NAS (iperf3.exe -c NAS -P 24):
[SUM]   0.00-10.01  sec  1.60 GBytes  1.37 Gbits/sec                  sender
[SUM]   0.00-10.01  sec  1.60 GBytes  1.37 Gbits/sec                  receiver

NAS to Laptop (iperf3.exe -c NAS -P 24 -R):
[SUM]   0.00-10.00  sec  2.00 GBytes  1.71 Gbits/sec    1             sender
[SUM]   0.00-10.00  sec  1.99 GBytes  1.71 Gbits/sec                  receiver

Not sure what the discrepancy between sending and receiving but it's a good 20% that I'll chalk up to being wifi.

fip005c, 2023-03-07:
Thanks for reporting and giving hint to use parallel connections on iperf3. I couldn't understand why i was getting low throughput on my setup using iperf3, but good throughput when copying over large files using MacOS Finder. Any idea why single client can't achieve 1Gpbs over single TCP connection? My laptop intel i7 processor is quite capable...

vseven, 2023-03-07:
Jumbo frames needs to be turned on possibly?  I never bothered to figure it out as I'm happy enough with the performance.