bb-qq/r8152 #287: Asustor AS-2.5G2(RTL8156B): does not appear in DS416j Network Interface list

chatreip, 2023-02-11:
Description of the problem
I installed the 2.15.0-1 driver, the installation process went smoothly, but there is no USB displayed in the network panel, the following is the system log, I hope to help you solve it, thank you!

Description of your products
DS416j
CPU  MARVELL Armada 388 88F6828
DDR3 512MB
DSM 6.2.4-25556 Update 6
used driver: bb-qq 2.15.0-1 r8152-armada38x-2.14.0-2 from https://github.com/bb-qq/r8152/releases

Output of dmesg command

lsusb:
|__usb1          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (Linux 3.10.105 xhci-hcd xHCI Host Controller f10f0000.usb3) hub
|__usb2          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (Linux 3.10.105 xhci-hcd xHCI Host Controller f10f0000.usb3) hub
|__usb3          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (Linux 3.10.105 xhci-hcd xHCI Host Controller f10f8000.usb3) hub
|__usb4          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (Linux 3.10.105 xhci-hcd xHCI Host Controller f10f8000.usb3) hub
  |__4-1         0bda:8156:3100 00  3.20 5000MBit/s 256mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 0013000001)
|__usb5          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd f1058000.usb) hub

----------------------------------------------
ifconfig:
eth0      Link encap:Ethernet  HWaddr 00:11:32:**:**:**  
          inet addr:192.168.1.60  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: 2409:8a1e:2743:20e0:****:****:****:****/64 Scope:Global
          inet6 addr: fe80::211:****:****:****/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:12077 errors:0 dropped:0 overruns:0 frame:0
          TX packets:3300 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1024 
          RX bytes:2812956 (2.6 MiB)  TX bytes:995358 (972.0 KiB)
          Interrupt:193 

eth1      Link encap:Ethernet  HWaddr 00:E0:4C:**:**:**  
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:134 errors:0 dropped:0 overruns:0 frame:0
          TX packets:134 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:32047 (31.2 KiB)  TX bytes:32047 (31.2 KiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
		  
--------------------------------
dmesg:

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 3.10.105 (root@build4) (gcc version 4.9.3 20150311 (prerelease) (crosstool-NG 1.20.0) ) #25556 SMP Thu Jul 1 14:27:54 CST 2021
[    0.000000] CPU: ARMv7 Processor [414fc091] revision 1 (ARMv7), cr=10c53c7d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine: Marvell Armada 380/381/382/383/384/385/388 (Device Tree), model: Synology DS416j
[    0.000000] Memory policy: ECC disabled, Data cache writealloc
[    0.000000] On node 0 totalpages: 131072
[    0.000000] free_area_init_node: node 0, pgdat 8066e340, node_mem_map 806d5000
[... 512 lines cut ...]
[  148.902116] loop: module loaded
[  153.407935] findhostd uses obsolete (PF_INET,SOCK_PACKET)
[  156.716312] r8152: module verification failed: signature and/or required key missing - tainting kernel
[  156.728014] usbcore: registered new interface driver r8152
[  156.891821] usb 4-1: reset SuperSpeed USB device number 2 using xhci-hcd
[  156.911818] usb 4-1: Parent hub missing LPM exit latency info.  Power management will be impacted.
[  156.961669] xhci-hcd f10f8000.usb3: xHCI xhci_drop_endpoint called with disabled ep 9c0d7600
[  156.970212] xhci-hcd f10f8000.usb3: xHCI xhci_drop_endpoint called with disabled ep 9c0d762c
[  156.978779] xhci-hcd f10f8000.usb3: xHCI xhci_drop_endpoint called with disabled ep 9c0d7658
[  164.408742] init: syno-check-disk-compatibility main process (10956) terminated with status 255
[  164.517938] Synotify use 16384 event queue size
[  164.522629] Synotify use 16384 event queue size
[  173.201818] netif_napi_add() called with weight 256 on device eth%d
[  173.251605] r8152 4-1:1.0 eth1: v2.14.0 (2020/09/24)
[  173.256661] r8152 4-1:1.0 eth1: This product is covered by one or more of the following patents:
               		US6,570,884, US6,115,776, and US6,327,625.
               
[  173.990470] usbcore: registered new interface driver snd-usb-audio
[  174.412816] usbcore: registered new interface driver snd-usb-hiface
[  200.681275] Synotify use 16384 event queue size
[  200.686002] Synotify use 16384 event queue size
[  201.181133] init: synocontentextractd main process ended, respawning
[  208.616428] Synotify use 16384 event queue size
[  208.621040] Synotify use 16384 event queue size
[  231.642605] Synotify use 16384 event queue size

chatreip, 2023-02-13:
Add some content, the log of the plug-in startup process:

`[179161.248138] usb 4-1: reset SuperSpeed USB device number 3 using xhci-hcd
[179161.268223] usb 4-1: Parent hub missing LPM exit latency info.  Power management will be impacted.
[179161.277469] xhci-hcd f10f8000.usb3: xHCI xhci_drop_endpoint called with disabled ep 9c64ac00
[179161.286067] xhci-hcd f10f8000.usb3: xHCI xhci_drop_endpoint called with disabled ep 9c64ac2c
[179161.294669] xhci-hcd f10f8000.usb3: xHCI xhci_drop_endpoint called with disabled ep 9c64ac58
[179161.498085] r8152 4-1:1.0 eth1: v2.14.0 (2020/09/24)
[179161.503178] r8152 4-1:1.0 eth1: This product is covered by one or more of the following patents:
                		US6,570,884, US6,115,776, and US6,327,625.
                
[179161.519172] usbcore: registered new interface driver r8152
[179166.701827] ifconfig: page allocation failure: order:4, mode:0x40d0
[179166.708248] CPU: 1 PID: 31638 Comm: ifconfig Tainted: PF          O 3.10.105 #25556
[179166.716049] [<8001462c>] (unwind_backtrace+0x0/0xec) from [<8001171c>] (show_stack+0x10/0x14)
[179166.724761] [<8001171c>] (show_stack+0x10/0x14) from [<8009ed8c>] (warn_alloc_failed+0xe4/0x11c)
[179166.733718] [<8009ed8c>] (warn_alloc_failed+0xe4/0x11c) from [<800a1048>] (__alloc_pages_nodemask+0x58c/0x788)
[179166.743908] [<800a1048>] (__alloc_pages_nodemask+0x58c/0x788) from [<7f7c49cc>] (alloc_rx_agg+0x4c/0x134 [r8152])
[179166.754359] [<7f7c49cc>] (alloc_rx_agg+0x4c/0x134 [r8152]) from [<7f7c70bc>] (rtl8152_open+0xb4/0x3e8 [r8152])
[179166.764563] [<7f7c70bc>] (rtl8152_open+0xb4/0x3e8 [r8152]) from [<803c7638>] (__dev_open+0x98/0xf8)
[179166.773794] [<803c7638>] (__dev_open+0x98/0xf8) from [<803c7878>] (__dev_change_flags+0x94/0x128)
[179166.782831] [<803c7878>] (__dev_change_flags+0x94/0x128) from [<803c7978>] (dev_change_flags+0x10/0x48)
[179166.792392] [<803c7978>] (dev_change_flags+0x10/0x48) from [<8041f168>] (devinet_ioctl+0x640/0x714)
[179166.801610] [<8041f168>] (devinet_ioctl+0x640/0x714) from [<803af69c>] (sock_ioctl+0x138/0x288)
[179166.810453] [<803af69c>] (sock_ioctl+0x138/0x288) from [<800ecb38>] (do_vfs_ioctl+0x4b4/0xaa8)
[179166.819208] [<800ecb38>] (do_vfs_ioctl+0x4b4/0xaa8) from [<800ed19c>] (SyS_ioctl+0x70/0x80)
[179166.827701] [<800ed19c>] (SyS_ioctl+0x70/0x80) from [<8000e080>] (ret_fast_syscall+0x0/0x38)
[179166.836277] Mem-info:
[179166.838655] Normal per-cpu:
[179166.841548] CPU    0: hi:  186, btch:  31 usd: 180
[179166.846444] CPU    1: hi:  186, btch:  31 usd:   0
[179166.851353] active_anon:11607 inactive_anon:14352 isolated_anon:0
                 active_file:15530 inactive_file:14309 isolated_file:0
                 unevictable:540 dirty:0 writeback:323 unstable:0
                 free:31737 slab_reclaimable:7145 slab_unreclaimable:24417
                 mapped:9568 shmem:5571 pagetables:3287 bounce:0
                 free_cma:0
[179166.884648] Normal free:126948kB min:8192kB low:10240kB high:12288kB active_anon:46428kB inactive_anon:57408kB active_file:62120kB inactive_file:57236kB unevictable:2160kB isolated(anon):0kB isolated(file):0kB present:524288kB managed:508188kB mlocked:2160kB dirty:0kB writeback:0kB mapped:38272kB shmem:22284kB slab_reclaimable:28580kB slab_unreclaimable:97668kB kernel_stack:3808kB pagetables:13148kB unstable:0kB bounce:0kB free_cma:0kB writeback_tmp:0kB pages_scanned:2 all_unreclaimable? no
[179166.928376] lowmem_reserve[]: 0 0
[179166.931828] Normal: 15339*4kB (UEM) 5847*8kB (UEM) 1066*16kB (EMR) 22*32kB (EM) 0*64kB 1*128kB (R) 1*256kB (R) 1*512kB (R) 0*1024kB 0*2048kB 0*4096kB = 126788kB
[179166.946578] 40310 total pagecache pages
[179166.950527] 4837 pages in swap cache
[179166.954207] Swap cache stats: add 51238, delete 46401, find 291772/292881
[179166.961132] Free swap  = 1939440kB
[179166.964641] Total swap = 2097084kB
[179166.971215] 131072 pages of RAM
[179166.974460] 32933 free pages
[179166.977443] 3109 reserved pages
[179166.980698] 31550 slab pages
[179166.983676] 61912 pages shared
[179166.986827] 4837 pages swap cached`

bb-qq, 2023-03-18:
It appears to be running out of memory.
There are changes in the latest drivers to reduce memory consumption. Perhaps that change might help.
Can you upgrade DSM to 7?

Barlos, 2023-05-27:
Try to change to other USB port.

bb-qq, 2023-07-15:
I have made the latest driver available for DSM 6.2 as well, so please try it.

bb-qq, 2023-08-13:
I would like to close this issue since there has been no response. If you have any questions, feel free to re-open it.

closed 2023-08-13 by bb-qq