bb-qq/r8152 #289: Noob help with DS218+

ltuff1, 2023-02-12:
Description of Problem:

Okay, so I’ve tried to install this (r8152-apollolake-2.16.3-3) on my DS-281+ but was unsuccessful. Logged into my NAS (via Windows 10/Microsoft Edge), enabled SSH, and tried to manually install the .spk. It tells me that it’s not supported on the platform and incompatible with my DSM (I’m on 7.1.1-42962 Update 3), which I assume is just the “installation will fail the first time”. I open a Command Prompt in Win10 and run and successfully connect to my NAS via SSH (though it tells me it cannot change the directory to var/packages…/user because there’s no such file directory). When I run sudo install…/spk_su command it tells me it cannot stat as there’s no such file or directory. Have tried a couple of times (attempting manual install once or twice before sending the SSH command) and have attempted using root privileges as well. As I don’t really know what I’m doing [noob] otherwise (so not sure about how to get output from dmesg, lsub or ifconfig -a, if needed), I’m pretty much stuck at this point thinking I’m missing something quite basic (as others with the 218+ have been successful)…??

Description of Products:

Synology DS218+ NAS
DSM Version 7.1.1-42962 Update 3
Ethernet Dongle Club 3D USB 3.2 Gen1 Type-A to RJ45 2.5Gbps

Description of Environment:

Connected to Win10 PC (Intel I217-V) through Asus RT-AX89X router and QNAP QSW-2104-2T switch (all by ethernet CAT6/CAT 7 cabling) (though have a Win11 system with a 2.5g NIC that I will be using shortly)

dreimer1986, 2023-02-17:
Who told you to install anything from SSH? Check the 5 installation steps again and do as they say.

ltuff1, 2023-02-18:
Sorry, so if I prepare by setting up SSH access as instructed and then the steps are:

1. Go to "Package Center"
2. Press "Manual Install"
3. Chose a driver package downloaded from the [release page](https://github.com/bb-qq/r8152/releases).
4. [DSM7] The installation will fail the first time. After that, run the following command from the SSH terminal: sudo install -m 4755 -o root -D /var/packages/r8152/target/r8152/spk_su /opt/sbin/spk_su
5. [DSM7] Retry installation. (You don't need above DSM7 specific steps at the next time.)

what exactly am I doing in step 4 when I open up a command prompt in Windows, connect to my NAS, and then run the command? Have I totally got this wrong?

dreimer1986, 2023-02-18:
You install the package on the NAS WebUI and there in the Packet Center under Manual Installation or similar (Mine is set to German, so I roughly translated it). Then it fails. THEN you open SSH and insert the command and then you go back to the WebUI and install again. This time with it succeeding.

ltuff1, 2023-02-18:
Yes, that's exactly what I do...

dreimer1986, 2023-02-18:
OK, I misread your first post. Sorry for that. But this one liner in SSH should do it.

ltuff1, 2023-02-18:
Hmm…so in the Synology WebUI doing a Manual Install: Upload a Package of the .spk file returns:
“This package is not supported on the platform of Synology NAS or is incompatible with the current DSM version”
I then open a command prompt in Windows 10 and run “username@NAS-IP -p22” and enter my password and it returns:
“Could not chdir to home directory /var/services/homes/username: No such file or directory”
If I go ahead anyway and run the one line “sudo install -m 4755 -o root…/spk_su” copied and pasted from the installation instructions (and my password again) and it returns:
“install: cannot stat '/var/packages/r8152/target/r8152/spk_su': No such file or directory”
Going back into the NAS WebUI and repeating the manual install then returns the same response as the first time through.
Seems like I’m missing something very basic about where the install is looking for files or the directory I’m in...

dreimer1986, 2023-02-18:
"This package is not supported on the platform of Synology NAS or is incompatible with the current DSM version" should not happen. CPU seems to be right though... No clue what's going on there. I have a 420+ aka Geminilake and it worked.

ltuff1, 2023-02-18:
Oh crap...just realized I was in fact trying to install it on my DS220+ (I have both a 218+ and a 220+) which is a Geminilake chipset,  and not the Apollolake (duh!), so you're right. Will try it with the right NAS. But this is actually good news, I wasn't expecting that I might also be able to do it on my 220+ - which package did you use on your 420+ that worked?

dreimer1986, 2023-02-19:
This explains it ^^ So it indeed was the wrong CPU.
https://kb.synology.com/de-de/DSM/tutorial/What_kind_of_CPU_does_my_NAS_have
Here is the list of the CPUs used in the NAS systems. I use the newest Geminilake on mine and your's seems to have the same CPU.

bb-qq, 2023-03-18:
I am closing this issue as the issue appears to be resolved.

closed 2023-03-18 by bb-qq