bb-qq/r8152 #292: Worked for DS920+ and Cable Matters USB C to 2.5 Gigabit Ethernet Adapter

fip005c, 2023-03-05:
### Description of the problem
Worked for DS920+ and Cable Matters USB C to 2.5 Gigabit Ethernet Adapter.

After first reboot the driver didn't start automatically and I had to start it manually from UI. On subsequent reboots the driver starts automatically as expected so I could unplug 1Gbps connection and stay on 2.5Gpbs as permanent solution.

### Description of your products

NAS: 
$uname -a
Linux NAS-DS920plus 4.4.180+ #42962 SMP Tue Oct 18 15:07:03 CST 2022 x86_64 GNU/Linux synology_geminilake_920+
DSM: 7.1.1-42962 Update 3

Dongle: Cable Matters USB C to 2.5 Gigabit Ethernet Adapter, USB to 2.5 Gigabit Ethernet Adapter, Not Compatible with M2 Macs (2.5g Ethernet to USB, 2.5gbps Ethernet Port) -Compatible with Thunderbolt 4 USB 4
https://www.amazon.com/gp/product/B0856V9FDH/ref=ppx_yo_dt_b_asin_title_o01_s01?ie=UTF8&psc=1

### Description of your environment

I connect dongle to NAS back USB port.
Both dongle and my PC connected to TRENDnet 8-Port Unmanaged 2.5G Switch (https://www.amazon.com/gp/product/B08XWKF55C/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1)
All cabling is CAT6A (variant A)

### Output of ``dmesg`` command

$ dmesg | grep r8152 
[   50.802158] systemd[1]: Created slice r8152's slice.
[   50.812364] systemd[1]: Starting r8152's slice.
[   94.378980] r8152: module verification failed: signature and/or required key missing - tainting kernel
[   94.395475] usbcore: registered new interface driver r8152
[   94.676758] r8152 2-2:1.0 eth2: v2.16.3 (2022/07/06)
[   94.682325] r8152 2-2:1.0 eth2: This product is covered by one or more of the following patents:
[   94.699789] r8152 2-2:1.0 eth2: chip rev 14
[   94.704477] r8152 2-2:1.0 eth2: rx_buf_sz will be overridden to 16384
[   98.439078] r8152 2-2:1.0 eth2: carrier on

### Output of ``lsusb`` command

$ lsusb
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology Diskstation 7F00167F851A2E40)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-2         0bda:8156:3100 00  3.20 5000MBit/s 256mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 0013000001)

### Output of ``ifconfig -a`` command

ifconfig -a
docker0   Link encap:Ethernet  HWaddr 02:42:CF:6F:3A:6B  
          inet addr:172.17.0.1  Bcast:172.17.255.255  Mask:255.255.0.0
          inet6 addr: fe80::42:cfff:fe6f:3a6b/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:5 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:438 (438.0 B)

docker350 Link encap:Ethernet  HWaddr 76:66:27:56:58:39  
          inet6 addr: fe80::7466:27ff:fe56:5839/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:5 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:0 (0.0 B)  TX bytes:438 (438.0 B)

docker7a3 Link encap:Ethernet  HWaddr 9E:C4:54:06:D3:85  
          inet6 addr: fe80::9cc4:54ff:fe06:d385/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:10 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:0 (0.0 B)  TX bytes:876 (876.0 B)

eth0      Link encap:Ethernet  HWaddr 00:11:32:FD:60:1B  
          inet addr:192.168.50.77  Bcast:192.168.50.255  Mask:255.255.255.0
          inet6 addr: fd3c:f45c:382a:c8a2:211:32ff:fefd:601b/64 Scope:Global
          inet6 addr: fe80::211:32ff:fefd:601b/64 Scope:Link
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:7885 errors:0 dropped:0 overruns:0 frame:0
          TX packets:6295 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:3488241 (3.3 MiB)  TX bytes:12959241 (12.3 MiB)
          Interrupt:91 base 0x6000  

eth1      Link encap:Ethernet  HWaddr 00:11:32:FD:60:1C  
          inet addr:169.254.121.37  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:90 base 0x4000  

eth2      Link encap:Ethernet  HWaddr 5C:85:7E:3D:DD:04  
          inet addr:192.168.50.209  Bcast:192.168.50.255  Mask:255.255.255.0
          inet6 addr: fe80::5e85:7eff:fe3d:dd04/64 Scope:Link
          inet6 addr: fd3c:f45c:382a:c8a2:5e85:7eff:fe3d:dd04/64 Scope:Global
          UP BROADCAST RUNNING MULTICAST  MTU:9000  Metric:1
          RX packets:174751 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1206781 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:18818516 (17.9 MiB)  TX bytes:1735119165 (1.6 GiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:5450 errors:0 dropped:0 overruns:0 frame:0
          TX packets:5450 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:1220888 (1.1 MiB)  TX bytes:1220888 (1.1 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

tun0      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  
          inet addr:10.8.112.54  P-t-P:10.8.112.54  Mask:255.255.255.0
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:831 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:500 
          RX bytes:0 (0.0 B)  TX bytes:92640 (90.4 KiB)

HepoH3, 2023-03-06:
Hi, thank you for your feedback!
May I ask you to help with that part (I'm quite a noob):

> I had to start it manually from UI

fip005c, 2023-03-06:
Hi Andrei,
You just need to login into synology web UI. Go to "Package Center",
"Installed", "RTL8152/RTL8153 driver" and click on "Run" button so that the
screen will look like below.
On the next system restart you do not need these steps anymore so it's just
one time (you may need to do this on system DSM upgrade, but i'm not sure
yet). This seems to be a common issue for everyone complains about.

[image: image.png]

HepoH3, 2023-03-06:
> You just need to login into synology web UI. Go to "Package Center", "Installed", "RTL8152/RTL8153 driver" and click on "Run" button so that the screen will look like below. 

Thank you!

MilesTEG1, 2023-03-10:
For me, this adaptator wasn't stable. I had to send it back to the reseler.
After some high transfers, it crashes the driver. had to reboot the NAS to get access again.

fip005c, 2023-03-11:
Thanks for update. Was it one time incident? How large was the transfer?
I’d like to reproduce the load to rule out systematic issue.

I have 2 of these adapters. One was working fine as main network connection
for my work MacBook for a month. After that I got second one for NAS and so
far it’s working fine. I have hourly backups from MacBook and occasional
photos sync from family phones.

MilesTEG1, 2023-03-13:
> Thanks for update. Was it one time incident? How large was the transfer? I’d like to reproduce the load to rule out systematic issue. I have 2 of these adapters. One was working fine as main network [...]

Hello,
I already sent back this adapter to amazon to get a refund and buy another brand. I'm sorry I can't help you more on this Cable Matters adapter.

I'll create a new issue for the new one I received yesterday.

ryaneford, 2023-12-28:
Adding to this.  Worked for me like a dream. Thanks for adding!