bb-qq/r8152 #296: DS920+ and Asustor AS-U2.5G2 adapter : works but not at full speed for some client/server...

MilesTEG1, 2023-03-13:
### Description of the problem

> I bought an Asustor `AS-U2.5G2` adapter (not the same as the first gen `AS-U2.5G`) to replace a probably defect Cable Matters 2,5G adapter.
> I plugged it correctly in the USB3-USBC adapter supplied with the `AS-U2.5G2` (otherwise the 2,5G adapter is show as 1000M instead of 2500M). I also disable IPv6 on the LAN3 (eth2).
> [image]
>
> I launched severals iperf3 tests with this command while the NAS is the server with `iperf3 -s` :
> ```bash
>  # Launched on two differents clients : Asustor AS6704T, Macbook Air M2 with an Asustor `AS-U2.5G` adapter
>  iperf3 -c 192.168.2.201 -t 20 && iperf3 -c 192.168.2.201 -t 20 -R
>  ```
> The results are distrubing...
> 
> With `DS920+` as a server :
> - As a client: my MBA M2 with Asustor `AS-U2.5G` adapter : 
>     - `MBA M2 --> DS920+ ≈ 1,54 Gbits/sec` :warning:
>     - `DS920+ --> MBA M2 ≈ 2,35 Gbits/sec` (with `--reverse` mode)
> - As a client: Asustor `AS6704T` on LAN1 (eth0) (or on LAN2 (eth1)) port (2,5G) : 
>     - `AS6704T --> DS920+ ≈ 2,13 Gbits/sec`
>     - `DS920+ --> AS6704T ≈ 820 Mbits/sec` :warning: :warning: (with `--reverse` mode)
>
> With `AS6704T` as a server :
> - As a client: my MBA M2 with Asustor `AS-U2.5G` adapter: 
>     - `MBA M2 --> AS6704T ≈ 1,54 Gbits/sec` :warning:
>     - `AS6704T --> MBA M2 ≈ 2,35 Gbits/sec` (with `--reverse` mode)
> - As a client: Synology `DS920+` with  `AS-U2.5G2` adapter: 
>     - `DS920+ --> AS6704T ≈ 898 Mbits/sec` :warning: :warning:
>     - `AS6704T --> DS920+ ≈ 2,32 Gbits/sec` (with `--reverse` mode)
>

Not Tested yet :
> With `MBA M2` (with Asustor `AS-U2.5G` adapter) as a server :
> - As a client: Asustor `AS6704T`: 
>     - `AS6704T --> MBA M2 ≈ xxxx Gbits/sec`
>     - `MBA M2 --> AS6704T ≈ xxxx Gbits/sec` (with `--reverse` mode)
> - As a client: Synology `DS920+` with  `AS-U2.5G2` adapter: 
>     - `DS920+ --> MBA M2 ≈ xxxx Gbits/sec`
>     - `MBA M2 --> DS920+ ≈ xxxx Gbits/sec` (with `--reverse` mode)

The results with :warning: emoji are weird because of the other results...
I don't understand what's going wrong...

### Description of your products

> - Synology DS920+
>     - and output of ``uname -a`` command :
>       ```
>       Linux Syno-DS920Plus 4.4.180+ #42962 SMP Tue Jan 31 23:18:09 CST 2023 x86_64 GNU/Linux synology_geminilake_920+
>       ```
>     - DSM version : `7.1.1-42962 Update 4`
> - the product name of the ethernet dongle : `Asustor AS-U2.5G2`

### Description of your environment

> - connect ***using a Ethernet switch***
>     - Ethernet switch: `Asustor ASW205T`
> - cable category : `Cat.6 and Cat.6a`
> - The other machines connected to this switch:
>     - Macbook Air M2 + `Asustor AS-U2.5G` (the first edition)
>     - Asustor AS6704T (already has two 2,5G ethernet ports)

<details><summary>Here the results for the iperf3 tests between and Synology DS920+ Asustor AS6704T with DS920+ as server</summary>
<p>

#### `AS6704T --> DS920+` from LAN1 = eth0

```log
iperf3 -c 192.168.2.201 -t 20
Connecting to host 192.168.2.201, port 5201
[  5] local 192.168.2.203 port 52880 connected to 192.168.2.201 port 5201
[... 236 lines cut ...]
          inet addr:192.168.2.200  Bcast:192.168.2.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1076233098 errors:0 dropped:0 overruns:0 frame:0
          TX packets:655539717 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:1132580045758 (1.0 TiB)  TX bytes:77135358200 (71.8 GiB)
          Interrupt:97 base 0xc000

eth1      Link encap:Ethernet  HWaddr 00:11:32:xx:xx:xx
          inet addr:169.254.26.251  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:6301674 errors:0 dropped:0 overruns:0 frame:0
          TX packets:5607354 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:7251967853 (6.7 GiB)  TX bytes:12489679152 (11.6 GiB)
          Interrupt:96 base 0x8000

eth2      Link encap:Ethernet  HWaddr 00:E0:4C:xx:xx:xx
          inet addr:192.168.2.201  Bcast:192.168.2.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:22947233 errors:0 dropped:0 overruns:0 frame:0
          TX packets:122585745 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:3623196154 (3.3 GiB)  TX bytes:178057328592 (165.8 GiB)
````

MilesTEG1, 2023-03-13:
Update previous message with last tests results : the server on my MBA M2.
It appears, that it's a bit low in performance. I tested the same adapter with my wife's iMac M1, and there it's better.

bb-qq, 2023-03-19:
From the iperf3 output, it appears that there are a lot of TCP retransmissions and congestion control is reducing throughput.
(see the `Retr` column)

The combination of MBA M2 and AS6704T also seems to be causing the slowdown, so you may want to suspect the Ethernet cable or switch.

You may also find the [Troubleshooting page](https://github.com/bb-qq/r8152/wiki/Troubleshooting) useful.

MilesTEG1, 2023-03-19:
> From the iperf3 output, it appears that there are a lot of TCP retransmissions and congestion control is reducing throughput. (see the `Retr` column)

That's what I thought, but I wasn't sure the meanings of the `Retr` column...

> The combination of MBA M2 and AS6704T also seems to be causing the slowdown, so you may want to suspect the Ethernet cable or switch.
> 
> You may also find the [Troubleshooting page](https://github.com/bb-qq/r8152/wiki/Troubleshooting) useful.

I already look into the troubleshooting page :)
You say that the problem should be the ethernet cable ?
It's a 6a cat, 5m length. I can try another one.
The switch seems to be OK, I already try another port.
And as the problem doesn't occur between the iMac and the Asustor, I suspect the problem is located on the DS920+.
My MBA seems to have only uploads problems. From the iMac, it seems there is no problem.

So, for now, I just had to deal with this problem.
Next month, I'll buy another adapter for the iMac, so I could test this new one on the Synology.

PS : I didn't change the MTU values, because I can't set a value greater than 1500 on my mac:
[image]
I keep the LAN1 (eth0) plugged in another switch (a gigabit one) for all my critical services (reverse proxy, servers, etc...)
I set a static route for the Asustor NAS IP :
[image]

I also tried to remove temporally the 1GB link and the tests have the same results.

bb-qq, 2023-03-19:
To isolate the cause, it may help to connect the PC and NAS directly with a shorter cable instead of through a switch.

If the problem is on the NAS side, it may be that burst transfers are causing a temporary memory shortage. In this case, you may get some error output to `dｍesg`.

MilesTEG1, 2023-03-19:
> To isolate the cause, it may help to connect the PC and NAS directly with a shorter cable instead of through a switch.
> 
> If the problem is on the NAS side, it may be that burst transfers are causing a temporary memory shortage. In this case, you may get some error output to `dｍesg`.

I tried another cable, and ran an iperf3 from the mac  :
```shell
Connecting to host 192.168.2.201, port 5201
[  5] local 192.168.2.30 port 57222 connected to 192.168.2.201 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec   162 MBytes  1.36 Gbits/sec
[  5]   1.00-2.01   sec   212 MBytes  1.77 Gbits/sec
[  5]   2.01-3.00   sec   254 MBytes  2.14 Gbits/sec
[  5]   3.00-4.00   sec   267 MBytes  2.24 Gbits/sec
[  5]   4.00-5.00   sec   265 MBytes  2.22 Gbits/sec
[  5]   5.00-6.00   sec   266 MBytes  2.23 Gbits/sec
[  5]   6.00-7.00   sec   276 MBytes  2.32 Gbits/sec
[  5]   7.00-8.00   sec   250 MBytes  2.10 Gbits/sec
[  5]   8.00-9.00   sec   250 MBytes  2.10 Gbits/sec
[  5]   9.00-10.00  sec   216 MBytes  1.81 Gbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-10.00  sec  2.36 GBytes  2.03 Gbits/sec                  sender
[  5]   0.00-10.00  sec  2.36 GBytes  2.03 Gbits/sec                  receiver

iperf Done.
Connecting to host 192.168.2.201, port 5201
Reverse mode, remote host 192.168.2.201 is sending
[  5] local 192.168.2.30 port 57244 connected to 192.168.2.201 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  96.8 MBytes   812 Mbits/sec
[  5]   1.00-2.00   sec  99.1 MBytes   832 Mbits/sec
[  5]   2.00-3.00   sec  98.7 MBytes   828 Mbits/sec
[  5]   3.00-4.00   sec   100 MBytes   839 Mbits/sec
[  5]   4.00-5.00   sec  95.1 MBytes   798 Mbits/sec
[  5]   5.00-6.00   sec   100 MBytes   839 Mbits/sec
[  5]   6.00-7.00   sec   101 MBytes   848 Mbits/sec
[  5]   7.00-8.00   sec  99.9 MBytes   838 Mbits/sec
[  5]   8.00-9.00   sec  98.6 MBytes   827 Mbits/sec
[  5]   9.00-10.00  sec   101 MBytes   844 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   991 MBytes   831 Mbits/sec  372             sender
[  5]   0.00-10.00  sec   990 MBytes   830 Mbits/sec                  receiver

iperf Done.
``` 

I extract this `dmesg` logs starting this morning when I installed the latest preview build of your driver:
dmesg return this:
```log
[Sun Mar 19 10:55:04 2023] Module [r8152] is removed.
[Sun Mar 19 10:55:04 2023] usbcore: deregistering interface driver r8152
[Sun Mar 19 10:55:10 2023] audit_printk_skb: 12 callbacks suppressed
[Sun Mar 19 10:55:10 2023] audit: type=1325 audit(1679219860.776:11575): table=filter family=2 entries=98
[Sun Mar 19 10:55:10 2023] audit: type=1325 audit(1679219860.799:11576): table=filter family=2 entries=98
[Sun Mar 19 10:55:10 2023] audit: type=1325 audit(1679219860.881:11577): table=filter family=2 entries=98
[Sun Mar 19 10:55:10 2023] audit: type=1325 audit(1679219860.898:11578): table=filter family=10 entries=59
[Sun Mar 19 10:55:10 2023] audit: type=1325 audit(1679219860.914:11579): table=filter family=10 entries=59
[Sun Mar 19 10:55:10 2023] audit: type=1325 audit(1679219860.930:11580): table=filter family=10 entries=59
[Sun Mar 19 10:55:10 2023] audit: type=1325 audit(1679219860.948:11581): table=filter family=2 entries=102
[Sun Mar 19 10:55:10 2023] audit: type=1325 audit(1679219860.964:11582): table=filter family=10 entries=63
[Sun Mar 19 10:55:10 2023] audit: type=1325 audit(1679219860.980:11583): table=filter family=2 entries=88
[Sun Mar 19 10:55:10 2023] audit: type=1325 audit(1679219861.001:11584): table=filter family=10 entries=43
[Sun Mar 19 10:55:15 2023] usb 2-1: reset SuperSpeed USB device number 11 using xhci_hcd
[Sun Mar 19 10:55:15 2023] r8152 2-1:1.0 eth2: v2.16.3 (2022/07/06)
[Sun Mar 19 10:55:15 2023] r8152 2-1:1.0 eth2: This product is covered by one or more of the following patents:
                           		US6,570,884, US6,115,776, and US6,327,625.

[Sun Mar 19 10:55:15 2023] r8152 2-1:1.0 eth2: chip rev 14
[Sun Mar 19 10:55:15 2023] r8152 2-1:1.0 eth2: rx_buf_sz will be overridden to 16384
[... 42 lines cut ...]
Reverse mode, remote host 192.168.2.200 is sending
[  5] local 192.168.2.30 port 60215 connected to 192.168.2.200 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  96.2 MBytes   807 Mbits/sec
[  5]   1.00-2.00   sec   103 MBytes   861 Mbits/sec
[  5]   2.00-3.00   sec  99.9 MBytes   838 Mbits/sec
[  5]   3.00-4.00   sec  98.0 MBytes   822 Mbits/sec
[  5]   4.00-5.00   sec   103 MBytes   862 Mbits/sec
[  5]   5.00-6.00   sec   100 MBytes   841 Mbits/sec
[  5]   6.00-7.00   sec   102 MBytes   859 Mbits/sec
[  5]   7.00-8.00   sec   103 MBytes   862 Mbits/sec
[  5]   8.00-9.00   sec   102 MBytes   858 Mbits/sec
[  5]   9.00-10.00  sec   101 MBytes   850 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  1010 MBytes   847 Mbits/sec  435             sender
[  5]   0.00-10.00  sec  1008 MBytes   846 Mbits/sec                  receiver

iperf Done.
```
Even on the built-in ethernet port (and connected to another switch), there are some Retries!
(this test was conduct while the mac were connected to the 2,5G switch, and the NAS to the 1G switch, the 2 switches are linked of course).
I tried both connected to the 1G switch, résultats are the same : much Retries.

If I do the same tests with the Asustor as server, I have much much less Retries, 0-20.

MilesTEG1, 2023-04-05:
Hello,
Here some news !

Fist of all, I bought another adaptor, the [Club3D one, a CAC-1420 USB 3.2 Gen1 Type-A to RJ46 2.5Gbps](https://amzn.eu/d/hzwIxQ8).
But sadly, I had the same results... upload from my 920+ was pooly bad... like my previous tests.

After cleaning the NAS from the driver and driver's preferences (see #299 ), I installed it again, from scratch.
And, after some reboots, I didn't see any new interface eth2... only my built-in eth0 and eth1... 
I try to uninstall from DSM, reboot, and reinstall it again... with no positive result. 😞  Still no eth2 interface, even if the driver were well loaded and functional...

So I hit `lsusb` command to see if there is my adapter... No one were listed... Crap...
Before the first cleansing of the driver and preference's files, it was on the list... and worked (ok not very well...), but according to the FAQ, maybe a USB Hub with power should resolve stability and low upload throughput... So I bought [a powered usb3 hub](https://amzn.eu/d/fu9Kfaa), and plug my adaptor on it, and it worked... but after uninstalling/reinstalling the driver, it works no more...

I unplug the adaptor from the USB-HUB, and plug it into the front USB-A connector, restart the driver with my script (see #295 ) and Oh-my ! ETH2 was back !!
Yes.
So, I did some iperf3 tests between my 920+ and my Asustor and see what I get:
```bash
192.168.2.201 est une adresse IP valide. Le test iperf3 va pouvoir s'exéctuer.
Connecting to host 192.168.2.201, port 5201
[  5] local 192.168.2.203 port 53348 connected to 192.168.2.201 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   283 MBytes  2.37 Gbits/sec    0    622 KBytes
[  5]   1.00-2.00   sec   280 MBytes  2.35 Gbits/sec    0    687 KBytes
[  5]   2.00-3.00   sec   279 MBytes  2.34 Gbits/sec    0    913 KBytes
[  5]   3.00-4.00   sec   275 MBytes  2.31 Gbits/sec    0   1.12 MBytes
[  5]   4.00-5.00   sec   280 MBytes  2.35 Gbits/sec    0   1.12 MBytes
[  5]   5.00-6.00   sec   280 MBytes  2.35 Gbits/sec    0   1.12 MBytes
[  5]   6.00-7.00   sec   281 MBytes  2.36 Gbits/sec    0   1.12 MBytes
[  5]   7.00-8.00   sec   280 MBytes  2.35 Gbits/sec    0   1.23 MBytes
[  5]   8.00-9.00   sec   279 MBytes  2.34 Gbits/sec    0   1.36 MBytes
[  5]   9.00-10.00  sec   281 MBytes  2.36 Gbits/sec    0   1.36 MBytes
[  5]  10.00-11.00  sec   275 MBytes  2.31 Gbits/sec    0   1.86 MBytes
[  5]  11.00-12.00  sec   281 MBytes  2.36 Gbits/sec    0   1.86 MBytes
[  5]  12.00-13.00  sec   279 MBytes  2.34 Gbits/sec    0   1.95 MBytes
[  5]  13.00-14.00  sec   280 MBytes  2.35 Gbits/sec    0   2.04 MBytes
[  5]  14.00-15.00  sec   281 MBytes  2.36 Gbits/sec    0   2.04 MBytes
[  5]  15.00-16.00  sec   280 MBytes  2.35 Gbits/sec    0   2.04 MBytes
[  5]  16.00-17.00  sec   280 MBytes  2.35 Gbits/sec    0   2.04 MBytes
[  5]  17.00-18.00  sec   281 MBytes  2.36 Gbits/sec    0   2.04 MBytes
[  5]  18.00-19.00  sec   279 MBytes  2.34 Gbits/sec    0   2.04 MBytes
[  5]  19.00-20.00  sec   280 MBytes  2.35 Gbits/sec    0   2.04 MBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-20.00  sec  5.46 GBytes  2.35 Gbits/sec    0             sender
[  5]   0.00-20.00  sec  5.46 GBytes  2.35 Gbits/sec                  receiver

iperf Done.
Connecting to host 192.168.2.201, port 5201
Reverse mode, remote host 192.168.2.201 is sending
[  5] local 192.168.2.203 port 53352 connected to 192.168.2.201 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec   277 MBytes  2.33 Gbits/sec
[  5]   1.00-2.00   sec   276 MBytes  2.32 Gbits/sec
[  5]   2.00-3.00   sec   281 MBytes  2.35 Gbits/sec
[  5]   3.00-4.00   sec   281 MBytes  2.35 Gbits/sec
[  5]   4.00-5.00   sec   279 MBytes  2.34 Gbits/sec
[  5]   5.00-6.00   sec   279 MBytes  2.34 Gbits/sec
[  5]   6.00-7.00   sec   280 MBytes  2.35 Gbits/sec
[  5]   7.00-8.00   sec   281 MBytes  2.35 Gbits/sec
[  5]   8.00-9.00   sec   279 MBytes  2.34 Gbits/sec
[  5]   9.00-10.00  sec   281 MBytes  2.35 Gbits/sec
[  5]  10.00-11.00  sec   281 MBytes  2.35 Gbits/sec
[  5]  11.00-12.00  sec   281 MBytes  2.35 Gbits/sec
[  5]  12.00-13.00  sec   281 MBytes  2.35 Gbits/sec
[  5]  13.00-14.00  sec   280 MBytes  2.35 Gbits/sec
[  5]  14.00-15.00  sec   280 MBytes  2.35 Gbits/sec
[  5]  15.00-16.00  sec   281 MBytes  2.35 Gbits/sec
[  5]  16.00-17.00  sec   281 MBytes  2.35 Gbits/sec
[  5]  17.00-18.00  sec   280 MBytes  2.35 Gbits/sec
[  5]  18.00-19.00  sec   281 MBytes  2.35 Gbits/sec
[  5]  19.00-20.00  sec   281 MBytes  2.35 Gbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-20.00  sec  5.47 GBytes  2.35 Gbits/sec    0             sender
[  5]   0.00-20.00  sec  5.47 GBytes  2.35 Gbits/sec                  receiver

iperf Done.
```

A nearly perfect score 😄  for both ways : up and down ! ! With no retry !

After this success, I try to activate again my static routes I had set before to ensure all traffic go through eth2 interface :
[image]
But this caused all bad retry to occurs again with the poor upload throughput...
As soon as I deactivated those statics routes rules, my upload throughput was back to full 2,5G speed.

So, for now I let thoses static routes behind, and seek a way to ensure all traffic goes though eth2 interface, while maintaining my eth0 interface up and running.

I just have to decide what to do with the usb hub : give it back to amazon, or keep it in case of I need to plug another usb thing in the 920+, because the other port is used by my UPS.

So for now, all is working fine !

bb-qq, 2023-04-22:
I am glad to hear that it worked successfully.

Is the problem now that throughput is affected when static routes are configured?
I am not sure about the configuration of your network, but what happens when you separate the segments?

MilesTEG1, 2023-04-25:
> Is the problem now that throughput is affected when static routes are configured?
> I am not sure about the configuration of your network, but what happens ***when you separate the segments?***

I'm not sure I correctly understood your questions here...

If static routes are set, the throughput is affected: low throughput.
What do you mean with "***separate the segments***"?

bb-qq, 2023-04-30:
I wrote the previous comment in the hope that by creating an independent subnet for the desired route, it would be possible to communicate over the desired route without explicitly assigning a static route.

Sorry if I'm missing the point.

bb-qq, 2024-06-24:
I would like to close this issue as I have not heard back from you in a long time.
If you have any questions, feel free to re-open.

closed 2024-06-24 by bb-qq