bb-qq/r8152 #297: Compatibility ds 123 ds 223

fabien67160, 2023-03-26:
Bonjour il faut prendre quelle pilote pour le Realtek RTD1619B ?

fabien67160, 2023-04-22:
Is there anyone who wants to answer?

bb-qq, 2023-04-22:
I created a driver for RTD1619B platforms.
https://github.com/bb-qq/r8152/releases/download/2.16.3-4/r8152-rtd1619b-2.16.3-4_7.1.spk

I hope you will try it and report back with your results.

angelgfoto, 2023-07-01:
hey bb-qq, what model of adapter is it specifically for, this driver?

TheS1R, 2023-07-01:
[https://github.com/bb-qq/r8152/wiki/Compatibility](url)

bb-qq, 2023-07-02:
It seems to have worked successfully, so I would close this issue.

After using it for a while and being able to evaluate its stability, I would be happy to have you report back to me again on this issue.
https://github.com/bb-qq/r8152/issues/135

closed 2023-07-02 by bb-qq