bb-qq/r8152 #298: plugin says error, also seems that  every restart the nic dissapears

evanrich, 2023-03-26:
### Description of the problem

[image].

Also, it seems after a restart the nic doesn't show up.   I saw a post here that involved like a `/etc/init.d/rtl8156 stop/start` command but will have to look for it, thats the only way this worked.

### Description of your products

DS620Slim
`Linux ImageStore 4.4.180+ #42962 SMP Fri Feb 10 16:32:29 CST 2023 x86_64 GNU/Linux synology_apollolake_620slim`
[image]

### Description of your environment

> - connect directly to a PC or using a Ethernet switch
>     - PC: NIC model name or chipset name
>     - Ethernet switch: model name
> - cable category (Cat.5e or Cat.6 or Cat.7)

### Output of ``dmesg`` command
[
````

[pastebin](https://pastebin.com/Y8DHJSA8)

### Output of ``lsusb`` command

````
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 650023C913EF6055)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-2         0bda:8156:3104 00  3.20 5000MBit/s 256mA 2IFs (Realtek USB 10/100/1G/2.5G LAN 401000001)

````

### Output of ``ifconfig -a`` command

````
bond0     Link encap:Ethernet  HWaddr 90:09:D0:17:B7:27
          inet addr:192.168.1.15  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::9209:d0ff:fe17:b727/64 Scope:Link
          UP BROADCAST RUNNING MASTER MULTICAST  MTU:1500  Metric:1
          RX packets:48379672 errors:0 dropped:0 overruns:0 frame:0
          TX packets:106233271 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:3228554885 (3.0 GiB)  TX bytes:216413183744 (201.5 GiB)

docker0   Link encap:Ethernet  HWaddr 02:42:85:CE:98:CB
          inet addr:172.17.0.1  Bcast:172.17.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

docker-54 Link encap:Ethernet  HWaddr 02:42:E8:55:5A:EB
          inet addr:172.18.0.1  Bcast:172.18.255.255  Mask:255.255.0.0
          inet6 addr: fe80::42:e8ff:fe55:5aeb/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:74188 errors:0 dropped:0 overruns:0 frame:0
          TX packets:74195 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:24200458 (23.0 MiB)  TX bytes:40347026 (38.4 MiB)

docker48c Link encap:Ethernet  HWaddr DA:BD:22:67:CA:9C
          inet6 addr: fe80::d8bd:22ff:fe67:ca9c/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:70104 errors:0 dropped:0 overruns:0 frame:0
          TX packets:70112 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:16480766 (15.7 MiB)  TX bytes:39881868 (38.0 MiB)

docker572 Link encap:Ethernet  HWaddr 2A:24:50:1E:2A:02
          inet6 addr: fe80::2824:50ff:fe1e:2a02/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:4084 errors:0 dropped:0 overruns:0 frame:0
          TX packets:4096 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:8758324 (8.3 MiB)  TX bytes:466256 (455.3 KiB)

eth0      Link encap:Ethernet  HWaddr 90:09:D0:17:B7:27
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:48344703 errors:0 dropped:0 overruns:0 frame:0
          TX packets:53672831 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:3226279746 (3.0 GiB)  TX bytes:110408515562 (102.8 GiB)
          Interrupt:115 base 0x4000

eth1      Link encap:Ethernet  HWaddr 90:09:D0:17:B7:28
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:34975 errors:0 dropped:0 overruns:0 frame:0
          TX packets:52560444 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:2275505 (2.1 MiB)  TX bytes:106004677062 (98.7 GiB)
          Interrupt:116

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:75223 errors:0 dropped:0 overruns:0 frame:0
          TX packets:75223 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:11018365 (10.5 MiB)  TX bytes:11018365 (10.5 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
````

evanrich, 2023-03-26:
to get it to work i have to run these commands
```
sudo /var/packages/r8152/scripts/start-stop-status stop
sudo /var/packages/r8152/scripts/start-stop-status start
```

evanrich, 2023-03-27:
> Install the package through the GUI again and it will be successful and the error will go away. I've not been able to get the package to autostart though following a reboot.

thank you! that solved the "error",  its a shame that scheduled tasks doesn't have a "on reboot" option like cron does.  Might be able to sudo to root and add it to crontab though on my own, or write a script to check the status of the interface and just add it to the scheduled tasks section.  I'll mess around with it for a bit.

evanrich, 2023-03-30:
> I confirm, my network interface also disappears after a reboot. DSM 7.1.1-42962 Update 4 (DS720+)

check my last update above for a workaround via cron.

ISnotes, 2023-03-31:
I can’t check, reinstalling the application after the first reboot helped me. But now I am faced with a new problem: if you pull out the network card and insert it again during operation, the interface turns off and goes into the DOWN state. I had to restart the app again. I would like to switch completely to a 2.5G network, I would not like to lose the ability to manage in one moment. Is it possible to fix it somehow @nmiller0113?

evanrich, 2023-04-03:
> I can’t check

You can't read the comment I posted above?  

> I would like to switch completely to a 2.5G network

You can.  Give the 2.5G adapter a static IP, and then just move all your mounts to use the IP of the 2.5G adapter.  You can still use the 1Gb interface for management.   I'm using the 1G for management tasks, and the 2.5G for all my data mounts, effectively using the 1G as an OOB interface only to log into the box.  This is how you would set up most servers anyway (management and data on different nics), so it works.   If you really need a single nic solution (not recommended), then you should get a synology with a dedicated 10G nic (or one that supports a 10G addon card)

ISnotes, 2023-04-03:
> a single nic solution (not recommended), then you

It's workaround

bb-qq, 2023-04-22:
Could you put the contents of the `/var/packages/r8152/var/log/start-stop-status.log` here?

evanrich, 2023-05-08:
```
++ grep /usb
+ initial_count=0
+ '[' -e /var/packages/r8152/target/r8152/mii.ko ']'
+ '[' -e /var/packages/r8152/target/r8152/usbnet.ko ']'
+ /sbin/insmod /var/packages/r8152/target/r8152/r8152.ko
+ '[' -r /usr/lib/udev/rules.d/51-usb-r8152-net.rules ']'
++ seq 1 60
+ for i in `seq 1 60`
++ count_usb_interfaces
++ ls -l /sys/class/net/
++ grep /usb
++ wc -l
+ '[' 1 -ne 0 ']'
+ break
+ set_interfaces up
+ action=up
+ target_interface_name=
++ ls /sys/class/net
+ all_interface_names='docker0
docker-54909108
docker904cc6a
docker9fe85f5
eth0
eth1
eth2
lo
sit0'
+ '[' -n ']'
+ '[' '!' -e /sys/class/net/ ']'
+ for interface_name in ${all_interface_names}
+ [[ -n '' ]]
+ '[' '!' -e /sys/class/net/docker0/device/driver ']'
+ continue
+ for interface_name in ${all_interface_names}
+ [[ -n '' ]]
+ '[' '!' -e /sys/class/net/docker-54909108/device/driver ']'
+ continue
+ for interface_name in ${all_interface_names}
+ [[ -n '' ]]
+ '[' '!' -e /sys/class/net/docker904cc6a/device/driver ']'
+ continue
+ for interface_name in ${all_interface_names}
+ [[ -n '' ]]
+ '[' '!' -e /sys/class/net/docker9fe85f5/device/driver ']'
+ continue
+ for interface_name in ${all_interface_names}
+ [[ -n '' ]]
+ '[' '!' -e /sys/class/net/eth0/device/driver ']'
++ ls -ld /sys/class/net/eth0/device/driver
+ driver_location='lrwxrwxrwx 1 root root 0 Apr 18 16:12 /sys/class/net/eth0/device/driver -> ../../../../bus/pci/drivers/r8168'
++ echo 'lrwxrwxrwx 1 root root 0 Apr 18 16:12 /sys/class/net/eth0/device/driver -> ../../../../bus/pci/drivers/r8168'
++ grep r8152
+ '[' '!' -z '' ']'
+ for interface_name in ${all_interface_names}
+ [[ -n '' ]]
+ '[' '!' -e /sys/class/net/eth1/device/driver ']'
++ ls -ld /sys/class/net/eth1/device/driver
+ driver_location='lrwxrwxrwx 1 root root 0 Apr 18 16:12 /sys/class/net/eth1/device/driver -> ../../../../bus/pci/drivers/r8168'
++ echo 'lrwxrwxrwx 1 root root 0 Apr 18 16:12 /sys/class/net/eth1/device/driver -> ../../../../bus/pci/drivers/r8168'
++ grep r8152
+ '[' '!' -z '' ']'
+ for interface_name in ${all_interface_names}
+ [[ -n '' ]]
+ '[' '!' -e /sys/class/net/eth2/device/driver ']'
++ ls -ld /sys/class/net/eth2/device/driver
+ driver_location='lrwxrwxrwx 1 root root 0 Apr 18 16:33 /sys/class/net/eth2/device/driver -> ../../../../../../bus/usb/drivers/r8152'
++ echo 'lrwxrwxrwx 1 root root 0 Apr 18 16:33 /sys/class/net/eth2/device/driver -> ../../../../../../bus/usb/drivers/r8152'
++ grep r8152
+ '[' '!' -z 'lrwxrwxrwx 1 root root 0 Apr 18 16:33 /sys/class/net/eth2/device/driver -> ../../../../../../bus/usb/drivers/r8152' ']'
+ config_file=/etc/sysconfig/network-scripts/ifcfg-eth2
+ config_storage_location=/var/packages/r8152/etc/ifcfg-eth2
+ '[' -f /etc/sysconfig/network-scripts/ifcfg-eth2 ']'
+ '[' up = down ']'
+ '[' up = up ']'
+ '[' -f /var/packages/r8152/etc/ifcfg-eth2 ']'
+ cp /var/packages/r8152/etc/ifcfg-eth2 /etc/sysconfig/network-scripts/ifcfg-eth2
+ ifconfig eth2 up
+ set_ovs_interface eth2 up
+ interface_name=eth2
+ action=up
+ config_file_ovs=/etc/sysconfig/network-scripts/ifcfg-ovs_eth2
+ config_storage_location_ovs=/var/packages/r8152/etc/ifcfg-ovs_eth2
+ is_ovs_enable
+ '[' '!' -f /usr/syno/etc/synoovs/ovs_reg.conf ']'
+ return 1
+ '[' up = up ']'
+ is_ovs_enable
+ '[' '!' -f /usr/syno/etc/synoovs/ovs_reg.conf ']'
+ return 1
+ '[' -r /etc/sysconfig/network-scripts/ifcfg-eth2 ']'
+ source /etc/sysconfig/network-scripts/ifcfg-eth2
++ DEVICE=eth2
++ BOOTPROTO=static
++ ONBOOT=yes
++ IPADDR=192.168.50.50
++ NETMASK=255.255.255.248
+ apply_if_config eth2
+ target_interface_name=eth2
+ [[ ! -v BOOTPROTO ]]
+ '[' static = dhcp ']'
+ source /etc/VERSION
++ majorversion=7
++ minorversion=1
++ major=7
++ minor=1
++ micro=1
++ productversion=7.1.1
++ buildphase=GM
++ buildnumber=42962
++ smallfixnumber=5
++ nano=5
++ base=42962
++ builddate=2023/04/08
++ buildtime=20:01:57
+ '[' 42962 -ge 42962 ']'
+ [[ -v IPADDR ]]
+ [[ -v NETMASK ]]
+ ifconfig eth2 inet 192.168.50.50 netmask 255.255.255.248
+ netstat -nr
Kernel IP routing table
Destination     Gateway         Genmask         Flags   MSS Window  irtt Iface
0.0.0.0         192.168.1.1     0.0.0.0         UG        0 0          0 eth0
169.254.0.0     0.0.0.0         255.255.0.0     U         0 0          0 eth1
172.17.0.0      0.0.0.0         255.255.0.0     U         0 0          0 docker0
172.18.0.0      0.0.0.0         255.255.0.0     U         0 0          0 docker-54909108
192.168.1.0     0.0.0.0         255.255.255.0   U         0 0          0 eth0
192.168.50.48   0.0.0.0         255.255.255.248 U         0 0          0 eth2
+ ip route
+ grep -E '^default'
default via 192.168.1.1 dev eth0  src 192.168.1.15
+ '[' up = up ']'
+ '[' -e /var/packages/r8152/scripts/apply-private-flags ']'
+ for interface_name in ${all_interface_names}
+ [[ -n eth2 ]]
+ [[ lo != \e\t\h\2 ]]
+ continue
+ for interface_name in ${all_interface_names}
+ [[ -n eth2 ]]
+ [[ sit0 != \e\t\h\2 ]]
+ continue
+ exit 0
```

bb-qq, 2023-05-13:
Since the log ends with exit 0, the process appears to have completed successfully. If I had the entire log, I might be able to investigate further.

XingXing-404, 2023-05-16:
I am very sorry that I can only communicate with you through translation software.
I had a similar problem, I tried reinstalling the system and it still didn't work. Every time I start the NAS, I have to go to the package center and click repair to use it, or add a task "/var/packages/r8152/scripts/start-stop-status start" to the scheduled task. Here is my error log.
```
+ date
Tue May 16 16:59:39 CST 2023
+ grep -E '^(version|create_time)' /var/packages/r8152/INFO
version="2.16.3-4"
create_time="20230319-14:37:13"
+ echo start
start
+ case $1 in
+ '[' -e /var/packages/r8152/scripts/apply-memory-setting ']'
+ '[' -w /sys/module/usbcore/parameters/autosuspend ']'
+ echo -1
++ count_usb_interfaces
++ ls -l /sys/class/net/
++ grep /usb
++ wc -l
+ initial_count=1
+ '[' -e /volume3/@appstore/r8152/r8152/mii.ko ']'
+ '[' -e /volume3/@appstore/r8152/r8152/usbnet.ko ']'
+ /sbin/insmod /volume3/@appstore/r8152/r8152/r8152.ko
insmod: ERROR: could not insert module /volume3/@appstore/r8152/r8152/r8152.ko: File exists

```

Crazy-FuQing, 2023-05-24:
I also encountered a similar problem. After installing the apk, the usb network card was successfully recognized, but the network card could not be found after the group restarted, and the reinstallation of the spk did not solve the problem.
Below is my log file

xzqiaochu, 2023-06-21:
I am the same problem with @XingXing-404 (need click repair after rebooting) and @ISnotes (DOWN after reinserting).
Version is DSM 7.1.1-42962 Update 5. 
I use 2.16.3-4

xzqiaochu, 2023-06-21:
> I also encountered a similar problem. After installing the apk, the usb network card was successfully recognized, but the network card could not be found after the group restarted, and the reinstallation of the spk did not solve the problem. Below is my log file /var/packages/r8152/var/log/start-stop-status.log [start-stop-status.log](https://github.com/bb-qq/r8152/files/11551326/start-stop-status.log)

Are you using ARPL? I encountered a similar problem and resolved it by disabling cdc_ncm. As a result, usb0 was changed to eth1. This article might be useful: https://www.mi-d.cn/5556

bb-qq, 2023-07-02:
Indeed, using my driver with other kernel modules may cause problems. I can investigate if I have the dmesg logs.

n3o4po11o, 2023-07-30:
> Indeed, using my driver with other kernel modules may cause problems. I can investigate if I have the dmesg logs.

because ARPL enable r8152 module by default, this script `/var/packages/r8152/scripts/start-stop-status` will insmod "r8125" kernel module, but system already load this module, the  /var/packages/r8152/var/log/start-stop-status.log will said,

`insmod: ERROR: could not insert module /var/packages/r8152/target/r8152/r8152.ko: File exists`
as #234 
and then this script `/var/packages/r8152/scripts/start-stop-status` will exit 1 and execue `shutdown` interface. so will show need repair, and rmmod r8152 kernel module.

I am using ARPL,  I uninstall r8152.spk and do nothing. everthing works by default.

keepworld, 2024-09-07:
Adding a sleep in the crontab fix the issues :
`@reboot root sleep 120 && /var/packages/r8152/scripts/start-stop-status stop && /var/packages/r8152/scripts/start-stop-status start`

bb-qq, 2024-09-22:
If it is a timing issue, enabling hot plugging may help.
https://github.com/bb-qq/r8152/wiki/Troubleshooting#hot-plugging-does-not-work

bb-qq, 2024-09-22:
I added a tip about ARPL.
https://github.com/bb-qq/r8152/wiki/Troubleshooting#for-arpl-users

closed 2025-01-02 by bb-qq