bb-qq/r8152 #299: How to completely uninstall the driver I order to reinstall from scratch ?

MilesTEG1, 2023-04-05:
Hello,
I want to try an installation from scratch but in order to do this , I need to completely remove the driver and all of it’s parameters and files.
But I don’t know if I uninstall the package from dsm it will remove everything concerning the driver.

is there a way to achieve a complete uninstall of the driver ?

thanks

MilesTEG1, 2023-04-05:
Hello,
As I didn't receive an answer, here what I did :

1. Execute `find / -iname "r8152"` to search all files/folders related to the r8152 driver.
2. Uninstall from DSM the package.
3. Execute again `find / -iname "r8152"` to search all files/folders related to the r8152 driver after uninstalling it.
4. Delete all folders and files found with `find` command, and for me this was :
  ```bash
  /volume1/@appdata/r8152
  /volume1/@apphome/r8152
  /volume1/@appconf/r8152
  /volume1/@userpreference/r8152
  /usr/syno/etc/packages/r8152
  /usr/syno/etc/packages/SecureSignIn/preference/r8152
```
5. reboot the NAS
6. Reinstall the package

bb-qq, 2023-04-22:
I think that procedure is fine.
The driver reads the settings saved under `@appconf/r8152`, so it may be sufficient to just delete this.

bb-qq, 2023-07-16:
I would like to close this issue since the question seems to have been resolved.

closed 2023-07-16 by bb-qq