bb-qq/r8152 #300: Two pieces of Club 3D CAC-1420 in one NAS - The 2nd CAC-1420 doesn't start automatically and loses its IP configuration

hy-ts, 2023-04-06:
### Description of the problem
I've two pieces of Club 3D CAC-1420 in my NAS: the first plugged into the front USB port (assigned to LAN 3 in DSM), the second into the rear USB port (assigned to LAN 4 in DSM).
After restarting the NAS, the LAN 4 in DSM (2nd CAC-1420 at the rear USB port) doesn't start automatically and loses its IP configurations. I made in Task Scheduler a new task (named: LAN 4 Start): User: root; Event: System boot; User defined script: sudo ip link set eth3 up. This script starts the LAN 4 (eth3) at boot. But in DSM at Network interface settings some manual configuration settings are lost: IP address, Subnet mask, Gateway, DNS server. (MTU=9000 is retained.)
Now I'm looking for automatically setting these configs in Task Scheduler, too. (Not yet succeded, but I think, it must be simple. But I'm not a Linux guru, only a simple user.)

So it would be nice to have these IP settings retained in case of the second CAC-1420 after every NAS reboot.

And thank you very much for your work, because these two adapters are working well, when I do manual IP settings in case of the second adapter.

### Description of your products
Model(platform): DS918+ (apollolake)
DSM version: 7.2-64216
Driver version: 2.16.3-4 (r8152-apollolake-2.16.3-4_7.2.spk)
Ethernet device: Club 3D CAC-1420 - Two pieces of them in one NAS
Working status: Only one CAC-1420: Working;  Two of CAC-1420: Partly working
Maximum Throughput: Not yet tested

Output of ``uname -a`` command:
Linux fileserver 4.4.302+ #64216 SMP Tue Mar 7 17:11:43 CST 2023 x86_64 GNU/Linux synology_apollolake_918+

### Description of your environment
PC: Under change, but now:  1 piece of Intel I225-V (2.5Gbit) and a I350-T2V2 card with 2 ports of 1 Gbit
Ethernet switch: Zyxel MG-108
Cable category Cat.7

### Output of ``dmesg`` command
[dmesg.txt](https://github.com/bb-qq/r8152/files/11170145/dmesg.txt)

### Output of ``lsusb`` command
/$ lsusb
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F008C5A16DA5A47)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-1         0bda:8156:3100 00  3.20 5000MBit/s 256mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 001000001)
  |__2-2         0bda:8156:3100 00  3.20 5000MBit/s 256mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 001000000)

### Output of ``ifconfig -a`` command
(I deleted "inet6 addr" datas)
/$ ifconfig -a
eth0      Link encap:Ethernet  HWaddr 00:11:32:A1:33:E9
          inet addr:192.168.0.13  Bcast:192.168.0.255  Mask:255.255.255.0
          inet6 addr: 
          inet6 addr: 
          inet6 addr: 
          inet6 addr: 
          UP BROADCAST RUNNING MULTICAST  MTU:9000  Metric:1
          RX packets:73460 errors:0 dropped:0 overruns:0 frame:0
          TX packets:42518 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:22818311 (21.7 MiB)  TX bytes:25652829 (24.4 MiB)

eth1      Link encap:Ethernet  HWaddr 00:11:32:A1:33:EA
          inet addr:192.168.0.14  Bcast:192.168.0.255  Mask:255.255.255.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth2      Link encap:Ethernet  HWaddr A0:CE:C8:F9:28:EE
          inet addr:192.168.0.11  Bcast:192.168.0.255  Mask:255.255.255.0
          inet6 addr: 
          inet6 addr: 
          inet6 addr: 
          inet6 addr: 
          UP BROADCAST RUNNING MULTICAST  MTU:9000  Metric:1
          RX packets:384123 errors:0 dropped:0 overruns:0 frame:0
          TX packets:918318 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:41468238 (39.5 MiB)  TX bytes:1130398710 (1.0 GiB)

eth3      Link encap:Ethernet  HWaddr A0:CE:C8:F9:26:38
          inet addr:192.168.0.12  Bcast:192.168.0.255  Mask:255.255.255.0
          inet6 addr: 
          inet6 addr: 
          inet6 addr: 
          inet6 addr: 
          UP BROADCAST RUNNING MULTICAST  MTU:9000  Metric:1
          RX packets:37374 errors:0 dropped:0 overruns:0 frame:0
          TX packets:5526 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:4227950 (4.0 MiB)  TX bytes:1155755 (1.1 MiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:28679 errors:0 dropped:0 overruns:0 frame:0
          TX packets:28679 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:4066346 (3.8 MiB)  TX bytes:4066346 (3.8 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

tun0      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00
          inet addr:10.8.0.1  P-t-P:10.8.0.2  Mask:255.255.255.255
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:500
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

Bafti71, 2023-04-20:
@hy-ts 
Just like you, I want to use 2 x Club 3D CAC-1420 in my NAS. Unfortunately, my Synology only shows ONE of the connected USB-Adapters unter "Control Panel - Network". 

The first plugged into the front USB port, the second into the rear USB port.

How can I make both Club 3D CAC-1420 work? I have tried uninstalling/reinstalling the driver. Reboot. Still only shows 1 Adapter.

Can you help?

**Description of your products**

Model(platform): DS1817+ (avoton)
DSM version: DSM 7.1.1-42962 Update 5
Driver version: r8152-avoton-2.16.3-4_7.1.spk
Ethernet device: Club 3D CAC-1420 - Two pieces of them in one NAS

**Output of lsusb command:**

|__usb1          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd 0000:00:16.0) hub
  |__1-1         8087:07db:0003 09  2.00  480MBit/s 0mA 1IF  ( ffffffd1ffffffb2ffffffdbffffffad) hub
    |__1-1.1     f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 65007A6F146DAE60)
|__usb2          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (Linux 3.10.108 etxhci_hcd-170202 Etron xHCI Host Controller 0000:09:00.0) hub
  |__2-3         0463:ffff:0001 00  1.10  1.5MBit/s 20mA 1IF  (EATON 5S ffffffe7ffffffd2)
|__usb3          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (Linux 3.10.108 etxhci_hcd-170202 Etron xHCI Host Controller 0000:09:00.0) hub
  |__3-1         0bda:8156:3104 00  3.20 5000MBit/s 256mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 401000001)
  |__3-4         0bda:8151:3104 00  3.20 5000MBit/s 256mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 401000000)

**Output of ifconfig -a command**

bond0     Link encap:Ethernet  HWaddr 00:11:32:81:5C:96
          inet addr:192.168.10.3  Bcast:192.168.10.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MASTER MULTICAST  MTU:1500  Metric:1
          RX packets:10709 errors:0 dropped:0 overruns:0 frame:0
          TX packets:3740 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:2211192 (2.1 MiB)  TX bytes:2547892 (2.4 MiB)

eth0      Link encap:Ethernet  HWaddr 00:11:32:81:5C:96
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:2380 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1172 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:291815 (284.9 KiB)  TX bytes:141523 (138.2 KiB)

eth1      Link encap:Ethernet  HWaddr 00:11:32:81:5C:96
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:1246 errors:0 dropped:0 overruns:0 frame:0
          TX packets:2482 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:165276 (161.4 KiB)  TX bytes:2395869 (2.2 MiB)

eth2      Link encap:Ethernet  HWaddr 00:11:32:81:5C:96
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:1332 errors:0 dropped:0 overruns:0 frame:0
          TX packets:44 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:192436 (187.9 KiB)  TX bytes:5292 (5.1 KiB)

eth3      Link encap:Ethernet  HWaddr 00:11:32:81:5C:96
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:5751 errors:0 dropped:0 overruns:0 frame:0
          TX packets:42 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:1561665 (1.4 MiB)  TX bytes:5208 (5.0 KiB)

eth4      Link encap:Ethernet  HWaddr A0:CE:C8:6E:79:5B
          inet addr:192.168.11.2  Bcast:192.168.11.7  Mask:255.255.255.248
          inet6 addr: fe80::a2ce:c8ff:fe6e:795b/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:2012 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1638 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:247496 (241.6 KiB)  TX bytes:1091043 (1.0 MiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:2282 errors:0 dropped:0 overruns:0 frame:0
          TX packets:2282 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:268742 (262.4 KiB)  TX bytes:268742 (262.4 KiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

hy-ts, 2023-04-20:
@Bafti71 
Please post the output of the "ip addr" command, especially the values of the eth4 and eth5 ports. I hope you have the eth5 LAN port.

hy-ts, 2023-04-20:
@Bafti71 
From the output of the "lsusb" command we can see, that both CAC-1420 devices are recognized at the hardware level:
"|__3-1 0bda:8156:3104 00 3.20 5000MBit/s 256mA 1IF (Realtek USB 10/100/1G/2.5G LAN 401000001)
 |__3-4 0bda:8151:3104 00 3.20 5000MBit/s 256mA 1IF (Realtek USB 10/100/1G/2.5G LAN 401000000)"

But it seems to me, that the driver is not loading for the 2nd CAC-1420. So I think the task is to start the driver for the second CAC-1420.
Unfortunately I don't know the solution, but I think, the problem can be solved.
It may also be that the system memory is low (if it only has 2GB of RAM).
I would try the following:
- As a test, I would unplug all other USB devices if there are any (except the two CAC-1420s).
- I would transfer the CAC-1420 from the rear USB port to another rear one.
- I would delete the 4 network bonds (LAN1-LAN4): this would free up a (virtual) network connection. Maybe this might allow the second CAC-1420. Of course, canceling the bond can cause quite a few problems: e.g. DHCP server, binding other services to your LAN IP address, etc.
- And finally, if you upgrade to DSM 7.2, I would still try to see if the two CAC-1420s work.

Bafti71, 2023-04-20:
@hy-ts 
Thank you very much for your answers and suggestions.

- My Syno has 8MB of RAM
- I unplugged the other USB-Devices --> no success
- I tried different USB-Portconfigurations --> no success
- I deleted the Bond --> no success

When I stop the r8152 driver, the Syno sees both USB devices in the Info-Center. But they don't appear in the Network-Center. When I start the driver, only one USB-Adapter is listed in the Info-Center.

Locks like it's not ready to work with 2 similar USB-LAN-Adapters on my Syno.

hy-ts, 2023-04-20:
@Bafti71 
> When I stop the r8152 driver, the Syno sees both USB devices in the Info-Center. But they don't appear in the Network-Center. When I start the driver, only one USB-Adapter is listed in the Info-Center.

This behaviour is very interesting. I think that "bb-qq", the maker of the driver might be able to answer this question.

Bafti71, 2023-04-20:
@hy-ts 
I found another thread where this problem has been discussed earlier:
#140 
bb-qq is informed and I am curious to find out what the next steps and solutions will be.

hy-ts, 2023-04-20:
@Bafti71 
Well, I never thought that products from different manufacturers could be a solution.
https://github.com/bb-qq/r8152/issues/140#issuecomment-1407640731
Please let me know here, if there is a solution to your problem.

bb-qq, 2023-04-22:
Could you try what happens when you enable the HotPlug feature?
This feature does not handle multiple adapters properly, but the result may work fine.

(Note: This command modifies your system outside of package management.)
```
sudo bash /var/packages/r8152/scripts/install-udev-rules
```

Bafti71, 2023-04-22:
@bb-qq 
I have tried this command. Unfortunately it doesn't change anything.

hy-ts, 2023-04-22:
@bb-qq 
For me, the above command (HotPlug feature) works perfectly! The two CAC-1420s work well at the same time.

The steps I took:
1. I ran the command.
2. I deactivated my "Start LAN 4" script in the DSM Task Manager.
3. I restarted DS918+.
4. Everything works perfectly: In the "DSM / Network interface" menu, all 4 of my LANs are visible and all of them retain their network settings. Both CAC-1420 ("USB 10/100/1G/2.5G LAN - Realtek semiconductor Corp.") are listed in the DSM Information Center.

So my original issue, which I described in the opening comment of this particular thread, has been solved! Thank you!
I'm curious to see if everything will still work well after next week's DSM 7.2 RC update.

hy-ts, 2023-04-25:
@hy-ts 
**Using new DSM 7.2RC (7.2-64551):**
File copy speed tests with file size of 104452.6 MB (MByte);
NAS-to-PC: 104452.6 MB / 222 s = 470.5 MB/s
PC-to-NAS: 104452.6 MB / 245 s = 426.3 MB/s

bb-qq, 2023-04-30:
Thank you for sharing your results. It is interesting to note that the results are slightly better than with a single 5 Gbps adapter.

I will close this issue as the problem seems to be resolved.

I would be happy to share the summary in this thread as well.
https://github.com/bb-qq/r8152/issues/135

closed 2023-04-30 by bb-qq

hy-ts, 2023-07-05:
@bb-qq 
Just FYI: there is a new generic Linux USB driver (version 2.17.1) on Realtek's website.
[https://www.realtek.com/en/component/zoo/category/network-interface-controllers-10-100-1000m-gigabit-ethernet-usb-3-0-software](url)