bb-qq/r8152 #301: Question : How to activate Wake-on-Lan for the interface created with the adaptor ?

MilesTEG1, 2023-04-07:
Hello,
I'm looking for a way to activate the Wake-on-Lan for the interface created with the adaptor, in my case it's foe eth2 (=LAN3 in DSM).
Because on DSM, the only interface for witch I can activate the WOL are LAN1 & LAN2:
[image]
LAN3 isn't here...

So, how can I activate the WOL on the LAN3 interface ?

Thanks

bb-qq, 2023-04-22:
The WOL function of the device can be enabled with the following command However, it seems that Synology's BIOS does not support wake-up from USB, so it does not actually work. (See #178)
```
sudo ethtool -s eth2 wol g
```

MilesTEG1, 2023-04-25:
Hello,
The command you gave me here doesn't return an error, or any message at all...
But see this:
```shell
root@Syno-DS920Plus:~$ ethtool eth2
Settings for eth2:
	Supported ports: [ MII ]
	Supported link modes:   10baseT/Half 10baseT/Full
	                        100baseT/Half 100baseT/Full
	                        1000baseT/Full
	                        2500baseX/Full
	Supported pause frame use: No
	Supports auto-negotiation: Yes
	Supported FEC modes: Not reported
	Advertised link modes:  10baseT/Half 10baseT/Full
	                        100baseT/Half 100baseT/Full
	                        1000baseT/Full
	                        2500baseX/Full
	Advertised pause frame use: Symmetric Receive-only
	Advertised auto-negotiation: Yes
	Advertised FEC modes: Not reported
	Link partner advertised link modes:  10baseT/Half 10baseT/Full
	                                     100baseT/Half 100baseT/Full
	                                     1000baseT/Full
	Link partner advertised pause frame use: No
	Link partner advertised auto-negotiation: Yes
	Link partner advertised FEC modes: Not reported
	Speed: 2500Mb/s
	Duplex: Full
	Port: MII
	PHYAD: 32
	Transceiver: internal
	Auto-negotiation: on
Cannot get wake-on-lan settings: Operation not permitted
	Current message level: 0x00007fff (32767)
			       drv probe link timer ifdown ifup rx_err tx_err tx_queued intr tx_done rx_status pktdata hw wol
	Link detected: yes
```

It seems WoL isn't possible, you're right.

bb-qq, 2023-04-30:
Unfortunately, there is no way to enable it, so this issue is closed.

closed 2023-04-30 by bb-qq