bb-qq/r8152 #303: How to install readme.md needs a step 7

AlexTVFP, 2023-04-25:
### Description of the problem

> In the section "How to install" on the page https://github.com/bb-qq/r8152 there are 6 steps in the section "Installation". I actually had to do a 7th step to make it run. I had to go in the UI , go in the Package Center - Installed pannel , look up the RTL8152/RTL8153 driver installed package, and click "run". After that Trendnet adaptor was visible in the network cards, not before.

It's working now, but I did had to search like a maniac :P Maybe it is because of the Update 5? I do not know... or is it suppose to auto-run? I had the assumption that since it was a driver instead of an actual software that it would auto-start.

### Description of your products

> - DS920+ 
>     - DSM 7.1.1-42962 Update 5
> - TUC-ET2G(v2.0R) - TRENDnet

### Description of your environment

> - Ethernet switch
>     - PC: Realtek PCIe 2.5GbE Family Controller
>     - Ethernet switch: QNAP QSW-1105-5T
> - cable category Cat.5e

### Output of ``dmesg`` command

````
if you really need it, let me know.
````

### Output of ``lsusb`` command

````
if you really need it, let me know.
````

### Output of ``ifconfig -a`` command

````
if you really need it, let me know.
````

TheS1R, 2023-04-25:
The driver is supposed to auto-run by design.

AlexTVFP, 2023-04-27:
Currently Yes, so let's call it a day ;)

bb-qq, 2023-07-15:
Since there seems to be no problem, I close this issue.

closed 2023-07-15 by bb-qq