bb-qq/r8152 #304: DS720+ Version: 7.1.1-42962 Update 5 Driver could not be started

motorradbill, 2023-04-25:
### Description of the problem

DS720+ Version: 7.1.1-42962 Update 5 Driver could not be started r8152-geminilake-2.16.3-4_7.2.spk

### Description of your products

> DS720+
>    admin@DISKSTATION-1:~$ uname -a
Linux DISKSTATION-1 4.4.180+ #42962 SMP Sat Apr 8 00:14:24 CST 2023 x86_64 GNU/Linux synology_geminilake_720+
>     - DSM Version: 7.1.1-42962 Update 5
> - Trendnet TUC-ET2G Ethernet Adapter USB-C 3.1 to 2.5Gbase-T

### Description of your environment

> - connect  using a Ethernet switch
>     - PC: NIC WIFI
>     - Ethernet Trendnet
> - cable category Cat. 6

### Output of ``dmesg`` command

````
(paste output here, or attach as a file)
[  179.931188] r8152: version magic '4.4.302+ SMP mod_unload ' should be '4.4.180+ SMP mod_unload '
[  287.868041] r8152: version magic '4.4.302+ SMP mod_unload ' should be '4.4.180+ SMP mod_unload '

### Output of ``lsusb`` command

````
admin@DISKSTATION-1:~$ lsusb
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 65002440F1EF6043)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-2         20f4:e02b:3004 00  3.20 5000MBit/s 512mA 2IFs (Realtek USB 10/100/1G/2.5G LAN 400000001)
````

### Output of ``ifconfig -a`` command

````
admin@DISKSTATION-1:~$ ifconfig -a
bond0     Link encap:Ethernet  HWaddr 90:09:D0:1F:0D:15
          inet addr:192.168.97.200  Bcast:192.168.97.255  Mask:255.255.255.0
          inet6 addr: fe80::9209:d0ff:fe1f:d15/64 Scope:Link
          inet6 addr: 2002:5901:bfbe:0:9209:d0ff:fe1f:d15/64 Scope:Global
          UP BROADCAST RUNNING MASTER MULTICAST  MTU:1500  Metric:1
          RX packets:18146 errors:0 dropped:0 overruns:0 frame:0
          TX packets:13099 errors:0 dropped:27 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:6594546 (6.2 MiB)  TX bytes:6520266 (6.2 MiB)

eth0      Link encap:Ethernet  HWaddr 90:09:D0:1F:0D:15
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:9083 errors:0 dropped:0 overruns:0 frame:0
          TX packets:6366 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:3589875 (3.4 MiB)  TX bytes:1011450 (987.7 KiB)
          Interrupt:98 base 0xc000

eth1      Link encap:Ethernet  HWaddr 90:09:D0:1F:0D:15
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:9063 errors:0 dropped:0 overruns:0 frame:0
          TX packets:6733 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:3004671 (2.8 MiB)  TX bytes:5508816 (5.2 MiB)
          Interrupt:97 base 0x8000

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:4217 errors:0 dropped:0 overruns:0 frame:0
          TX packets:4217 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:1117382 (1.0 MiB)  TX bytes:1117382 (1.0 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
````

hy-ts, 2023-04-25:
@motorradbill 
The driver "r8152-geminilake-2.16.3-4_**_7.2_**.spk" is for DSM 7.2, but you are using DSM 7.1.1. So I would remove this driver, and install "r8152-geminilake-2.16.3-4_**_7.1_**.spk" version.

motorradbill, 2023-04-25:
THX

I got this fixed with r8152-geminilake-2.16.3-4_7.1.spk

 

greetz

closed 2023-04-25 by motorradbill