bb-qq/r8152 #305: [Works] DS716+II with ULT-WIIQ USB 2.5Gb

phires, 2023-05-04:
### Description of the problem

Works on my DS716+II with ULT-WIIQ USB 2.5Gb (Rear USB)

### Description of your products

- DS716+II
  - ``uname -a``: ``Linux synology 3.10.108 #42962 SMP Fri Mar 24 12:30:20 CST 2023 x86_64 GNU/Linux synology_braswell_716+II``
  - DSM 7.1.1-42962 Update 5
- ULT-WIIQ USB 2.5Gb
https://www.amazon.de/dp/B09GBFZ9DZ

### Description of your environment

Dongle is connected to the Back USB port. Network connection using a CAT6a cable to a Zyxel XGS1250-12 syncs with 2.5 Gbps.

Sjekke, 2023-05-04:
On Amazon users saying that this adapter is getting hot? True?

phires, 2023-05-05:
I'm running it in my basement, where it is not warm at all. The adapter is pretty cold, not even hand-warm.
But - and I'm not yet sure, if it is my cable installation or the adapter - I'm only getting about 1.4 Gbps with iperf3:

```
$ iperf3 -c 192.168.1.200
Connecting to host 192.168.1.200, port 5201
[  5] local 192.168.1.47 port 58942 connected to 192.168.1.200 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   181 MBytes  1.52 Gbits/sec  912    133 KBytes
[  5]   1.00-2.00   sec   172 MBytes  1.45 Gbits/sec  784    130 KBytes
[  5]   2.00-3.00   sec   172 MBytes  1.45 Gbits/sec  946    117 KBytes
[  5]   3.00-4.00   sec   174 MBytes  1.46 Gbits/sec  804    141 KBytes
[  5]   4.00-5.00   sec   175 MBytes  1.47 Gbits/sec  898    143 KBytes
[  5]   5.00-6.00   sec   164 MBytes  1.37 Gbits/sec  908    107 KBytes
[  5]   6.00-7.00   sec   174 MBytes  1.46 Gbits/sec  980    113 KBytes
[  5]   7.00-8.00   sec   169 MBytes  1.42 Gbits/sec  993    136 KBytes
[  5]   8.00-9.00   sec   162 MBytes  1.36 Gbits/sec  834    106 KBytes
[  5]   9.00-10.00  sec   170 MBytes  1.43 Gbits/sec  898    133 KBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  1.67 GBytes  1.44 Gbits/sec  8957             sender
[  5]   0.00-10.00  sec  1.67 GBytes  1.43 Gbits/sec                  receiver
```

I will investigate further.

Sjekke, 2023-05-05:
Mmm ... the speed should higher ...

bb-qq, 2023-05-13:
From looking at the iperf results, it appears that a lot of retransmissions are occurring. (see the `Retr` column in the results).

It could be a MTU configuration issue or a physical layer issue such as a cable.

phires, 2023-05-15:
Yes, my cabeling is sub-par. Won't be able to change anything about this currently. But the adapter is still working well.

Barlos, 2023-05-27:
I had to put 2.5gb USB NIC to the front USB port of my DS716+II.

Ethernet device: Diewu RTL815B (manufactured date 2020)
Working status: Worked.

Speed test software: OpenSpeedTest - on Docker
Setting:
DS716+II (16GB RAM with 3.5" 4TB *2, Jumbo Frame MTU: 9000) <<<===>>> Diewu RTL815B <<<== direct RJ-45 cable connection ==>>> Diewu RTL815B <<<===>>> PC (Jumbo frame: 9014)

Download: 2,073.4 Mbps
Upload: 2,429.9 Mbps
PING: 3ms
JITTER: 0.2ms

VAllens, 2024-10-25:
> I had to put 2.5gb USB NIC to the front USB port of my DS716+II.
>
> Ethernet device: Diewu RTL815B (manufactured date 2020) Working status: Worked.
>
> Speed test software: OpenSpeedTest - on Docker [...]

Wow... man. @Barlos 
I have the same model DS716+II, but it only has 8GB of RAM from a combination of two memory sticks.
I would like to inquire what is the brand of your 16GB memory stick?
Is it a two memory stick combo?
I'm hesitant to upgrade to 16GB because I'm worried about compatibility.