bb-qq/r8152 #306: DS720+ High Availability

NebzHB, 2023-05-18:
Hi,

I'm wanting to use external NIC as heartbeat interface in an High Availability Synology cluster.

It seems when a node isn't used, packages are unloaded... so the driver...

do you have an idea to keep it loaded ? have someone done the same ?

thank you in advance

bb-qq, 2023-07-02:
Since the driver does not usually unload on its own, there appears to be a driver stability issue.
If you have the dmesg output, we might be able to find out something more.

NebzHB, 2023-07-02:
Sorry I've sent back the devices to Amazon and I'm using the build in only

bb-qq, 2023-07-02:
Okay, then I can't investigate any further, so I'm closing this Issue.

closed 2023-07-02 by bb-qq

lazosweb, 2024-11-22:
Hi @bb-qq,

I have the same "issue" as the OP. It seems that during the HA switch over the packages/services are turned off from one server to another because when I switch over back to the active server which has your package the LAN network will be available again.

Another thing that might also help is that I have also another "Synology" with the Arc Loader which loads the r8152 drivers directly therefore your package is not needed. During the switch over the LAN will disappear also.

What I believe is that during the switch over Synology unbinds/ejects any attached USB devices. I do not know if there is any way to overcome that behaviour.

bb-qq, 2024-11-23:
I don't know the details, but if USB device rebinding is happening, activating the hot plug feature may help.
https://github.com/bb-qq/r8152/wiki/Troubleshooting#hot-plugging-does-not-work

lazosweb, 2024-11-23:
> I don't know the details, but if USB device rebinding is happening, activating the hot plug feature may help. https://github.com/bb-qq/r8152/wiki/Troubleshooting#hot-plugging-does-not-work

@bb-qq I do not think the problem is just hot plugging but because the package is also stopped during the switch over.

bb-qq, 2024-11-23:
>the package is also stopped during the switch over.

Ok, understand. In such a situation, it seems that the driver needs to be loaded outside of the package system. 

Please try the following steps:

- Stop the r8152 package without uninstalling it.
- Load the driver module by executing the following command
````
   sudo /sbin/insmod /var/packages/r8152/target/r8152/r8152.ko
````
If this works, you may want to register the `insmod` command in the task scheduler.

lazosweb, 2024-11-23:
> Ok, understand. In such a situation, it seems that the driver needs to be loaded outside of the package system.
>
> Please try the following steps:
>
> * Stop the r8152 package without uninstalling it.
> * [...]

It worked. During the switchover LAN 3 is still there now on the Original Synology.
Now I have to figure out what is going on the ARC Loader one because is doing other stuff.
I will investigate and back to you if I have to.