bb-qq/r8152 #311: DS918+ and CableCreation 2.5Gbps adapter and DSM 7.1.1-42962 Update 5 - driver doesn't seem to install

hetzjagd, 2023-06-06:
### Description of the problem

Everything seems to happen as per the installation steps provided by the devs, installing from the .spk file gives a fail error, running the ssh command seems to be accepted happily, then if I understand the instructions I attempted installing from .spk file again before restarting. Upon restarting the driver still shows as stopped in the list of Packages and asks to be repaired.

I have confirmed the usb adapter works when plugged into a Windows 10 PC.
Front and back USB ports get the same result.

Stacks more info here https://github.com/bb-qq/r8152/discussions/310

### Description of your products

> Synology DS918+
>    DSM 7.1.1-42962 Update 5
>   CableCreation 2.5G Ethernet to USB Adapter (Type-A) (CD0673)

### Description of your environment

> connect directly to modemrouter, the Telstra Smart Modem Gen 2 DJA0231
> - cable category Cat.5e

### Output of ``dmesg`` command

https://pastebin.com/vDxXkvj2

### Output of ``lsusb`` command

[redacted]@Nightmaresonwax:~$ lsusb
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F00835A19DA5A83)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-1         0bda:8156:3104 00  3.20 5000MBit/s 256mA 2IFs (Realtek USB 10/100/1G/2.5G LAN 401000001)

### Output of ``ifconfig -a`` command

https://pastebin.com/SxVGM8S1

hetzjagd, 2023-06-06:
I made a grave mistake.  I was using the 7.2 driver, not 7.1.  This entire thread can be deleted as it won't help anyone.

bb-qq, 2023-07-02:
There seems to be no problem, so I will close this issue.

closed 2023-07-02 by bb-qq