bb-qq/r8152 #312: Driver Fails to run after updating the 7.1.5

slvrscoobie, 2023-06-10:
Driver shows "failed to run" after updating the DSM to 7.1.5 on Geminilake DS920+

### Description of your products

> - DS920+
>     Linux Diskstation 4.4.180+ #42962 SMP Sat Apr 8 00:14:24 CST 2023 x86_64 GNU/Linux synology_geminilake_920+
>     - DSM version 7.1.5
> - 

### Description of your environment

Connected to Netgear 10/1gigE switch

````
(paste output here, or attach as a file)
````

|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-2         2109:2817:03a4 09  2.10  480MBit/s 0mA 1IF  (VIA Labs, Inc.          USB2.0 Hub              ffffff94ffffffb4ffffff94ffffffb4) hub
    |__1-2.2     04f9:0042:0100 00  2.00  480MBit/s 2mA 1IF  (Brother HL-2270DW series L1N859643)
    |__1-2.3     051d:0002:0090 00  2.00   12MBit/s 2mA 1IF  (American Power Conversion Back-UPS BX1500G FW:866.L5 .D USB FW:L5  4B1103P32327  )
    |__1-2.4     2109:2817:03a4 09  2.10  480MBit/s 0mA 1IF  (VIA Labs, Inc.          USB2.0 Hub              ffffff94ffffffb4ffffff94ffffffb2) hub
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 650022699713F204)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-1         0bda:8156:3100 00  3.20 5000MBit/s 256mA 2IFs (Realtek USB 10/100/1G/2.5 LAN 001000001)
  |__2-2         2109:0817:03a4 09  3.10 5000MBit/s 0mA 1IF  (VIA Labs, Inc.          USB3.0 Hub              ffffff94ffffffb4ffffff94ffffffb0) hub
    |__2-2.4     2109:0817:03a4 09  3.10 5000MBit/s 0mA 1IF  (VIA Labs, Inc.          USB3.0 Hub              ffffff94ffffffb4ffffff94ffffffb1) hub

````

````

### Output of ``ifconfig -a`` command

````
bond0     Link encap:Ethernet  HWaddr 90:09:D0:18:73:98  
          inet addr:192.168.44.77  Bcast:192.168.44.255  Mask:255.255.255.0
          inet6 addr: fdb9:fc5:4c6d:4ac1:9209:d0ff:fe18:7398/64 Scope:Global
          inet6 addr: fe80::9209:d0ff:fe18:7398/64 Scope:Link
          UP BROADCAST RUNNING MASTER MULTICAST  MTU:1500  Metric:1
          RX packets:3611884 errors:0 dropped:0 overruns:0 frame:0
          TX packets:8
[dmseg.txt](https://github.com/bb-qq/r8152/files/11712428/dmseg.txt)
5615 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:2932000118 (2.7 GiB)  TX bytes:23864804 (22.7 MiB)

docker0   Link encap:Ethernet  HWaddr 02:42:6B:A4:83:2D  
          inet addr:172.17.0.1  Bcast:172.17.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth0      Link encap:Ethernet  HWaddr 90:09:D0:18:73:98  
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:3605461 errors:0 dropped:0 overruns:0 frame:0
          TX packets:24674 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:2930500350 (2.7 GiB)  TX bytes:8621587 (8.2 MiB)
          Interrupt:99 base 0x2000  

eth1      Link encap:Ethernet  HWaddr 90:09:D0:18:73:99  
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:6423 errors:0 dropped:0 overruns:0 frame:0
          TX packets:60941 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:1499768 (1.4 MiB)  TX bytes:15243217 (14.5 MiB)
          Interrupt:98 base 0x8000  

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:6414 errors:0 dropped:0 overruns:0 frame:0
          TX packets:6414 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:2796450 (2.6 MiB)  TX bytes:2796450 (2.6 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

tailscale Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  
          inet addr:100.71.4.41  P-t-P:100.71.4.41  Mask:255.255.255.255
          inet6 addr: fd7a:115c:a1e0:ab12:4843:cd96:6247:429/128 Scope:Global
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1280  Metric:1
          RX packets:3576188 errors:0 dropped:0 overruns:0 frame:0
          TX packets:58972 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:500 
          RX bytes:2625666335 (2.4 GiB)  TX bytes:5912965 (5.6 MiB)

````

slvrscoobie, 2023-06-10:
[dmseg.txt](https://github.com/bb-qq/r8152/files/11712431/dmseg.txt)

slvrscoobie, 2023-06-10:
of COURSE after I post this I try one more thing that works! 
For anyone else that finds this, this is what fixed it: 
1) Uninstall old package (or package that says it needs repair)./ 
2)Install correct driver, but de-select "run after install"
3) After installation completes, run the sudo install -m 4755 -o root -D /var/packages/r8152/target/r8152/spk_su /opt/sbin/spk_su command
4) start package. Worked fine in my system.

mrjohnpoz, 2023-06-12:
> updating the DSM to 7.1.5

Typo? did you mean 7.1.1 update 5?  there is no 7.1.5 release of dsm..

slvrscoobie, 2023-06-12:
> > updating the DSM to 7.1.5
> 
> 
> 
> Typo? did you mean 7.1.1 update 5?  there is no 7.1.5 release of dsm.. 

Sorry. Yes, It's 7.1.1 Update 5.

bb-qq, 2023-07-02:
The problem seems to have been resolved by reinstalling the driver, so I close this issue.

closed 2023-07-02 by bb-qq