bb-qq/r8152 #313: DS16+ and Cable Matters adapter : works but not at full speed

extric99, 2023-06-19:
### Description of the problem

Result are not consistent between upload and downloads like reported elsewhere on the forum (https://github.com/bb-qq/r8152/issues/296) and not reaching the theoretical speeds possible.

I tried reinstalling but I got only a minor improvement not real fix.

I can get max iperf results between the NUC11 and another PC so the issue is not on the hardware I am using.

[image]

Iperf from NUC to DS916

````
Connecting to host 192.168.0.73, port 5201
[  5] local 192.168.0.5 port 37028 connected to 192.168.0.73 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   155 MBytes  1.30 Gbits/sec    1    242 KBytes       
[  5]   1.00-2.00   sec   155 MBytes  1.30 Gbits/sec    0    245 KBytes       
[  5]   2.00-3.00   sec   184 MBytes  1.55 Gbits/sec    0    291 KBytes       
[  5]   3.00-4.00   sec   184 MBytes  1.55 Gbits/sec    1    310 KBytes       
[  5]   4.00-5.00   sec   196 MBytes  1.64 Gbits/sec    8    338 KBytes       
[  5]   5.00-6.00   sec   176 MBytes  1.48 Gbits/sec    0    342 KBytes       
[  5]   6.00-7.00   sec   202 MBytes  1.69 Gbits/sec    0    366 KBytes       
[  5]   7.00-8.00   sec   246 MBytes  2.06 Gbits/sec    0    400 KBytes       
[  5]   8.00-9.00   sec   187 MBytes  1.57 Gbits/sec   26    403 KBytes       
[  5]   9.00-10.00  sec   177 MBytes  1.48 Gbits/sec    1    403 KBytes       
[  5]  10.00-11.00  sec   214 MBytes  1.80 Gbits/sec    0    403 KBytes       
[  5]  11.00-12.00  sec   195 MBytes  1.63 Gbits/sec    0    403 KBytes       
[  5]  12.00-13.00  sec   178 MBytes  1.49 Gbits/sec    1    403 KBytes       
[  5]  13.00-14.00  sec   195 MBytes  1.64 Gbits/sec    0    409 KBytes       
[  5]  14.00-15.00  sec   175 MBytes  1.47 Gbits/sec    0    409 KBytes       
[  5]  15.00-16.00  sec   218 MBytes  1.83 Gbits/sec    2    441 KBytes       
[  5]  16.00-17.00  sec   263 MBytes  2.21 Gbits/sec    6    494 KBytes       
[  5]  17.00-18.00  sec   259 MBytes  2.17 Gbits/sec    0    516 KBytes       
[  5]  18.00-19.00  sec   265 MBytes  2.22 Gbits/sec    0    540 KBytes       
[  5]  19.00-20.00  sec   222 MBytes  1.86 Gbits/sec    5    544 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-20.00  sec  3.95 GBytes  1.70 Gbits/sec   51             sender
[  5]   0.00-20.00  sec  3.95 GBytes  1.69 Gbits/sec                  receiver

````
Iperf from NUC to DS916 Reverse

````
Connecting to host 192.168.0.73, port 5201
Reverse mode, remote host 192.168.0.73 is sending
[  5] local 192.168.0.5 port 34176 connected to 192.168.0.73 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec   261 MBytes  2.19 Gbits/sec                  
[  5]   1.00-2.00   sec   264 MBytes  2.22 Gbits/sec                  
[  5]   2.00-3.00   sec   265 MBytes  2.23 Gbits/sec                  
[  5]   3.00-4.00   sec   246 MBytes  2.06 Gbits/sec                  
[  5]   4.00-5.00   sec   265 MBytes  2.23 Gbits/sec                  
[  5]   5.00-6.00   sec   267 MBytes  2.24 Gbits/sec                  
[  5]   6.00-7.00   sec   264 MBytes  2.22 Gbits/sec                  
[  5]   7.00-8.00   sec   266 MBytes  2.23 Gbits/sec                  
[  5]   8.00-9.00   sec   266 MBytes  2.24 Gbits/sec                  
[  5]   9.00-10.00  sec   255 MBytes  2.14 Gbits/sec                  
[  5]  10.00-11.00  sec   265 MBytes  2.22 Gbits/sec                  
[  5]  11.00-12.00  sec   265 MBytes  2.23 Gbits/sec                  
[  5]  12.00-13.00  sec   265 MBytes  2.23 Gbits/sec                  
[  5]  13.00-14.00  sec   264 MBytes  2.22 Gbits/sec                  
[  5]  14.00-15.00  sec   259 MBytes  2.18 Gbits/sec                  
[  5]  15.00-16.00  sec   259 MBytes  2.17 Gbits/sec                  
[  5]  16.00-17.00  sec   265 MBytes  2.23 Gbits/sec                  
[  5]  17.00-18.00  sec   265 MBytes  2.23 Gbits/sec                  
[  5]  18.00-19.00  sec   264 MBytes  2.21 Gbits/sec                  
[  5]  19.00-20.00  sec   266 MBytes  2.23 Gbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[... 292 lines cut ...]
ovs_eth2  Link encap:Ethernet  HWaddr 5C:85:7E:3E:F3:4A  
          inet addr:192.168.0.73  Bcast:192.168.0.255  Mask:255.255.255.0
          inet6 addr: fe80::5e85:7eff:fe3e:f34a/64 Scope:Link
          inet6 addr: 2a02:1811:1422:fd00:5e85:7eff:fe3e:f34a/64 Scope:Global
          inet6 addr: 2a02:1811:1422:fd00:3206:7edc:8d89:c4e/128 Scope:Global
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:62611299 errors:0 dropped:0 overruns:0 frame:0
          TX packets:17972060 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:80008247363 (74.5 GiB)  TX bytes:43798421148 (40.7 GiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

tap021132 Link encap:Ethernet  HWaddr EE:F3:C5:A9:87:3B  
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:5476260 errors:0 dropped:0 overruns:0 frame:0
          TX packets:11908484 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:822853158 (784.7 MiB)  TX bytes:6604656563 (6.1 GiB)
````

bb-qq, 2023-07-02:
Throughput is above 1Gbps both up and down, so there seems to be no problem with routing and other settings.

Have you tried using a larger MTU value?

extric99, 2023-07-06:
I did test with MTU 9000 but saw no performance increase. I am now on holidays but I will retest when I am back.

bb-qq, 2023-07-15:
Looking at the iperf results, it appears that retransmissions are occurring; `Retr` is usually 0.

So what I would like to check is the following.
- What category and length of LAN cable are you using?
- What is the model number revision of the NIC installed in the NUC?
  - Intel I225 is known to have a faulty revision.

extric99, 2023-08-09:
Can confirm the issue is resolved and everything is working at full speed (MTU9000)

[image]

closed 2023-08-09 by extric99

eightsheep67, 2023-09-24:
where can i get the beta release? I install the 2.17.1-1 version.
my nas is 916+ and i cannot get it work at full speed. and it works unstable, the bandwidth is up and down.
the best result is around 1.8Gb/s.
[image]
[image]

eightsheep67, 2023-09-25:
> I installed the beta release and it would appear it has solved the issue I will check the cabling in the weekend.
> 
> ```
> Connecting to host 192.168.0.73, port 5201
> [  5] local 192.168.0.5 port 54572 connected to 192.168.0.73 port 5201
> [ ID] Interval           Transfer     Bitrate         Retr  Cwnd
> [  5]   0.00-1.00   sec   243 MBytes  2.04 Gbits/sec   13    690 KBytes       
> [  5]   1.00-2.00   sec   279 MBytes  2.34 Gbits/sec    5   1.02 MBytes       
> [  5]   2.00-3.00   sec   264 MBytes  2.21 Gbits/sec    1   1022 KBytes       
> [  5]   3.00-4.00   sec   262 MBytes  2.20 Gbits/sec   12    551 KBytes       
> [  5]   4.00-5.00   sec   294 MBytes  2.46 Gbits/sec    1   1.67 MBytes       
> [  5]   5.00-6.00   sec   259 MBytes  2.17 Gbits/sec   14    848 KBytes       
> [  5]   6.00-7.00   sec   289 MBytes  2.42 Gbits/sec    0   1.75 MBytes       
> [  5]   7.00-8.00   sec   281 MBytes  2.36 Gbits/sec    2   1.60 MBytes       
> [  5]   8.00-9.00   sec   265 MBytes  2.22 Gbits/sec    4   1.28 MBytes       
> [  5]   9.00-10.00  sec   271 MBytes  2.28 Gbits/sec    8    891 KBytes       
> [  5]  10.00-11.00  sec   295 MBytes  2.47 Gbits/sec    0   1.86 MBytes       
> [  5]  11.00-12.00  sec   294 MBytes  2.46 Gbits/sec    0   2.48 MBytes       
> [  5]  12.00-13.00  sec   295 MBytes  2.47 Gbits/sec    0   2.97 MBytes       
> [  5]  13.00-14.00  sec   286 MBytes  2.40 Gbits/sec   10    830 KBytes       
> [  5]  14.00-15.00  sec   238 MBytes  1.99 Gbits/sec   15    157 KBytes       
> [  5]  15.00-16.00  sec   256 MBytes  2.15 Gbits/sec   13    848 KBytes       
> [  5]  16.00-17.00  sec   275 MBytes  2.31 Gbits/sec    4    813 KBytes       
> [  5]  17.00-18.00  sec   271 MBytes  2.28 Gbits/sec    2   1.02 MBytes       
> [  5]  18.00-19.00  sec   260 MBytes  2.18 Gbits/sec   21   1.49 MBytes       
> [  5]  19.00-20.00  sec   278 MBytes  2.33 Gbits/sec    9   1.42 MBytes       
> - - - - - - - - - - - - - - - - - - - - - - - - -
> [ ID] Interval           Transfer     Bitrate         Retr
> [  5]   0.00-20.00  sec  5.33 GBytes  2.29 Gbits/sec  134             sender
> [  5]   0.00-20.00  sec  5.32 GBytes  2.29 Gbits/sec                  receiver`
> ```

where is the beta release, pls tell me i'm dying to know。

extric99, 2023-09-25:
Use the latest release and make sure your mtu=9000. Not sure why you use 4088

Sjekke, 2023-09-25:
Only use mtu=9000 if you have a point2point connection else keep it on 1500. Also use the front usb port ...

eightsheep67, 2023-09-26:
> Use the latest release and make sure your mtu=9000. Not sure why you use 4088

i iperf two PCs connected by one switch,  4088byte jumbo frame is only one can up to 2.45Gb/s.
then i tested all the number on NAS, 4000 or 4088 is the best performance, otherwise would down to 1.5Gb/s

eightsheep67, 2023-09-26:
> Only use mtu=9000 if you have a point2point connection else keep it on 1500. Also use the front usb port ...

my 916+ front usb port seems not working with my usb adapter. I'll try again later.

eightsheep67, 2023-09-26:
> Only use mtu=9000 if you have a point2point connection else keep it on 1500. Also use the front usb port ...

man U saved me!!!! I tried many times and get the adapter work with the front port (plug and unplug), then everything back on track!!!
up and down is close to 2.45Gb/s !!!!