bb-qq/r8152 #315: Using two usb lan card at the same time?

yuickey, 2023-06-23:
I'm trying to use smb multichannel.

Has anyone tried using a 2.5G lan card in both the front and back usb ports of a Synology at the same time?

qiuyan81, 2023-06-25:
I have the same requirement, but I can't seem to get the answer.

bb-qq, 2023-07-02:
See comments here about using multiple adapters.
https://github.com/bb-qq/r8152/issues/300#issuecomment-1518530441

However, I am not sure if SMB multichannel will perform as expected. Depending on the product model, the USB host controller may not have enough bandwidth.

qiuyan81, 2023-07-03:
> See comments here about using multiple adapters. [#300 (comment)](https://github.com/bb-qq/r8152/issues/300#issuecomment-1518530441)
> 
> However, I am not sure if SMB multichannel will perform as expected. Depending on the product model, the USB host controller may not have enough bandwidth.

Thanks to the author's reply, I can already enable 2 2.5G network adapters, but I can't make them work under the aggregation network(LACP).
my machine model ds920,dsm7.2.

bb-qq, 2023-07-15:
Since LACP works completely outside of the driver, there is most likely a problem with the configuration or switch. How do you verify that it is working or not?

bb-qq, 2024-06-15:
Since there has been no response for a long time, this issue is closed. Please feel free to re-open it if you have any questions.
Other users have reported that LACP works, but settings are not persisted across reboots.

closed 2024-06-15 by bb-qq