bb-qq/r8152 #318: 2.5Gbe NIC Not Being Used as Primary After Changing Volume 1 to Encrypted Volume

abracadabra1111, 2023-07-21:
I posted details of the issue I'm encountering on reddit (https://www.reddit.com/r/synology/comments/151h36w/mystery_ds920_usb_adapter_no_longer_being_used_as/). But figure it would be worthwhile to come to the bb-qq source. 

The problem I'm encountering after changing volume 1 to an encrypted volume is that the 2.5Gbe USB NIC is no longer being used as the primary LAN interface. Several problems are evident:
1. DNS queries are all from the built-in LAN interface IP
2. Network UPS access (to another NAS) does not work when I whitelist the USB LAN IP. It only works when I whitelist the built-in LAN interface IP
3. Hyperbackup Server does not work properly when using the USB LAN IP, only the built-in LAN IP.

Happy to hear any thoughts on troubleshooting, but I'm mostly curious on how I can do a clean uninstall of the bb-qq drivers and then reinstall to see if that fixes it. I've used the Uninstall within Package Center and reinstalled drivers and that didn't change the noted behavior.

bb-qq, 2023-07-21:
When IP addresses of the same subnet are assigned to multiple interfaces, detailed settings are required to make the interfaces communicate as desired. Therefore, I recommend avoiding such a configuration as much as possible.

abracadabra1111, 2023-07-21:
> When IP addresses of the same subnet are assigned to multiple interfaces, detailed settings are required to make the interfaces communicate as desired. Therefore, I recommend avoiding such a configuration as much as possible.

The additional link on the same subnet provides redundancy and can be leveraged for SMB3 multicast. And I've used this configuration for over a year without problems. It was only after changing to an encrypted volume 1 that this problem appeared. Any thoughts on how to do a clean uninstall/reinstall?

bb-qq, 2023-08-11:
To clean everything, clear your `@appconf` directory after uninstallation.

```
$ ls -l /var/packages/r8152/
total 96
drwxr-xr-x 2 root root  4096 Aug  6 21:06 conf
-rw-r--r-- 1 root root     0 Aug  6 21:06 enabled
lrwxrwxrwx 1 root root    23 Mar 19 18:20 etc -> /volume8/@appconf/r8152
lrwxrwxrwx 1 root root    23 Mar 19 18:20 home -> /volume8/@apphome/r8152
-rw-r--r-- 1 root root 86887 Mar 19 18:20 INFO
drwxr-xr-x 2 root root  4096 Mar 19 15:37 scripts
lrwxrwxrwx 1 root root    24 Mar 19 18:20 target -> /volume8/@appstore/r8152
lrwxrwxrwx 1 root root    23 Mar 19 18:20 tmp -> /volume8/@apptemp/r8152
lrwxrwxrwx 1 root root    23 Mar 19 18:20 var -> /volume8/@appdata/r8152
```

In this case, your @appconf is `/volume8/@appconf/r8152`. My driver persists settings to here.
So manually remove this directory after uninstalling the package.

Of course, the location of the directory will vary depending on the volume on which the package is installed.

abracadabra1111, 2023-08-11:
> To clean everything, clear your `@appconf` directory after uninstallation.
>
> ```
> $ ls -l /var/packages/r8152/
> total 96
> drwxr-xr-x 2 root root  4096 Aug  6 21:06 conf
> -rw-r--r-- 1 root root     0 Aug [...]

Thanks! Still seeing the same behavior where DNS requests seem to be coming exclusively from the secondary adapter (1Gbe Integrated LAN port) vice the 2.5Gbe USB-to-ethernet device. Looks like this behavior is consistent for both of my adapters in DS920+ and DS420+ running DSM 7.2-64570 Update 1.

abracadabra1111, 2023-08-11:
Inadvertently closed.

bb-qq, 2023-08-11:
I don't know the details, but perhaps your problem can be solved by changing the Service Order. This setting is located in Control Panel - Network - Network Interface - Manage.

abracadabra1111, 2023-08-11:
> I don't know the details, but perhaps your problem can be solved by changing the Service Order. This setting is located in Control Panel - Network - Network Interface - Manage.

Unfortunately, that's part of the problem. It persists regardless of service order. But LAN3 (USB-to-E) is set as the primary in service order.

bb-qq, 2023-09-10:
The only other way I can come up with is to explicitly specify the interface priority with the `ifconfig metric` command.

FlorianHeigl, 2024-03-09:
what happens if you remove the gateway entry from the 1g interface?

abracadabra1111, 2024-03-10:
> what happens if you remove the gateway entry from the 1g interface?

I noted this in the reddit thread I referenced, but removing the 1g interface did not help. I haven't done this recently, but from what I recall there were no DNS queries generated when only the 2.5G interface was active.

bb-qq, 2024-06-15:
The results of the following commands might be more precise.
* `netstat -nr` 
* `ifconfig -a`
* `cat /etc/resolv.conf`

abracadabra1111, 2024-06-15:
> The results of the following commands might be more precise the situation.
> 
>     * `netstat -nr`
> 
>     * `ifconfig -a`
> 
>     * `cat /etc/resolv.conf`

Thanks - looks like more info that just continues to confound me. netstat command seems to show that eht2 is being used as the interface for DNS queries to the gateway.
`/$ netstat -nr
Kernel IP routing table
Destination     Gateway         Genmask         Flags   MSS Window  irtt Iface
0.0.0.0         192.168.100.1    0.0.0.0         UG        0 0          0 eth2
169.254.0.0     0.0.0.0         255.255.0.0     U         0 0          0 eth1
172.17.0.0      0.0.0.0         255.255.0.0     U         0 0          0 docker0
192.168.100.0    0.0.0.0         255.255.255.0   U         0 0          0 eth0
192.168.100.0    0.0.0.0         255.255.255.0   U         0 0          0 eth2
`
ifconfig shows that eth2 is being used for the majority of traffic (5.3TB RX/9.2 TB TX). But eth0 is still being used much more heavily than one would expect for a secondary, unused interface (550MB RX/850MB TX).

bb-qq, 2024-06-16:
The default gateway is not used for communication to the same subnet.

Also, if multiple interfaces belong to the same subnet, the value of the interface metric determines which one is used. This can be checked with `ifconfig -a`.

abracadabra1111, 2024-06-16:
> The default gateway is not used for communication to the same subnet.
> 
> Also, if multiple interfaces belong to the same subnet, the value of the interface metric determines which one is used. This can be checked with `ifconfig -a`.
First, thanks for continuing to try and troubleshoot this issue. 

Ran route -n and the interface metric appears identical for all - 0.
`Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
0.0.0.0         192.168.40.1    0.0.0.0         UG    0      0        0 eth2
169.254.0.0     0.0.0.0         255.255.0.0     U     0      0        0 eth1
172.17.0.0      0.0.0.0         255.255.0.0     U     0      0        0 docker0
192.168.40.0    0.0.0.0         255.255.255.0   U     0      0        0 eth0
192.168.40.0    0.0.0.0         255.255.255.0   U     0      0        0 eth2
`

bb-qq, 2024-06-23:
Here are my guesses:

According to the information in the routing table, eth0 is used for intra-LAN communication, because the first entry is used when routes with the same metric value exist.

For communication to the Internet, the default gateway is used. According to information in the routing table, eth2 is used for communication with the default gateway. Therefore, it is likely that different interfaces are used for LAN and WAN communication.

To avoid this situation, it is best to assign different subnets to each interface, but if you want to assign the same subnet, you can set the metric value manually.

abracadabra1111, 2024-06-23:
Closing this since it doesn't seem to be an issue with the underlying bbq driver, but rather a DSM limitation associated with primary interface assignment.

closed 2024-06-23 by abracadabra1111