bb-qq/r8152 #320: After updating driver, cant find shared folders in Explorer.

hirez19, 2023-07-22:
First, thanks for many nice releases.
But when updating, I am forced to change the Server name, on the synology NAS.

Its kind of an hassle each time to change all shortcuts so wonder if there is a quick fix for this? 
The same happened last time, so i alternate between several names, and switch between to keep access through windows.
Servers, apps, etc are working as its static IP but windows, and acessing shared folders goes a bit weird.
Have tried switching smb3 to 2 and all kinds of settings in windows discovery but its like it does not like the name and I have to change it.
Any thoughts would be appreciated.

bb-qq, 2023-08-11:
This is the first time I have heard of the symptoms of having to change the server name.
Wouldn't restarting the Windows PC solve the problem?

Information such as what other devices are connected to the network might tell us something more.

hirez19, 2023-08-23:
Its a large "home" network, so Sonos, cameras, tvs, a lot of devices are connected.
I have of course, restarted, tried over Wifi, lan, etc. Any help in the right direction would be great, i can try different settings, but good to hear i am the only one :)

bb-qq, 2024-06-15:
It has been a long time and I have no idea what is going on as I do not have sufficient knowledge of how the SMB works.

I would like to close this issue as I have no further advice to offer, but if you have learned anything new since then, please let me know.

hirez19, 2024-06-18:
I havent dared to update since last time, as I am worried, it will again throw things around. So now i try to use more like "IP" adresses for the nas access. But let me try to update to latest synology during weekend and report back maybe?

hirez19, 2024-08-10:
Update: 920 worked to update without having to re-install or change anything.. will ask my friend to to the same with his 918 and i guess something has been fixed, before we always had to reinstall the driver and change the windows display name for it to work, weird, its on an Asus 2,5 giga

hirez19, 2024-08-11:
918: same probs as before, but you can close the issue, its a lot of different version out there and i solved it manually..
But in case someone has problem accessing their folders, try renaming the "server name" as using ip it finds everything :)

closed 2024-09-22 by bb-qq

hirez19, 2024-09-24:
Of course i thank you for all the help and great drivers! I simply think if anyone else has this problem, update, restart all the windows machines, wait a day etc. and it actually got found. I tried releasing ips etc. but in the end it worked by just waiting a few hours.