bb-qq/r8152 #327: ASUS USB-C 2500(asu2.5g2) works on DS 1513+

marineyeh, 2023-08-11:
### Description of the problem

> ASUS USB-C 2500(asu2.5g2) works on DS 1513+. Speed double than before.

flypigxyz, 2023-11-29:
my machine is ds1513+, also use the asus usb-c 2500 gen2 , but it not worked and the speed half.

cupanther, 2024-07-26:
I have a DS1513+ as well and can only get 1.2G one way and 1G the other way.  I am using the same ASUS USB-C 2500.  I have verified the USB port accepts 5G, set the MTU to the recommended setting, verified SMB 3 is enabled and checked that the ethernet adapter reports it can do 2.5G.  The adapter is plugged directly into the left USB3 port, but have also tried the other one and no difference.  Is there anything else I need to check or change?