bb-qq/r8152 #329: Install error / Repair loop -  2.17.1-1 / DSM 7.2-64561 / DS916+

McMac, 2023-08-13:
Was running 2.16.3-3 on my 916+ which I let me update to DSM 7.2 without a problem (Club 3D adapter). Saw that there was a new package (r8152-braswell-2.17.1-1_7.2.spk) specifically for 7.2, plus I seem to have some intermittent disconnects to my NFS shares. I figured I'd update to confirmed version. And now I am stuck in the repair loop. Tried several iterations of complete uninstall #309 as well as re-installs all to no avail.

Found a thread providing this:

sudo /var/packages/r8152/scripts/start-stop-status stop
sudo /var/packages/r8152/scripts/start-stop-status start
(output see below)

Situation now is:
1. Driver is loaded manually
2. NIC shows up, gets IP via DHCP, accessible via ssh etc. 
3. UI shows error
4. Trying to re-install driver as per instructions, it gives the Error -> Repairing -> "Failed to repair "RTL8152/RTL8153" driver". 
5. Rebooting will again not start the driver (due to it's error state?)

--> Driver is working apparently but only (?) the UI isn't catching up preventing proper loading at startup

What additional information can I provide for further analysis? 

---

<code>
++ dirname /var/packages/r8152/scripts/start-stop-status
+ script_root=/var/packages/r8152/scripts
++ realpath -m /var/packages/r8152/scripts/..
+ package_root=/var/packages/r8152
++ basename /var/packages/r8152
+ driver_name=r8152
+ driver_root=/var/packages/r8152/target/r8152
+ log_root=/var/packages/r8152/var/log
+ mkdir -p /var/packages/r8152/var/log
+ exec
++ tee -a /var/packages/r8152/var/log/start-stop-status.log
+ echo ----------
----------
+ date
Sun Aug 13 03:09:01 PM CEST 2023
+ grep -E '^(version|create_time)' /var/packages/r8152/INFO
version="2.17.1-1"
create_time="20230715-10:17:25"
+ echo stop
stop
+ case $1 in
+ set_interfaces down
+ action=down
+ target_interface_name=
++ ls /sys/class/net
+ all_interface_names='eth0
eth1
lo
sit0'
+ '[' -n ']'
+ '[' '!' -e /sys/class/net/ ']'
+ for interface_name in ${all_interface_names}
+ [[ -n '' ]]
+ '[' '!' -e /sys/class/net/eth0/device/driver ']'
++ ls -ld /sys/class/net/eth0/device/driver
+ driver_location='lrwxrwxrwx 1 root root 0 Aug 13 15:07 /sys/class/net/eth0/device/driver -> ../../../../bus/pci/drivers/igb'
++ echo 'lrwxrwxrwx 1 root root 0 Aug 13 15:07 /sys/class/net/eth0/device/driver -> ../../../../bus/pci/drivers/igb'
++ grep r8152
+ '[' '!' -z '' ']'
+ for interface_name in ${all_interface_names}
+ [[ -n '' ]]
+ '[' '!' -e /sys/class/net/eth1/device/driver ']'
++ ls -ld /sys/class/net/eth1/device/driver
+ driver_location='lrwxrwxrwx 1 root root 0 Aug 13 15:07 /sys/class/net/eth1/device/driver -> ../../../../bus/pci/drivers/igb'
++ echo 'lrwxrwxrwx 1 root root 0 Aug 13 15:07 /sys/class/net/eth1/device/driver -> ../../../../bus/pci/drivers/igb'
++ grep r8152
+ '[' '!' -z '' ']'
+ for interface_name in ${all_interface_names}
+ [[ -n '' ]]
+ '[' '!' -e /sys/class/net/lo/device/driver ']'
+ continue
+ for interface_name in ${all_interface_names}
+ [[ -n '' ]]
+ '[' '!' -e /sys/class/net/sit0/device/driver ']'
+ continue
+ /sbin/rmmod /var/packages/r8152/target/r8152/r8152.ko
rmmod: ERROR: Module r8152 is not currently loaded
+ true
+ '[' -e /var/packages/r8152/target/r8152/usbnet.ko ']'
+ '[' -e /var/packages/r8152/target/r8152/mii.ko ']'
+ exit 0
</code>

however then manually loading it using the start command, the NIC shows up:

<code>
++ dirname /var/packages/r8152/scripts/start-stop-status
+ script_root=/var/packages/r8152/scripts
++ realpath -m /var/packages/r8152/scripts/..
+ package_root=/var/packages/r8152
++ basename /var/packages/r8152
+ driver_name=r8152
+ driver_root=/var/packages/r8152/target/r8152
+ log_root=/var/packages/r8152/var/log
+ mkdir -p /var/packages/r8152/var/log
+ exec
++ tee -a /var/packages/r8152/var/log/start-stop-status.log
+ echo ----------
----------
+ date
Sun Aug 13 03:09:13 PM CEST 2023
+ grep -E '^(version|create_time)' /var/packages/r8152/INFO
version="2.17.1-1"
create_time="20230715-10:17:25"
+ echo start
start
+ case $1 in
+ '[' -e /var/packages/r8152/scripts/apply-memory-setting ']'
+ '[' -w /sys/module/usbcore/parameters/autosuspend ']'
++ count_usb_interfaces
++ ls -l /sys/class/net/
++ grep /usb
++ wc -l
+ initial_count=0
+ '[' -e /var/packages/r8152/target/r8152/mii.ko ']'
+ '[' -e /var/packages/r8152/target/r8152/usbnet.ko ']'
+ /sbin/insmod /var/packages/r8152/target/r8152/r8152.ko
+ '[' -r /usr/lib/udev/rules.d/51-usb-r8152-net.rules ']'
++ seq 1 60
+ for i in `seq 1 60`
++ count_usb_interfaces
++ ls -l /sys/class/net/
++ grep /usb
++ wc -l
+ '[' 0 -ne 0 ']'
+ sleep 1
+ for i in `seq 1 60`
++ count_usb_interfaces
++ ls -l /sys/class/net/
++ grep /usb
++ wc -l
+ '[' 0 -ne 0 ']'
+ sleep 1
+ for i in `seq 1 60`
++ count_usb_interfaces
++ grep /usb
++ ls -l /sys/class/net/
++ wc -l
+ '[' 1 -ne 0 ']'
+ break
+ set_interfaces up
+ action=up
+ target_interface_name=
++ ls /sys/class/net
+ all_interface_names='eth0
eth1
eth2
lo
sit0'
+ '[' -n ']'
+ '[' '!' -e /sys/class/net/ ']'
+ for interface_name in ${all_interface_names}
+ [[ -n '' ]]
+ '[' '!' -e /sys/class/net/eth0/device/driver ']'
++ ls -ld /sys/class/net/eth0/device/driver
+ driver_location='lrwxrwxrwx 1 root root 0 Aug 13 15:07 /sys/class/net/eth0/device/driver -> ../../../../bus/pci/drivers/igb'
++ grep r8152
++ echo 'lrwxrwxrwx 1 root root 0 Aug 13 15:07 /sys/class/net/eth0/device/driver -> ../../../../bus/pci/drivers/igb'
+ '[' '!' -z '' ']'
+ for interface_name in ${all_interface_names}
+ [[ -n '' ]]
+ '[' '!' -e /sys/class/net/eth1/device/driver ']'
++ ls -ld /sys/class/net/eth1/device/driver
+ driver_location='lrwxrwxrwx 1 root root 0 Aug 13 15:07 /sys/class/net/eth1/device/driver -> ../../../../bus/pci/drivers/igb'
++ grep r8152
++ echo 'lrwxrwxrwx 1 root root 0 Aug 13 15:07 /sys/class/net/eth1/device/driver -> ../../../../bus/pci/drivers/igb'
+ '[' '!' -z '' ']'
+ for interface_name in ${all_interface_names}
+ [[ -n '' ]]
+ '[' '!' -e /sys/class/net/eth2/device/driver ']'
++ ls -ld /sys/class/net/eth2/device/driver
+ driver_location='lrwxrwxrwx 1 root root 0 Aug 13 15:09 /sys/class/net/eth2/device/driver -> ../../../../../../../bus/usb/drivers/r8152'
++ grep r8152
++ echo 'lrwxrwxrwx 1 root root 0 Aug 13 15:09 /sys/class/net/eth2/device/driver -> ../../../../../../../bus/usb/drivers/r8152'
+ '[' '!' -z 'lrwxrwxrwx 1 root root 0 Aug 13 15:09 /sys/class/net/eth2/device/driver -> ../../../../../../../bus/usb/drivers/r8152' ']'
+ config_file=/etc/sysconfig/network-scripts/ifcfg-eth2
+ config_storage_location=/var/packages/r8152/etc/ifcfg-eth2
+ '[' -f /etc/sysconfig/network-scripts/ifcfg-eth2 ']'
+ '[' up = up ']'
+ '[' -f /var/packages/r8152/etc/ifcfg-eth2 ']'
+ ifconfig eth2 up
+ set_ovs_interface eth2 up
+ interface_name=eth2
+ action=up
+ config_file_ovs=/etc/sysconfig/network-scripts/ifcfg-ovs_eth2
+ config_storage_location_ovs=/var/packages/r8152/etc/ifcfg-ovs_eth2
+ is_ovs_enable
+ '[' '!' -f /usr/syno/etc/synoovs/ovs_reg.conf ']'
+ return 1
+ '[' up = up ']'
+ is_ovs_enable
+ '[' '!' -f /usr/syno/etc/synoovs/ovs_reg.conf ']'
+ return 1
+ '[' -r /etc/sysconfig/network-scripts/ifcfg-eth2 ']'
+ apply_if_config eth2
+ target_interface_name=eth2
+ [[ ! -v BOOTPROTO ]]
+ synonet --dhcp eth2
me@<redacted>:~$ Host Name: <redacted>
Network interface: eth2
DHCP
IP: <redacted>
Mask: <redacted>
Gateway: <redacted>
DNS: <redacted>
MTU Setting: 9000
65535, unknown duplex, active mtu 1500
RX bytes: 0
TX bytes: 0
+ '[' up = up ']'
+ '[' -e /var/packages/r8152/scripts/apply-private-flags ']'
+ for interface_name in ${all_interface_names}
+ [[ -n eth2 ]]
+ [[ lo != \e\t\h\2 ]]
+ continue
+ for interface_name in ${all_interface_names}
+ [[ -n eth2 ]]
+ [[ sit0 != \e\t\h\2 ]]
+ continue
+ exit 0
</code>

What looks a bit odd is towards the end:

MTU Setting: 9000
65535, unknown duplex, active mtu 1500

bb-qq, 2023-09-10:
```
+ exit 0
```

The last line indicates that the driver startup script was completed successfully. There may be an error before or after this, but is this all the log?

Also, there may be some clues in `/var/log/synopkg.log`.

lazosweb, 2024-01-25:
For the 916+ users, it seems that if you install the package when u r at the 7.2.1 it wont work (repair loop).
If u are able to install 7.1.1, the package will work, and then u can update to 7.2.1.

bb-qq, 2024-06-15:
If you have difficulty submitting logs, please try a complete reinstallation.
https://github.com/bb-qq/r8152/wiki/Troubleshooting#uninstall-completely--full-reinstallation

McMac, 2024-07-28:
I saw that there is a new beta driver. What is the rough ETA on this? "Almost out" or "later this year"?

jameslashmar, 2024-09-05:
Having the same issue now on 920+.  The driver was installed, but I was having issues with the NAS over that IP, and decided to remove and reinstall. Stuck in this loop that when the driver fails the first time, it is stuck in this 'error' situation, that then can't be installed after the sudo command. The only option is to uninstall.  DSM 7.2.1-69057 Update 5 with bb-qq 1.3.3.0-12

bb-qq, 2024-09-19:
The latest driver is here. This is not beta.
https://github.com/bb-qq/r8152/releases/tag/2.18.1-1

closed 2025-01-02 by bb-qq