bb-qq/r8152 #330: get_registers -71 on DS2415+ and Realtek USB 10/100/1G/2.5G LAN 4013000001

rgry, 2023-08-16:
### Description of the problem

dmesg gives: a lot of these errors: r8152 3-4:1.0 eth4: get_registers -71
and the speed is about 1Gbits/s

### Description of your products
DS2415+ running DSM 7.1.1-42962 Update 6
Linux DS3 3.10.108 #42962 SMP Mon May 29 14:36:28 CST 2023 x86_64 GNU/Linux synology_avoton_2415+
Ethernet dongle is a no name REALTEK r8152 (Aliexpress)
Version of your Driver 2.17-1-1

Two windows computers with the same type of adapter connected to Zyxel MG-108-ZZ0101F (8 port 2.5GB) with cat 5e cables
iperf installed in docker on DS2415+

iperf between windows computers are 2.15Gb/s (through switch)
iperf between windows computer and NAS are  about 1Gb/s (2.5Gb adapter)
iperf between windows computer and NAS are  about 1Gb/s (builtin 1.0Gb adapter)

### Output of ``dmesg`` command
[1033288.365548] r8152 3-4:1.0 eth4: Promiscuous mode enabled
[1033288.372916] r8152 3-4:1.0 eth4: carrier on
[1033322.045906] r8152 3-4:1.0 eth4: Promiscuous mode enabled
[1033322.053033] r8152 3-4:1.0 eth4: carrier on
[1033463.842597] r8152 3-4:1.0 eth4: get_registers -71
[1033662.088002] r8152 3-4:1.0 eth4: get_registers -71
[1034961.954673] r8152 3-4:1.0 eth4: get_registers -71

### Output of ``lsusb`` command

|__usb1          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd 0000:00:16.0) hub
  |__1-1         8087:07db:0002 09  2.00  480MBit/s 0mA 1IF  ( ffffffd1ffffffb2ffffffdbffffffad) hub
    |__1-1.1     f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 650057EE94DA5A99)
|__usb2          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (Linux 3.10.108 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub
|__usb3          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (Linux 3.10.108 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub
  |__3-4         0bda:8156:3104 00  3.20 5000MBit/s 256mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 4013000001)

### Output of ``ifconfig -a`` command

docker0   Link encap:Ethernet  HWaddr 02:42:54:5D:2A:C1
          inet addr:172.17.0.1  Bcast:172.17.255.255  Mask:255.255.0.0
          inet6 addr: fe80::42:54ff:fe5d:2ac4/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:3971072 errors:0 dropped:0 overruns:0 frame:0
          TX packets:4562696 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:206931064 (197.3 MiB)  TX bytes:31842832584 (29.6 GiB)

docker-fa Link encap:Ethernet  HWaddr 02:42:DA:FB:F8:A1
          inet addr:172.18.0.1  Bcast:172.18.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

docker1ea Link encap:Ethernet  HWaddr 7E:ED:E8:17:2C:11
          inet6 addr: fe80::7ced:e8ff:fe17:2c15/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:3970777 errors:0 dropped:0 overruns:0 frame:0
          TX packets:4562484 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:262496716 (250.3 MiB)  TX bytes:31842796219 (29.6 GiB)

docker3ff Link encap:Ethernet  HWaddr 4A:22:25:B0:EF:81
          inet6 addr: fe80::4822:25ff:feb0:ef80/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:295 errors:0 dropped:0 overruns:0 frame:0
          TX packets:461 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:29356 (28.6 KiB)  TX bytes:238174 (232.5 KiB)
[... 82 lines cut ...]
          RX bytes:0 (0.0 B)  TX bytes:3161890 (3.0 MiB)

ovs_eth4  Link encap:Ethernet  HWaddr 00:E0:4C:68:00:F9
          inet addr:192.168.2.114  Bcast:192.168.2.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:9000  Metric:1
          RX packets:11551577 errors:0 dropped:0 overruns:0 frame:0
          TX packets:3500046 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:57141555175 (53.2 GiB)  TX bytes:24830195098 (23.1 GiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

tun1000   Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00
          inet addr:169.254.176.64  P-t-P:169.254.176.64  Mask:255.255.248.0
          inet6 addr: fec0:a2b2:8::40/112 Scope:Site
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1400  Metric:1
          RX packets:19216811 errors:0 dropped:0 overruns:0 frame:0
          TX packets:28982502 errors:0 dropped:495535 overruns:0 carrier:0
          collisions:0 txqueuelen:500
          RX bytes:1443152570 (1.3 GiB)  TX bytes:34386512042 (32.0 GiB)

Chavell3, 2023-10-06:
I have the same issue. My network link is even going down completely. can somebody help...

bb-qq, 2024-06-15:
Is the situation the same with the latest drivers?

rgry, 2024-06-15:
After reboot of nas

Dmesg gives:

[   18.702369] systemd[1]: Created slice r8152's slice.
[   18.709346] systemd[1]: Starting r8152's slice.
[   83.163409] r8152: module verification failed: signature and/or required key missing - tainting kernel
[   83.175643] usbcore: registered new interface driver r8152
[   84.558869] r8152 3-4:1.0 eth4: v2.18.1 (2024/05/20)
[   84.564624] r8152 3-4:1.0 eth4: This product is covered by one or more of the following patents:
[   84.582127] r8152 3-4:1.0 eth4: chip rev 14
[   87.946990] r8152 3-4:1.0 eth4: Promiscuous mode enabled
[   87.956429] r8152 3-4:1.0 eth4: carrier on
[  103.615764] r8152 3-4:1.0 eth4: Promiscuous mode enabled
[  265.869263] r8152 3-4:1.0: read type=0x0133, index=0xb5d4 fail -71
[  265.881697] WARNING: at //source/r8152/r8152.c:1395 ocp_read_word+0x7e/0x90 [r8152]()

Iperf3 test is getting about 1.8 - 2.2 Gbits/sec

Looks better now. What do you make of the warning?

Kind regards
Rune

bb-qq, 2024-06-16:
The symptoms occurring are probably the same as before. However, the new driver has enhanced error handling and recovery procedures, which may have improved performance.

bb-qq, 2024-06-16:
I would like to close this issue once as there does not seem to be anything more I can do.

If you have any questions, feel free to re-open it.

closed 2024-06-16 by bb-qq