bb-qq/r8152 #335: DS416Play - CAC-1420 only detected as USB2 device on any USB port

chvvkumar, 2023-08-27:
### Description of the problem

DS416Play - CAC-1420 only detected as USB2 device on any USB port. Tried it on all three ports but I only see the device is detected as only capable of 480Mbps. This is reflected in file copy speeds ~30MB/s

### Description of your products

Synology DS416Play>     - and output of ``uname -a`` command
DSM 7.1.1-42962
CLUB 3D CAC-1420 USB 3.2 Type A to RJ45 2.5 Gigabit LAN Ethernet Cable Adapter Windows 10, 8.1, Mac OSX 10.6 to 10.13

### Description of your environment

Directly connected to PC using CAT6 cable

PC NIC: Realtek Gaming 2.5GbE Family Controller rtl8125bg

### Output of ``dmesg`` command

[dmesg.txt](https://github.com/bb-qq/r8152/files/12447033/dmesg.txt)

### Output of ``lsusb`` command

````
chvvkumar@diskstation:~$ lsusb
|__usb1          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (xhci_hcd 0000:00:14.0) hub
  |__1-5         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 6500739E2DE41647)
|__usb2          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (xhci_hcd 0000:00:14.0) hub
|__usb3          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (Linux 3.10.108 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub
  |__3-1         0bda:8156:3104 00  2.10  480MBit/s 200mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 401000001)
|__usb4          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (Linux 3.10.108 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub
````

### Output of ``ifconfig -a`` command

````
chvvkumar@diskstation:~$ ifconfig -a
docker0   Link encap:Ethernet  HWaddr 02:42:A0:16:8B:3C
          inet addr:172.17.0.1  Bcast:172.17.255.255  Mask:255.255.0.0
          inet6 addr: fe80::42:a0ff:fe16:8b3c/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:10695 errors:0 dropped:0 overruns:0 frame:0
          TX packets:13101 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:958844 (936.3 KiB)  TX bytes:16755099 (15.9 MiB)

docker72d Link encap:Ethernet  HWaddr A2:C7:C7:F4:F6:B3
          inet6 addr: fe80::a0c7:c7ff:fef4:f6b3/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:10695 errors:0 dropped:0 overruns:0 frame:0
          TX packets:13104 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:1108574 (1.0 MiB)  TX bytes:16755357 (15.9 MiB)

eth0      Link encap:Ethernet  HWaddr 00:11:32:6A:F6:6B
          inet addr:192.168.1.5  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::211:32ff:fe6a:f66b/64 Scope:Link
          inet6 addr: fd5d:60d4:d2eb:de33:211:32ff:fe6a:f66b/64 Scope:Global
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:149769 errors:0 dropped:0 overruns:0 frame:0
          TX packets:104202 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:55005290 (52.4 MiB)  TX bytes:64238735 (61.2 MiB)

eth1      Link encap:Ethernet  HWaddr 00:11:32:6A:F6:6C
          inet addr:192.168.2.5  Bcast:192.168.2.255  Mask:255.255.255.0
          inet6 addr: fe80::211:32ff:fe6a:f66c/64 Scope:Link
          inet6 addr: fd5d:60d4:d2eb:de33:211:32ff:fe6a:f66c/64 Scope:Global
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:61279 errors:0 dropped:0 overruns:0 frame:0
          TX packets:2801 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:10445493 (9.9 MiB)  TX bytes:463928 (453.0 KiB)

eth2      Link encap:Ethernet  HWaddr A0:CE:C8:6E:7A:A6
          inet addr:192.168.10.5  Bcast:192.168.10.255  Mask:255.255.255.0
          inet6 addr: fe80::a2ce:c8ff:fe6e:7aa6/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:9000  Metric:1
          RX packets:23516 errors:0 dropped:0 overruns:0 frame:0
          TX packets:31606 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:1380879 (1.3 MiB)  TX bytes:273744261 (261.0 MiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:27933 errors:0 dropped:0 overruns:0 frame:0
          TX packets:27933 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:7278374 (6.9 MiB)  TX bytes:7278374 (6.9 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

````

chvvkumar, 2023-08-27:
Just did some testing to rule out my cabling as the cable goes through two different keystones and three different cable segments.

Herei s my original layout:
````
Desktop <-> Keystone Wall jack <-> Rack Patch Panel <-> DS416Play
````

Testing with my RaspberryPi
````
Desktop <-> Keystone Wall jack <-> Rack Patch Panel <-> RPi
````

````
pi@pihole:[~]: sudo ethtool eth1
Settings for eth1:
	Supported ports: [ TP MII ]
	Supported link modes:   10baseT/Half 10baseT/Full
	                        100baseT/Half 100baseT/Full
	                        1000baseT/Half 1000baseT/Full
	                        2500baseT/Full
	Supported pause frame use: No
	Supports auto-negotiation: Yes
	Supported FEC modes: Not reported
	Advertised link modes:  10baseT/Half 10baseT/Full
	                        100baseT/Half 100baseT/Full
	                        1000baseT/Full
	Advertised pause frame use: Symmetric Receive-only
	Advertised auto-negotiation: Yes
	Advertised FEC modes: Not reported
	Link partner advertised link modes:  10baseT/Half 10baseT/Full
	                                     100baseT/Half 100baseT/Full
	                                     1000baseT/Full
	                                     2500baseT/Full
	Link partner advertised pause frame use: Symmetric Receive-only
	Link partner advertised auto-negotiation: Yes
	Link partner advertised FEC modes: Not reported
	Speed: 1000Mb/s
	Duplex: Full
	Port: MII
	PHYAD: 32
	Transceiver: internal
	Auto-negotiation: on
	Supports Wake-on: pumbg
	Wake-on: g
	Current message level: 0x00007fff (32767)
			       drv probe link timer ifdown ifup rx_err tx_err tx_queued intr tx_done rx_status pktdata hw wol
	Link detected: yes

````

I noticed between the two ethtool outputs 
On the NAS: the advertised speeds of the partner (desktop) do not show 2.5G speeds
On the Pi: advertised speed now shows 2.5g but it still connects at 1g.

bb-qq, 2023-08-28:
> CAC-1420 only detected as USB2 device on any USB port. 

This symptom is strange.

* If you connect CAC-1420 to your PC will it be recognized as a USB 3.0 device?
* If you connect other USB 3.0 devices such as HDDs to the DS416Play, will they be recognized as USB 3.0 devices?

chvvkumar, 2023-08-29:
> * If you connect CAC-1420 to your PC will it be recognized as a USB 3.0 device?

Yes, it is recognized as a SuperSpeed device on my Windows desktop:

````
    =========================== USB Port1 ===========================

Connection Status        : 0x01 (Device is connected)
Port Chain               : 1-24-1
Properties               : 0x01
 IsUserConnectable       : yes
 PortIsDebugCapable      : no
 PortHasMultiCompanions  : no
 PortConnectorIsTypeC    : no
ConnectionIndex          : 0x01 (Port 1)
CompanionIndex           : 0
 CompanionHubSymLnk      : USB#VID_174C&PID_2074#5&2a6ec909&0&8#{f18a0e88-c30c-11d0-8815-00a0c906bed8}
 CompanionPortNumber     : 0x01 (Port 1)
 -> CompanionPortChain   : 1-8-1

      ========================== Summary =========================
Vendor ID                : 0x0BDA (Realtek Semiconductor Corp.)
Product ID               : 0x8156
USB Version              : 3.2 Gen 1
Port maximum Speed       : SuperSpeed
Device maximum Speed     : SuperSpeed
Device Connection Speed  : SuperSpeed
Self powered             : no
Demanded Current         : 256 mA
Used Endpoints           : 4

      ======================== USB Device ========================

        +++++++++++++++++ Device Information ++++++++++++++++++
Friendly Name            : Realtek Gaming USB 2.5GbE Family Controller
Device Description       : Realtek Gaming USB 2.5GbE Family Controller
Device Path 1            : \\?\USB#VID_0BDA&PID_8156#401000001#{a5dcbf10-6530-11d2-901f-00c04fb951ed} (GUID_DEVINTERFACE_USB_DEVICE)
Device Path 2            : \\?\USB#VID_0BDA&PID_8156#401000001#{cac88484-7515-4c03-82e6-71a87abac361}\{57ca6c7e-ace0-4c17-bcbf-e060e7798639} (GUID_DEVINTERFACE_NET)
Device Path 3            : \\?\USB#VID_0BDA&PID_8156#401000001#{ad498944-762f-11d0-8dcb-00c04fc3358c}\{57ca6c7e-ace0-4c17-bcbf-e060e7798639} (GUID_NDIS_LAN_CLASS)
Kernel Name              : \Device\USBPDO-18
Device ID                : USB\VID_0BDA&PID_8156\401000001
Hardware IDs             : USB\VID_0BDA&PID_8156&REV_3104 USB\VID_0BDA&PID_8156
Driver KeyName           : {4d36e972-e325-11ce-bfc1-08002be10318}\0016 (GUID_DEVCLASS_NET)
Driver                   : \SystemRoot\System32\DriverStore\FileRepository\rtucx21x64.inf_amd64_0fee41a4ed424d5a\rtucx21x64.sys (Version: 11.1.621.2021  Date: 2022-05-07)
Driver Inf               : C:\Windows\inf\rtucx21x64.inf
Legacy BusType           : PNPBus
Class                    : Net
Class GUID               : {4d36e972-e325-11ce-bfc1-08002be10318} (GUID_DEVCLASS_NET)
Service                  : rtucx21x64
Enumerator               : USB
Location Info            : Port_#0001.Hub_#0003
Location IDs             : PCIROOT(0)#PCI(1400)#USBROOT(0)#USB(24)#USB(1), ACPI(_SB_)#ACPI(PC00)#ACPI(XHCI)#ACPI(RHUB)#ACPI(SS08)#ACPI(EP01)
Container ID             : {d180beb6-017a-5986-8625-5465a523528c}
Manufacturer Info        : Realtek
Capabilities             : 0x94 (Removable, UniqueID, SurpriseRemovalOK)
Status                   : 0x0180600A (DN_DRIVER_LOADED, DN_STARTED, DN_DISABLEABLE, DN_REMOVABLE, DN_NT_ENUMERATOR, DN_NT_DRIVER)
Problem Code             : 0
WakeFromSleepState       : 1
Power State              : D0 (supported: D0, D1, D2, D3, wake from D0, wake from D1, wake from D2)

        +++++++++++++++++ Registry USB Flags +++++++++++++++++
HKEY_LOCAL_MACHINE
\SYSTEM\CurrentControlSet\Control\usbflags\0BDA81563104
 osvc                    : REG_BINARY 00 00

````
[USB Info.txt](https://github.com/bb-qq/r8152/files/12459551/USB.txt)

> * If you connect other USB 3.0 devices such as HDDs to the DS416Play, will they be recognized as USB 3.0 devices?

Yes, I connected a SanDisk flash drive and it is recognized properly:

````
chvvkumar@diskstation:~$ lsusb
|__usb1          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (xhci_hcd 0000:00:14.0) hub
  |__1-5         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 6500739E2DE41647)
|__usb2          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (xhci_hcd 0000:00:14.0) hub
  |__2-2         0781:5583:0100 00  3.20 5000MBit/s 896mA 1IF  ( USB  SanDisk 3.2Gen1 0101635bf66043cccf033f652a345b35e7e201d84c96d249a4ec2e69e80d17cc01b900000000000000000000d5bd05e5ff17460083558107bf2d509b)
|__usb3          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (Linux 3.10.108 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub
  |__3-2         051d:0002:0101 00  1.10  1.5MBit/s 2mA 1IF  (American Power Conversion Back-UPS ES 750G FW:908.W4 .D USB FW:W4  4B1614P44143  )
|__usb4          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (Linux 3.10.108 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub
````

chvvkumar, 2023-08-29:
Actually, not sure what happened but after I removed the USB Flash drive and reinstalled the USB NIC, it started working fine! Here's the relavent output from dmesg, ethtool and ifconfig:

````
[180935.423945] usb 2-2: new SuperSpeed USB device number 3 using xhci_hcd
[180935.442022] usb 2-2: Parent hub missing LPM exit latency info.  Power management will be impacted.
[180935.578117] usb 2-2: reset SuperSpeed USB device number 3 using xhci_hcd
[180935.596038] usb 2-2: Parent hub missing LPM exit latency info.  Power management will be impacted.
[180935.613427] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88010eebfc80
[180935.623690] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88010eebfcc0
[180935.633981] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88010eebfd00
[180935.716873] r8152 2-2:1.0 eth2: v2.17.1 (2023/06/13)
[180935.722554] r8152 2-2:1.0 eth2: This product is covered by one or more of the following patents:
                		US6,570,884, US6,115,776, and US6,327,625.

[180935.740355] r8152 2-2:1.0 eth2: chip rev 14
[180935.745172] r8152 2-2:1.0 eth2: rx_buf_sz will be overridden to 16384
chvvkumar@diskstation:~$ lsusb
|__usb1          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (xhci_hcd 0000:00:14.0) hub
  |__1-5         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 6500739E2DE41647)
|__usb2          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (xhci_hcd 0000:00:14.0) hub
  |__2-2         0bda:8156:3104 00  3.20 5000MBit/s 256mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 401000001)
|__usb3          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (Linux 3.10.108 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub
  |__3-2         051d:0002:0101 00  1.10  1.5MBit/s 2mA 1IF  (American Power Conversion Back-UPS ES 750G FW:908.W4 .D USB FW:W4  4B1614P44143  )
|__usb4          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (Linux 3.10.108 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub

chvvkumar@diskstation:~$ sudo ethtool eth2
Settings for eth2:
	Supported ports: [ MII ]
	Supported link modes:   10baseT/Half 10baseT/Full
	                        100baseT/Half 100baseT/Full
	                        1000baseT/Full
	                        2500baseX/Full
	Supported pause frame use: No
	Supports auto-negotiation: Yes
	Supported FEC modes: Not reported
	Advertised link modes:  10baseT/Half 10baseT/Full
	                        100baseT/Half 100baseT/Full
	                        1000baseT/Full
	                        2500baseX/Full
	Advertised pause frame use: Symmetric Receive-only
	Advertised auto-negotiation: Yes
	Advertised FEC modes: Not reported
	Link partner advertised link modes:  10baseT/Half 10baseT/Full
	                                     100baseT/Half 100baseT/Full
	                                     1000baseT/Full
	Link partner advertised pause frame use: No
	Link partner advertised auto-negotiation: Yes
	Link partner advertised FEC modes: Not reported
	Speed: 2500Mb/s
	Duplex: Full
	Port: MII
	PHYAD: 32
	Transceiver: internal
	Auto-negotiation: on
	Supports Wake-on: pumbg
	Wake-on: g
	Current message level: 0x00007fff (32767)
			       drv probe link timer ifdown ifup rx_err tx_err tx_queued intr tx_done rx_status pktdata hw wol
	Link detected: yes

````

bb-qq, 2023-09-10:
If you insert the USB 3.0 connector slowly, it will be recognized as a USB 2.0 device. It sounds like a lie, but it is true. Your case may be related to this.

Since your problem seems to be solved and I have added this issue to the troubleshooting page, I am closing this issue.

closed 2023-09-10 by bb-qq