bb-qq/r8152 #336: DS920+ DSM 7.2-64570 Update 3 - driver 2.17.1-1 does not work

Fjellvanderer, 2023-08-28:
After installation of the latest driver no connection is possible. Tried it sveral times but no other result.

Until update everthing works fine (driver 2.16.3-3).

Any idea for troubleshooting?
````

jvivona, 2023-09-07:
i can concur - we were running update 1 and everything was working.   moved to update 3 and the network interface is gone.  have tried reinstalling & reboot - no progress.

eriknelson, 2023-10-01:
I have a DSM920+, DSM 7.2-64570 Update 1. I attempted to install https://github.com/bb-qq/r8152/releases/download/2.17.1/r8152-geminilake-2.17.1-1_7.2.spk, choosing the 7.2 version for geminilake. Installed and failed as expected. Ran the instruction. At this point the instructions are not clear. I basically just X'd out of the window and left it in an "Error" install state. Then I tried to manually install again. It said it "Failed to repair the install".

So after the initial package fails to install, and the `install` cli command is run...exactly what steps should be take.?

DervilRus, 2023-10-03:
Today I updated the DSM version on DS920+ from DSM 7.2-64570 to DSM 7.2.1-69057. Everything works great on both versions without any problems

bb-qq, 2023-10-09:
@Fjellvanderer, did your problem resolve with a full reinstall?

Fjellvanderer, 2023-10-11:
Hi!

Full reinstall ist outstanding. Will be done on the course of new HDs in
the near future.
Currently ot is not possible to do this due to workload.

Best regards

Fjellvanderer, 2023-10-15:
After complete new setup with new and  larger drives the 2.5 GbE connection
seems to be stabile. Same Software Status as before.

Currently the full restore is running with 1 GbE to secure availability of
data next working day.

closed 2024-04-18 by bb-qq