bb-qq/r8152 #339: J5Create Seems to be working and stable

meallonardo, 2023-09-13:
### Working

> J5create JCH474.

### Description of your products

> - DS1813+
>   Linux MINK 3.10.108 #42962 SMP Mon May 29 14:36:28 CST 2023 x86_64 GNU/Linux synology_cedarview_1813+
>   DSM version 7.1.1-42962 Update 6
>   J5create JCH474

### Description of your environment

> - Connected through Netgear 1GB switch
>   
>    
> Cat 6

### Output of ``dmesg`` command
221.890942] usbcore: registered new interface driver aqc111
[  303.678858] usb 8-1: USB disconnect, device number 2
[  355.155634] usb 8-2: new SuperSpeed USB device number 3 using etxhci_hcd-170202
[  394.180652] usb 8-2: USB disconnect, device number 3
[  399.417308] usb 2-5: new high-speed USB device number 4 using ehci-pci
[  952.240838] Module [aqc111] is removed. 
[  952.244831] usbcore: deregistering interface driver aqc111
[  952.253618] Module [usbnet] is removed. 
[  952.260213] Module [mii] is removed. 
[ 1930.978664] usb 2-5: USB disconnect, device number 4
[ 2231.715745] usb 7-1: new high-speed USB device number 3 using etxhci_hcd-170202
[ 2231.734333] usb 7-1: not running at top speed
[ 2235.199462] usb 7-1: new high-speed USB device number 3 using etxhci_hcd-170202
[ 2235.218036] usb 7-1: not running at top speed
[ 2238.683094] usb 7-1: new high-speed USB device number 3 using etxhci_hcd-170202
[ 2238.701740] usb 7-1: not running at top speed
[ 2242.166721] usb 7-1: new high-speed USB device number 3 using etxhci_hcd-170202
[ 2242.185320] usb 7-1: not running at top speed
[ 2245.650352] usb 7-1: new high-speed USB device number 3 using etxhci_hcd-170202
[ 2245.668900] usb 7-1: not running at top speed
[ 2249.133974] usb 7-1: new high-speed USB device number 3 using etxhci_hcd-170202
[ 2249.152478] usb 7-1: not running at top speed
[ 2249.160846] Got empty serial number. Generate serial number from product.
[ 2249.168589] ethub 7-1:1.0: USB hub found
[ 2249.172844] ethub 7-1:1.0: 4 ports detected
[ 2250.875412] usb 7-1.4: new high-speed USB device number 4 using etxhci_hcd-170202
[ 2363.302363] usbcore: registered new interface driver r8152
[ 2364.435569] usb 7-1.4: reset high-speed USB device number 4 using etxhci_hcd-170202
[ 2364.586249] r8152 7-1.4:1.0 eth4: v2.17.1 (2023/06/13)
[ 2364.591451] r8152 7-1.4:1.0 eth4: This product is covered by one or more of the following patents:
               		US6,570,884, US6,115,776, and US6,327,625.

[ 2364.607229] r8152 7-1.4:1.0 eth4: chip rev 6
[ 2364.611574] r8152 7-1.4:1.0 eth4: rx_buf_sz will be overridden to 16384
[ 2365.553836] IPv6: ADDRCONF(NETDEV_UP): eth4: link is not ready
[ 2368.216128] IPv6: ADDRCONF(NETDEV_CHANGE): eth4: link becomes ready
[ 2368.226534] r8152 7-1.4:1.0 eth4: carrier on

````
(paste output here, or attach as a file)
````

### Output of ``lsusb`` command
lsusb
|__usb1          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd 0000:00:1a.7) hub
|__usb2          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd 0000:00:1d.7) hub
  |__2-1         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 650045FD10852167)
|__usb3          1d6b:0001:0310 09  1.10   12MBit/s 0mA 1IF  (uhci_hcd 0000:00:1a.0) hub
|__usb4          1d6b:0001:0310 09  1.10   12MBit/s 0mA 1IF  (uhci_hcd 0000:00:1d.0) hub
|__usb5          1d6b:0001:0310 09  1.10   12MBit/s 0mA 1IF  (uhci_hcd 0000:00:1d.1) hub
  |__5-2         051d:0002:0090 00  2.00   12MBit/s 2mA 1IF  (American Power Conversion Back-UPS ES 600M1 FW:928.a9 .D USB FW:a9  4B2216P16436  )
|__usb6          1d6b:0001:0310 09  1.10   12MBit/s 0mA 1IF  (uhci_hcd 0000:00:1d.2) hub
|__usb7          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (Linux 3.10.108 etxhci_hcd-170202 Etron xHCI Host Controller 0000:05:00.0) hub
  |__7-1         2109:2817:0214 09  2.10  480MBit/s 0mA 1IF  (VIA Labs, Inc.          USB2.0 Hub              ffffff94ffffffb4ffffff94ffffffb4) hub
    |__7-1.4     0bda:8153:3000 00  2.10  480MBit/s 200mA 1IF  (Realtek USB 10/100/1000 LAN 000001)
|__usb8          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (Linux 3.10.108 etxhci_hcd-170202 Etron xHCI Host Controller 0000:05:00.0) hub

````
(paste output here)
````

### Output of ``ifconfig -a`` command
ifconfig -a
bond0     Link encap:Ethernet  HWaddr 00:11:32:2E:D6:AE  
          inet addr:192.168.2.2  Bcast:192.168.2.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MASTER MULTICAST  MTU:1500  Metric:1
          RX packets:7434 errors:0 dropped:0 overruns:0 frame:0
          TX packets:8139 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:894692 (873.7 KiB)  TX bytes:839137 (819.4 KiB)

eth0      Link encap:Ethernet  HWaddr 00:11:32:2E:D6:AC  
          inet addr:192.168.0.151  Bcast:192.168.0.255  Mask:255.255.255.0
          inet6 addr: fe80::211:32ff:fe2e:d6ac/64 Scope:Link
          inet6 addr: 2600:1008:a002:9b20:211:32ff:fe2e:d6ac/64 Scope:Global
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:17829 errors:0 dropped:0 overruns:0 frame:0
          TX packets:21438 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:8711497 (8.3 MiB)  TX bytes:17624691 (16.8 MiB)

eth1      Link encap:Ethernet  HWaddr 00:11:32:2E:D6:AD  
          inet addr:169.254.175.56  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth2      Link encap:Ethernet  HWaddr 00:11:32:2E:D6:AE  
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:7434 errors:0 dropped:0 overruns:0 frame:0
          TX packets:8139 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:894692 (873.7 KiB)  TX bytes:839137 (819.4 KiB)

eth3      Link encap:Ethernet  HWaddr 00:11:32:2E:D6:AE  
          UP BROADCAST SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth4      Link encap:Ethernet  HWaddr 00:E0:4F:82:4C:FB  
          inet addr:192.168.0.79  Bcast:192.168.0.255  Mask:255.255.255.0
          inet6 addr: 2600:1008:a002:9b20:2e0:4fff:fe82:4cfb/64 Scope:Global
          inet6 addr: fe80::2e0:4fff:fe82:4cfb/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:3289 errors:0 dropped:0 overruns:0 frame:0
          TX packets:4960 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:487152 (475.7 KiB)  TX bytes:5860891 (5.5 MiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:13790 errors:0 dropped:0 overruns:0 frame:0
          TX packets:13790 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:1816916 (1.7 MiB)  TX bytes:1816916 (1.7 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

````
(paste output here)
````

Sjekke, 2023-09-13:
So you installed a 2,5gbit adapter on a 1gb switch?

meallonardo, 2023-09-13:
Yes, I needed another Ethernet port.  Didn't care about the speed.

Sjekke, 2023-09-13:
And you already had 4 gbit ports ... just out of curiosity... where do you need them for?

meallonardo, 2023-09-13:
It's my backup nas.  The 4 1 gig ports are bonded and connected directly to my other nas. Ds1819+   It replicates all my data. I needed 1 port for admin.  Didn't need to be fast.

meallonardo, 2023-09-13:
Sorry if my config is stupid.  Just wanted to report my success as thanks for making the driver available.