bb-qq/r8152 #341: DS916+ DSM 7.1.1-42962 Update 6 - driver 2.17.1-1 unstable does not work

johnbradley101, 2023-10-04:
### Description of the problem

Unstable Club 3D CAC-1420 USB 3.2 Gen1 Adapter Type A to 2.5 Gigabit Ethernet.
I can get it to run and it works fine for a period of time. then it stops working and it usually needs a system restart to bring it back up.
I can enable the driver and get the USB adapter to appear as LAN 3.  It works well for a period of time and they it becomes unresponsive and I have to revert to the built in ethernet ports at 1Gbps.
When it works I am able to transfer files at 200-240 MB per sec and the speed is consistent.

### Description of your products
Synology DS916+, 8GB RAM, 
Linux synology 3.10.108 #42962 SMP Mon May 29 14:35:41 CST 2023 x86_64 GNU/Linux synology_braswell_916+
DSM 7.1.1-42962 Update 6
Club 3D CAC-1420 USB 3.2 Gen1 Adapter Type A to 2.5 Gigabit Ethernet

bb-qq Driver version 2.17.1-1

> - the product name of your NAS model
>     - and output of ``uname -a`` command
>     - DSM version x.x
> - the product name of the ethernet dongle

### Description of your environment

Windows 10 PC -> Zyxel XGS1250-12 switch using Cat6 cable, 10Gb port, running at 2.5gbps
--> No issues with the PC's neetwork connection: it connects to the rest ofth ge LAN and the internet without issues.

Nas -> Zyxel XGS1250-12 switch using Cat6 cable, 10Gb port, running at 2.5gbps
Nas -> Zyxel XGS1250-12 switch using Cat6 cable, 1Gb port, running at 1gbps  (I'm using this as a fallback when the USB 2.5gb adapter fails)

The lan cables are only a few metres long and in the same room as the PC.

### Output of ``dmesg`` command

see attached file

### Output of ``lsusb`` command

|__usb1          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (xhci_hcd 0000:00:14.0) hub
  |__1-5         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 650065AEA6E41658)
|__usb2          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (xhci_hcd 0000:00:14.0) hub
  |__2-2         0bda:8156:3000 00  3.20 5000MBit/s 512mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 000000001)
|__usb3          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (Linux 3.10.108 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub
  |__3-1         2109:2817:9023 09  2.10  480MBit/s 0mA 1IF  (VIA Labs, Inc.          USB2.0 Hub              ffffff94ffffffb4ffffff94ffffffb4) hub
    |__3-1.4     051d:0002:0106 00  1.10  1.5MBit/s 24mA 1IF  (American Power Conversion Back-UPS XS 950U   FW:925.T2 .I USB FW:T2  3B1718X08717  )
  |__3-2         045b:0209:0100 09  2.10  480MBit/s 0mA 1IF  ( ffffffd1ffffffb2ffffffdbffffffad) hub
|__usb4          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (Linux 3.10.108 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub
  |__4-1         2109:0817:9023 09  3.10 5000MBit/s 0mA 1IF  (VIA Labs, Inc.          USB3.0 Hub              ffffff94ffffffb4ffffff94ffffffb0) hub
    |__4-1.1     174c:1153:0100 00  3.00 5000MBit/s 0mA 1IF  (Ugreen Ugreen Storage Device 26A1EE83184C)
    |__4-1.2     1058:25ed:1031 00  3.10 5000MBit/s 8mA 1IF  (Western Digital My Book 25ED 59354A5A42563055)
  |__4-2         045b:0210:0100 09  3.00 5000MBit/s 0mA 1IF  ( ffffffd1ffffffb2ffffffdbffffffa0) hub

### Output of ``ifconfig -a`` command

Taken when the adapter IS working
docker0   Link encap:Ethernet  HWaddr 02:42:E5:84:70:3D
          inet addr:172.17.0.1  Bcast:172.17.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:
[dmesg.txt](https://github.com/bb-qq/r8152/files/12801282/dmesg.txt)
0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

docker018 Link encap:Ethernet  HWaddr 36:69:17:F4:F6:26
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

docker0e3 Link encap:Ethernet  HWaddr 86:5C:33:0D:B2:3A
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth0      Link encap:Ethernet  HWaddr 00:11:32:60:BE:85
          inet addr:192.168.0.21  Bcast:192.168.0.255  Mask:255.255.255.0
          inet6 addr: fe80::211:32ff:fe60:be85/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:9000  Metric:1
          RX packets:17793 errors:0 dropped:0 overruns:0 frame:0
          TX packets:22698 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:11420385 (10.8 MiB)  TX bytes:20595957 (19.6 MiB)

eth1      Link encap:Ethernet  HWaddr 00:11:32:60:BE:86
          inet addr:169.254.175.207  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth2      Link encap:Ethernet  HWaddr A0:CE:C8:E8:8F:EE
          inet addr:192.168.0.20  Bcast:192.168.0.255  Mask:255.255.255.0
          inet6 addr: fe80::a2ce:c8ff:fee8:8fee/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:2031 errors:0 dropped:0 overruns:0 frame:0
          TX packets:219 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:178151 (173.9 KiB)  TX bytes:40173 (39.2 KiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:1240 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1240 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:174359 (170.2 KiB)  TX bytes:174359 (170.2 KiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

johnbradley101, 2023-10-04:
[dmesg.txt](https://github.com/bb-qq/r8152/files/12801290/dmesg.txt)

bb-qq, 2023-10-09:
- Does the same problem occur when using CAC-1420 with a 1 Gbps link?
- Is it the same when using the older driver 2.16.3-4?

johnbradley101, 2023-10-11:
Thanks for coming back to me.
I moved the cable to a 1 Gbps port : 
- same Cat6 cable 
- and same switch (just a different port)
- **So far it's been stable for about 30 hours:**  The highlighted time is when the change took place.
- 
[image]

**I haven't changed the driver yet.**
->  I was going to swap over to a different 10gb port on the switch (the one used by my PC) to rule out a dodgy port on the switch.

johnbradley101, 2023-10-11:
In the device settings I have also changed IEEE 802.3az EEE to disabled :
[image]

johnbradley101, 2023-10-12:
so far so good.... the connection has been solid and has remained up overnight. 
I think it must be the IEEE 802.3az EEE : energy saving setting. .

I'm going to close this issue as resolved: ensure the energy saving settings are set to disabled to avoid the adapter becoming unresponsive and being dropped by Synology.

johnbradley101, 2023-10-12:
thanks for all your help and the awesome driver.

closed 2023-10-12 by johnbradley101