bb-qq/r8152 #344: Could you help to release a new version to support monaco in dsm 6.2?

gengchaogit, 2023-10-09:
Hi,

I'm searching all the release note, but not found a release version can be support monaco in dsm 6.2.

https://github.com/bb-qq/r8152/releases/tag/2.17.1

[r8152-monaco-2.17.1-1_7.1.spk](https://github.com/bb-qq/r8152/releases/download/2.17.1/r8152-monaco-2.17.1-1_7.1.spk)
380 KB
Jul 15
[r8152-monaco-2.17.1-1_7.2.spk](https://github.com/bb-qq/r8152/releases/download/2.17.1/r8152-monaco-2.17.1-1_7.2.spk)
360 KB
Jul 15

Not found monaco platform dsm 6.2.

Best regards,
Chao

gengchaogit, 2023-10-09:
my usb netcard is 8156B

bb-qq, 2023-10-15:
I added [r8152-monaco-2.17.1-1_6.2.spk](https://github.com/bb-qq/r8152/releases/download/2.17.1/r8152-monaco-2.17.1-1_6.2.spk).

gengchaogit, 2023-10-15:
Oh, thank you !

bb-qq, 2023-10-29:
The work has been completed and this Issue is closed. It would be appreciated if you could let us know at [this issue](https://github.com/bb-qq/r8152/issues/135) whether it worked stably.

closed 2023-10-29 by bb-qq